fwupd/firmware-lenovo #369: Timeout too short for user interaction when updating Lenovo Thunderbolt 4 Dock (40B0) firmware

rclaasen, 2023-09-11:
I was updating my Lenovo Thunderbolt 4 Dock (40B0) firmware when I got the following message:

```
ubuntu@ubuntu:~$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • Integrated Camera
 • Prometheus
 • Prometheus IOTA Config
 • SSD 980 PRO 2TB
 • System Firmware
 • ThinkPad Thunderbolt 4 Dock
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB3.0 Hub
 • VMM6212
No updatable devices
ubuntu@ubuntu:~$ sudo fwupdmgr refresh
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 5 local devices supported
ubuntu@ubuntu:~$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • Integrated Camera
 • Prometheus IOTA Config
 • SSD 980 PRO 2TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB3.0 Hub
 • VMM6212
Devices with the latest available firmware version:
 • Embedded Controller
 • Prometheus
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade ThinkPad Thunderbolt 4 Dock from 10.15 to 10.16?                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Before continuing, ensure the following:                                     ║
║                                                                              ║
║ • Your computer battery life if over 25%.                                    ║
║ • Do not unplug the dock during update.                                      ║
║ • Do not put your computer into sleep or hibernate mode during update.       ║
║ • This update typically may take up to 15 minutes depending on your          ║
║ notebook and dock firmware version.                                          ║
║ • Please ensure fwupd version is 1.9.2 or above.                             ║
║                                                                              ║
║ ThinkPad Thunderbolt 4 Dock and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Action Required: Please unplug the USB cable. The update will continue in 30 seconds from now.
Action Required: The update will continue when the device USB cable has been re-inserted.
Waiting…                 [************************************** ]
failed to wait for cleanup replug: device cf9a31bd537d3261c5ae2979f5db1ff4b8c29721 did not come back
ubuntu@ubuntu:~$ sudo fwupdmgr update
```

Reason was simple: the update was taking a long time, and I left my laptop to do the update unattended.  

Bad idea, since there was an action required on my part: disconnect & reconnect the dock. 

Now the good news is that I could re-apply the firmware update using `fwupdmgr install`. That worked flawlessly, but it took me a while to find that. 

Suggestions: 

 1. Warn the user upfront that user interaction is needed; this should be in the list "Before continuing, ensure the following".
 2. Remove the timeout, and ask for user input instead (eg. "Please unplug the USB cable, and press ENTER to continue"). Increasing the time-out is another option. Given that the install takes a quarter of an hour, I suggest it should be really long (an hour?). 

Again, the firmware install went flawlessly the second time, but the messaging could  be clearer.

Thanks for all the work!

hughsie, 2023-09-11:
What fwupd version? for 1.9.5 we merged https://github.com/fwupd/fwupd/commit/f954512b253100f283ddaaf4dcf452fb101bc25c

rclaasen, 2023-09-11:
1.9.4 on an Ubuntu 23.10 development image (running Ubuntu 22.04 LTS as daily). 

From reading through https://github.com/fwupd/fwupd/commit/f954512b253100f283ddaaf4dcf452fb101bc25c it seems this is already addressed. 

Many thanks, apologies for the inconvenience!

```
ubuntu@ubuntu:~$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.4
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.3.0-7-generic
runtime   org.freedesktop.fwupd         1.9.4
```

closed 2023-09-11 by rclaasen