fwupd/firmware-lenovo #371: Battery capacity reported as 0% after firmware upgrade on Thinkpad P15 Gen 2i (Model 20YRS0BM00)

ohearnk, 2023-09-22:
Hello,

After recent EC and BIOS firmware updates on a brand new Thinkpad P15 Gen 2i (Model 20YRS0BM00) via fwupd/LVFS, battery charge information has disappeared and charge status is reported as 0%.  Interestingly, the battery charge percentage indicator was fine through a couple of charge/discharge cycles on stock firmware (i.e., before upgrade).  

Diagnostic information below.

<details>

<summary>fwupdmgr get-history</summary>

fwupdmgr get-history:
LENOVO 20YRS0BM00
│
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Previous version:   10.01.3273255
│ │   Update State:       Success
│ │   Last modified:      2023-09-15 20:39
│ │   GUID:               8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Updatable
│ │                       • Signed Payload
│ │ 
│ └─Prometheus Fingerprint Reader Update:
│       New version:      10.01.3478575
│       Remote ID:        lvfs
│       Release ID:       10694
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       License:          Proprietary
│       Size:             425.5 kB
│       Created:          2021-06-09
│       Urgency:          Medium
│       Tested by Lenovo:
│         Tested:         2023-05-05
│         Distribution:   ubuntu 22.04
│         Old version:    10.01.3273255
│         Version[fwupd]: 1.7.9
│       Tested by Lenovo:
│         Tested:         2023-02-28
│         Distribution:   ubuntu 22.04
│         Old version:    10.01.3273255
│         Version[fwupd]: 1.7.9
│       Tested by Lenovo:
│         Tested:         2022-07-07
│         Distribution:   ubuntu 22.04
│         Old version:    10.01.3158804
│         Version[fwupd]: 1.7.5
│       Vendor:           Synaptics
│       Duration:         2 seconds
│       Release Flags:    • Trusted metadata
│       Description:      
│       New features and enhancements:
│       
│       • Fix an authentication issue on FOP project.
│       Checksum:         9a3173547a0f44cd369c03b561018ce1d86834d6fef24e1ebb95fa1f8391521a
│     
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Previous version:   240.41.2142
│ │   Update State:       Success
│ │   Last modified:      2023-09-15 20:49
│ │   GUID:               666705ce-0624-4943-8738-09829a26c8ee
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
[... 148 lines cut ...]
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              0 Wh
    energy-empty:        0 Wh
    energy-full:         0 Wh
    energy-full-design:  0 Wh
    energy-rate:         0 W
    charge-cycles:       N/A
    percentage:          0%
    capacity:            100%
    icon-name:          'battery-caution-charging-symbolic'

</details>

<summary>journalctl  -b| grep upower</summary>

Sep 21 20:40:38 REDACTED upowerd[1348]: no valid voltage value found for device /sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:1e/PNP0C09:00/PNP0C0A:00/power_supply/BAT0, assuming 10V

</details>

Let me know if you need further information to diagnose this issue.

Thank you.

ohearnk, 2023-09-22:
Also, attempting to downgrade the firmware via fwupd results in the message `Check power failure, please insert the power source or charge the battery.` after rebooting despite the AC charger being plugged in.  This seems to be a similar issue encountered in #348 and #350.  Can you provide instructions for working around these power checks when downgrading the firmware?

ChiWei-Chen, 2023-09-22:
@ohearnk thanks for reporting this problem.
Have you tried the pin hole reset?
https://download.lenovo.com/pccbbs/pubs/p15_gen2_t15g_gen2_p17_gen2/ug/html_en/index.html#t=en%2FBottom_view.html&rhsearch=hole&rhhlterm=hole&rhsyns=%20

ohearnk, 2023-09-22:
Thank you for the response.  Can you clarify whether the pin hole reset is meant to resolve the battery capcity being reported as 0% or is it meant to allow firmware downgrade / reinstall?  If it is the latter, I still am receiving `Check power failure, please insert the power source or charge the battery.` when trying to downgrade the firmware.

Also, after trying the pin hole reset a few times, I did monentarily see a non-zero battery capacity upon rebooting without the AC adapter connected to the system.  However, after a few seconds the capacity reverts back to zero.  And there are different results from `upower` and `journalctl`:

<details>
<summary>upower<\summary>

```
upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               Celxpert
  model:                5B10W13959
  serial:               3786
  power supply:         yes
  updated:              Fri 22 Sep 2023 09:30:10 AM EDT (16 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              0 Wh
    energy-empty:        0 Wh
    energy-full:         94.57 Wh
    energy-full-design:  94 Wh
    energy-rate:         0 W
    charge-cycles:       7
    percentage:          0%
    capacity:            100%
    technology:          lithium-polymer
    icon-name:          'battery-caution-charging-symbolic'
  History (charge):
    1695389380	0.000	pending-charge
    1695389371	53.000	discharging
    1695389355	0.000	unknown
  History (rate):
    1695389380	0.000	pending-charge
    1695389371	32.633	discharging
    1695389355	0.000	unknown
```

<\details> 

<details>
<summary>journalctl<\summary>

```
Sep 22 09:29:15 REDACTED upowerd[1328]: energy_full (94.570000) is greater than energy_full_design (94.000000)

```

<\details>

ChiWei-Chen, 2023-09-25:
@ohearnk , thanks for the feedback. I shared this with FW team, I'll let you know if any progress (LO-2670 for tracking internally.)

ChiWei-Chen, 2023-10-03:
@ohearnk , as our team cannot reproduce this issue on multiple machines (with BIOS version 46w(1.27) and EC 30W(1.12)) and suspected this could be a HW issue. 
Please contact customer support for this issue. Thanks!