fwupd/firmware-lenovo #372: ThinkCentre M70s/M70t/M80s/M80t/M90s/M90t is outdated

PhilippWendler, 2023-09-22:
The [latest firmware for ThinkCentre M70s/M80s/M90s on LVFS](https://fwupd.org/lvfs/devices/com.lenovo.M90ts_M80ts_M70ts.BIOS.firmware) is 0.1.79 / M2TKT4FA, whereas the [latest firmware on the Lenovo support page](https://pcsupport.lenovo.com/de/de/products/desktops-and-all-in-ones/thinkcentre-m-series-desktops/thinkcentre-m70s/11db/11dbsapp00/downloads/ds545436-flash-bios-update-thinkcentre-m70t-m70s-m80t-m80s-m90t-m90s?category=BIOS) is M2TKT55A, 5 versions and at least one year newer. Please upload it also to LVFS.

My output of `fwupdmgr get-devices` is the following:
```
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.79
│ │   Minimum Version:    0.1.79
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              5deafc1a-69b6-4a95-8572-2c888cabf533
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Full disk encryption secrets may be invalidated when updating
```

mrhpearson, 2023-11-01:
Sorry for slow reply - I had missed this one in my inbox swamp.
Internal ticket LO-2728 created asking the FW team for an update.
Mark

RaymondmLei, 2023-11-16:
Latest BIOS M2TKT56A has been finished the LVFS publish, please check it. Thanks.

PhilippWendler, 2023-11-16:
Hm, even after a `fwupdmgr refresh --force`, `fwupdmgr get-updates` still does not list an update for me.

Full output of `get-devices`:
```
11DBSAPP00
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.11
│     Minimum Version:    0.1.11
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               a7f3a1a9-fd4e-4df9-8c9b-0ce37d1a7a6e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    96.33.1125
│     Minimum Version:    96.33.1125
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               6159fb9c-e80b-4d13-9018-bb00bd2f353a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.79
│ │   Minimum Version:    0.1.79
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              5deafc1a-69b6-4a95-8572-2c888cabf533
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Full disk encryption secrets may be invalidated when updating
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            711e4890-7203-50a5-80aa-2c06988616e6 ← UEFI\CRT_B4731FEF902AA3DA869F0803C84D732D790B0C23D095EDB2CCCCCB304FB00C53
│                         25c2b9af-7c95-564e-9b07-eccceacb46e8 ← UEFI\CRT_B4731FEF902AA3DA869F0803C84D732D790B0C23D095EDB2CCCCCB304FB00C53&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    458837
│     Minimum Version:    458837
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               5ae4922c-4451-4879-b74b-4aa65467f160
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─WDC PC SN730 SDBQNTY-256G-1001:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    11170101
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      21360J805280
      GUIDs:              7b974856-317d-5538-a915-de0a6353750f ← NVME\VEN_15B7&DEV_5006&SUBSYS_15B75006&REV_00
                          12c86995-0b90-5ec5-98f3-7a6ed4ca50e0 ← NVME\VEN_15B7&DEV_5006&SUBSYS_15B75006
                          a39943dd-3afb-54f8-b110-c5a21f071200 ← NVME\VEN_15B7&DEV_5006&REV_00
                          fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
                          1836f81c-3a3b-52b2-bb89-e5dc480ca9ec ← WDC PC SN730 SDBQNTY-256G-1001
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```
I am using fwupd 1.7.9.

PhilippWendler, 2023-11-16:
Ok, it seems I just needed to wait longer for mirrors to update or whatever. Update to 0.1.86 was offered now and applied successfully on this M70s. Thanks!

closed 2023-11-16 by PhilippWendler