fwupd/firmware-lenovo #374: ThinkPad L380 Yoga recognized as X280

Prroet, 2023-09-30:
Hi,

I tried to update the firmware of my Lenovo ThinkPad L380 to the latest version. The update with the command `fwupdmgr update`  reboots and it tries to do some update. During this reboot it shows fwupd, then a version number and then something like `reset`.
After this the Lenovo Logo appears for a moment. Then it reboots again. 
Once rebooted I executed the following command: `fwupdmgr get-devices` which returns in its first line: `LENOVO 20M7001BGE`.
A quick search revealed that this represents my notebook model Lenovo ThinkPad L380 Yoga. The output also states, that the last firmware update was unsuccessful. 
Anotheer command I tried was `fwupdmgr get-updates` which shows a few components, that can be updated, but for some reason it shows ThinkPad X280 instead of ThinkPad L380 Yoga, which I would expect. 

I also tried downloading and installing the following firmware: https://fwupd.org/downloads/92ebdf5b5ed5eafa71ccb57887d6f7c1e9a879d11dd93d8ed40d1da3007dbf16-Lenovo-ThinkPad-L380-SystemFirmware-0.1.33.cab 
The command I tried to update was `fwupdmgr install 92ebdf5b5ed5eafa71ccb57887d6f7c1e9a879d11dd93d8ed40d1da3007dbf16-Lenovo-ThinkPad-L380-SystemFirmware-0.1.33.cab`
But that resulted in: `No supported devices found`

I have the feeling that the Lenovo ThinkPad L380 and the Lenovo ThinkPad L380 Yoga are not compatible to another, but I am not sure. I still have to investigate on that. 

Does someone know how I can fix that I get updates for X280?

Prroet, 2023-09-30:
I am opening this issue here, because the closed issue: https://github.com/fwupd/firmware-lenovo/issues/2 appears to be very similar to my issue. I hope that was ok.

mrhpearson, 2023-11-01:
Doing an inbox cleanup and I'd missed this one - can you post the output from 'fwupdmgr get-devices' so I can check what the GUIDs used are please?

przemys1aw, 2023-11-10:
Hello there,

same here for L380 non-yoga version.

As I understand executing `fwupdmgr get-devices` is going to send the info to the project owners without the need to paste the dump here.

OpenSauce04, 2024-01-10:
I've also run into this issue on my L380 (non-yoga), and it also considers my laptop to be a ThinkPad X280

OpenSauce04, 2024-01-10:
@mrhpearson Here is the `fwupdmgr get-devices` output for my L380:
[devices.txt](https://github.com/fwupd/firmware-lenovo/files/13894571/devices.txt)

OpenSauce04, 2024-05-01:
@mrhpearson Any updates on this?

mrhpearson, 2024-05-01:
Thanks for the nudge.

I'm not seeing the GUID from your system matching what is on LVFS.

From your get-devices for system firmware:
│ │   GUIDs:              74edf5c5-58bb-46aa-aa7d-d655dd77284c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
And checking the x280 FW on LVFS
    <firmware type="flashed">6d265843-ce23-4ba3-a86e-64271ad0aa3f</firmware>

Can you confirm which FW you get offered please?
Mark

kmauleon, 2024-05-01:
hi I think ODM owner used X280 MEFW GUID again for L380 Yoga... this platform is out of my scope actually but I will check with my ODM contact if they know the owner.  will get back to you later...

kmauleon, 2024-05-02:
hi just an update... owner already replied and is checking this... will update again once I received info from owner...

kmauleon, 2024-05-06:
hi...as per ODM owner LVFS BIOS fix ETA is on 7/12... thank you very much

kmauleon, 2024-07-08:
hi all... BIOS 1.37 has been released to LVFS.  please check >>> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR0RET.firmware

kmauleon, 2024-08-21:
closing ticket.... please let us know if need to reopen... thank you very much

closed 2024-08-21 by kmauleon