fwupd/firmware-lenovo #375: Thinkpad T14 Gen1

davvore33, 2023-10-12:
I've updated just today the Intel Management Engine to the version 225.70.2228 and the touchpad doesn't work anymore even tho it doesn't seem to be related, I'm using archlinux well updated

```fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.6
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.2
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
runtime   org.freedesktop.Passim        0.1.3
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.7
runtime   org.kernel                    6.5.7-arch1-1
runtime   org.freedesktop.fwupd         1.9.6

```

and about the device in particular
```fwupdmgr get-devices 0d7baa94-7f7f-4896-b1d3-8a6c594ff635
Selected device: Intel Management Engine
LENOVO 20S00012IX
│
└─Intel Management Engine:
      Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
      Summary:            UEFI ESRT device
      Current version:    225.70.2228
      Minimum Version:    0.0.1
      Vendor:             Lenovo (DMI:LENOVO)
      Update State:       Success
      Problems:           • Device requires AC power to be connected
      GUID:               0d7baa94-7f7f-4896-b1d3-8a6c594ff635
      Device Flags:       • Internal device
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
```

is there anyway to have it rollback for now?

hughsie, 2023-10-12:
"fwupdmgr downgrade" will downgrade the version if the old version exists on the LVFS and if the ME allows a downgrade. Touchpad stopped working is a new one on me. Does sound still work too?

davvore33, 2023-10-13:
> "fwupdmgr downgrade" will downgrade the version if the old version exists on the LVFS and if the ME allows a downgrade. Touchpad stopped working is a new one on me. Does sound still work too?

unluckily the ME won't allow me to do so, someone suggested to reset the bios using the button beneath the cover, didn't work as well, I've tried "fwupdmgr downgrade" both with the ME and the System firmware, so far none of those are working sadly

kmauleon, 2023-10-16:
have you tried fwupdmgr reinstall ?  and choose MEFW... will ping ODM dev regarding this issue

kmauleon, 2023-10-17:
@davvore33 what was your previous MEFW version?  ODM dev will try to recreate your issue

kmauleon, 2023-10-17:
@davvore33 have you updated System Firmware as well prior to having the issue?

kmauleon, 2023-10-18:
@davvore33 anything else you updated aside from MEFW?  ODM dev did not encounter the issue

davvore33, 2023-10-19:
> have you tried fwupdmgr reinstall ? and choose MEFW... will ping ODM dev regarding this issue

hi, yes I've tried that without much result, sadly I can't help you with any other info, I've sent it to repair ^^"

kmauleon, 2023-10-19:
okay ^_^  let's just have this ticket opened until you have your laptop fixed... thank you very much!

kmauleon, 2023-12-11:
@davvore33 just wonderin' if your laptop was fixed already?