fwupd/firmware-lenovo #378: Cannot update TPM on T16 Gen 1

kmauleon, 2023-11-07:
I'm having the exact same issue here:
```
TPM:
  Device ID:            f098ca39a715eccf184c23361aede08540bf345b
  Previous version:     0.15.21
  Update State:         Failed
  Update Error:         failed to update to 0: error-unsuccessful
  Last modified:        2023-07-10 10:52
  GUID:                 8d5056e5-7a0a-4bcd-bf92-7d16212b72aa
  Device Flags:         • Internal device
                        • Updatable
                        • System requires external power source
                        • Needs a reboot after installation
                        • Device is usable for the duration of the update
```
We also have custom Secure Boot keys enrolled. 
Unfortunately, I can't use the default keys, that are shipped with my `Lenovo ThinkPad T16 Gen1`, which would result in Arch not booting anymore.

Installed Software:
```
~ pacman -Q fwupd linux 
fwupd 1.9.2-2
linux 6.4.1.arch2-1
```

_Originally posted by @VeldoraTheDragon in https://github.com/fwupd/firmware-lenovo/issues/353#issuecomment-1628819046_

kmauleon, 2023-11-09:
@VeldoraTheDragon statement from ODM dev as below...
the problems are confirmed in the process of updating T16 GEN1 INTEL IFX TPM. I will confirm the problems in SW environment (BIOS/EC/ME/Linux OS/ fwupdmgr version) and Perform an update operation after the machine I get. Check whether the TPM FW itself is a problem~ 

will feedback again once I get updates from them... thank you very much

kmauleon, 2023-11-13:
@VeldoraTheDragon ODM dev did not replicate the issue
[image]
[image]

for confirmation may we have similar information from your side... thank you
BIOS
EC
ME
Linux OS
fwupdmgr --version

kmauleon, 2023-11-22:
@VeldoraTheDragon please see instructions from ODM owner
[image]

kmauleon, 2023-12-18:
@VeldoraTheDragon the owner can't replicate the issue.  is it possible to update your BIOS/EC first again.  then after successful update please update TPM firmware to latest as well.  BTW, please make sure AC adapter is attached and battery is at least 25% for all updates.   Thank you very much.

kmauleon, 2024-01-29:
hi @VeldoraTheDragon another instruction from owner... can you please check... thank you very much!

Please let user check whether Bitlocker Driver Encryption is OFF. If yes, please turn it off before updating TPM FW.
The reason for this is that overseas users have opened the Bitlocker Driver Encryption before, resulting in PD firmware has not been updated

Bitlocker Driver Encryption:  Control PanelSystem and Security Bitlocker Driver Encryption

kmauleon, 2024-01-31:
@VeldoraTheDragon unfortunately TPM owner cannot replicate the issue using Ubuntu system... not sure how can they assist further

kmauleon, 2024-05-06:
closing as can not replicate with ubuntu. no update since jan'24

bukajsytlos, 2025-04-07:
@kmauleon did the owner have custom keys enrolled?

weimdall, 2026-10-04:
I ran into the exact same issue.
-> The update fails with custom Secure Boot keys, but **succeeds once Secure Boot is turned off**

kmauleon, 2026-10-08:
@weimdall can we get your system details please... and SS of BIOS Setup Menu... thank you very much

weimdall, 2026-10-08:
Here is a screenshot of the BIOS Setup Menu.
System Details:
- Lenovo ThinkPad P14s Gen 3
- Arch Linux - Kernel 7.2.8
- fwupd version 2.1.8
- fwupdmgr status post successful upgrade (disabling Secure Boot was needed)
```
└─TPM:
  │   Device ID:          891d5fcfe7676c2c5ab824cfa90c7be26b21c1e0
  │   Previous version:   0.15.21
  │   Update State:       Success
  │   Last modified:      2026-10-04 18:21:19
  │   GUID:               8d5056e5-7a0a-4bcd-bf92-7d16212b72aa
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen3/T16 Gen1 TPM Update:
        New version:      0.15.22
        Remote ID:        lvfs
        Release ID:       38455
        Summary:          Lenovo ThinkPad T14 Gen3/T16 Gen1 Infenion TPM Firmware
        License:          Proprietary
        Size:             1.3 MB
        Created:          2022-05-09 00:00:00
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N3BCZIFX152216832
        Vendor:           Lenovo
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        This stable release fixes the following issues:
        
        • This version tpm firmware fixed security issue
        Checksum:         aca3da190ce920d6891fc92415c397e106199229b3963358aa6250a6ecffc99d
```

[image]

kmauleon, 2026-10-08:
hi @weimdall as per ODM CO, they are working on a new release.  ETA around Nov.  Will follow up again by then.  Thank you very much!