fwupd/firmware-lenovo #379: Thinkpad P14s Gen2: Intel Management Engine: failed to update to 0: AC power required

DanBenHa, 2023-11-12:
I'm trying to update the firmware for Intel Management Engine, but it fails, apparently due to missing AC power adapter:
```
20VXCTO1WW
│
├─Battery:
│ │   Device ID:          f95c9218acd12697af946874bfe4239587209232
│ │   Previous version:   0
│ │   Update State:       Success
│ │   GUID:               869eab4c-7872-466f-a759-817c49a34782
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      1.1.10246
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   0.1.55
│ │   Update State:       Success
│ │   GUID:               973a10fa-79c0-453e-b78c-27b55cc0e6eb
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2:
│       New version:      0.1.57
│       Remote ID:        lvfs
│       Release ID:       44065
│       Summary:          Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2 System Firmware
│       License:          Proprietary
│       Size:             18,3 MB
│       Created:          2023-06-20
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Some of the platform secrets may be invalidated when updating this firmware. Please ensure you have the volume recovery key before continuing.
│       
│       This stable release fixes the following issues:
│       
│       • Updated includes a security fix.
│       
│       Some new functionality has also been added:
│       
│       • Updated the Diagnostics module to version 04.30.000.
│     
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Previous version:   10.01.3273255
│ │   Update State:       Success
│ │   Last modified:      2023-01-06 16:12
│ │   GUID:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │   Device Flags:       • Updatable
│ │                       • Is in bootloader mode
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─  New version:      10.01.3478575
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   0
  │   Update State:       Transient failure
  │   Update Error:       [31m[1mfailed to update to 0: AC power required[0m
  │   Last modified:      2023-11-11 18:03
  │   GUID:               5d19c74d-6a93-489a-a5ed-65cb7f1e13bf
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2:
        New version:      240.45.2411
        Remote ID:        lvfs
        Release ID:       53374
        Summary:          Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2 Consumer ME Firmware
        License:          Proprietary
        Size:             3,8 MB
        Created:          2023-08-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N34RN19W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Update includes a security fix.
      
```
AC power was plugged in the entire time. I don't see any power-related message when the actual update is tried. `IgnorePower=true` is set in `/etc/fwupd/daemon.conf`, but it's probably irrelevant for the actual update.

hughsie, 2023-11-12:
What does `upower --dump` say?

DanBenHa, 2023-11-12:
> What does `upower --dump` say?

```
Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              So 12 Nov 2023 16:24:44 CET (50 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               LGC
  model:                5B10W51826
  serial:               5822
  power supply:         yes
  updated:              So 12 Nov 2023 16:24:45 CET (49 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              42,34 Wh
    energy-empty:        0 Wh
    energy-full:         42,44 Wh
    energy-full-design:  50,5 Wh
    energy-rate:         2,942 W
    voltage:             12,662 V
    charge-cycles:       191
    percentage:          99%
    capacity:            84,0396%
    technology:          lithium-polymer
    icon-name:          'battery-full-charging-symbolic'

Device: /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o001
  native-path:          ucsi-source-psy-USBC000:001
  power supply:         yes
  updated:              So 12 Nov 2023 14:10:23 CET (8111 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              no
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o002
  native-path:          ucsi-source-psy-USBC000:002
  power supply:         yes
  updated:              So 12 Nov 2023 16:24:45 CET (49 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/mouse_dev_FF_42_48_BB_0F_27
  native-path:          /org/bluez/hci0/dev_FF_42_48_BB_0F_27
  model:                MX Vertical
  serial:               FF:42:48:BB:0F:27
  power supply:         no
  updated:              So 12 Nov 2023 14:48:57 CET (5797 seconds ago)
  has history:          yes
  has statistics:       no
  mouse
    present:             yes
    rechargeable:        no
    state:               unknown
    warning-level:       none
    percentage:          100%
    icon-name:          'battery-missing-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              So 12 Nov 2023 16:24:43 CET (51 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               pending-charge
    warning-level:       none
    energy:              42,34 Wh
    energy-full:         42,44 Wh
    energy-rate:         2,942 W
    charge-cycles:       N/A
    percentage:          99%
    icon-name:          'battery-full-charging-symbolic'

Daemon:
  daemon-version:  0.99.17
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  yes
  critical-action: PowerOff

```

hughsie, 2023-11-13:
> /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o001

Is that a UPS? Maybe that's confusing the capsule updater?

DanBenHa, 2023-11-13:
> > /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o001
> 
> Is that a UPS? Maybe that's confusing the capsule updater?

No, I don't own a UPS. Maybe this dump from a fresh restart is more informative.
```
Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              Mo 13 Nov 2023 18:39:29 CET (107 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              no
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               LGC
  model:                5B10W51826
  serial:               5822
  power supply:         yes
  updated:              Mo 13 Nov 2023 18:39:29 CET (107 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       none
    energy:              41,75 Wh
    energy-empty:        0 Wh
    energy-full:         42,44 Wh
    energy-full-design:  50,5 Wh
    energy-rate:         20,177 W
    voltage:             11,89 V
    charge-cycles:       192
    time to empty:       2,1 hours
    percentage:          98%
    capacity:            84,0396%
    technology:          lithium-polymer
    icon-name:          'battery-full-symbolic'
  History (charge):
    1699897169	0,000	unknown
  History (rate):
    1699897169	0,000	unknown

Device: /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o001
  native-path:          ucsi-source-psy-USBC000:001
  power supply:         yes
  updated:              Mo 13 Nov 2023 18:39:29 CET (107 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              no
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o002
  native-path:          ucsi-source-psy-USBC000:002
  power supply:         yes
  updated:              Mo 13 Nov 2023 18:39:29 CET (107 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              no
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Mo 13 Nov 2023 18:39:29 CET (107 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               discharging
    warning-level:       none
    energy:              41,75 Wh
    energy-full:         42,44 Wh
    energy-rate:         20,177 W
    charge-cycles:       N/A
    time to empty:       2,1 hours
    percentage:          98%
    icon-name:          'battery-full-symbolic'

Daemon:
  daemon-version:  0.99.17
  on-battery:      yes
  lid-is-closed:   no
  lid-is-present:  yes
  critical-action: PowerOff
```
The `line_power_ucsi_source_psy_USBC` is still there, albeit at a different USB port.

hughsie, 2023-11-14:
@mrhpearson do you happen to know why "AC power required" would be returned from the capsule updater, even though the AC state is connected and the battery is fully charged?

mrhpearson, 2023-11-14:
Afraid I don't know. @kmauleon - is this something you can look at?

Is the power supply the one that comes supplied with the laptop?

DanBenHa, 2023-11-14:
> Is the power supply the one that comes supplied with the laptop?

Yes.

Maybe this is of help: When I plug in the charger both of these power supplies get updated to `online: yes`

DanBenHa, 2023-11-14:
Maybe the issue is not about the power but about getting the current version.

I think `fwupd` is trying to install this: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34RN.firmware, which says it's firmware version `15.0.45.2411`. I verified in the BIOS menu that this is the current version that the computer is running.

mrhpearson, 2023-11-14:
Oh - yeah, we've had a bunch of issues with ME FW and versions. Katherene should know what the status of those fixes are - but for now ignore the update as you're already running the right version.

No idea why it's reporting it as a power issue though. @hughsie  - how does fwupdmgr get that status message from the FW?

kmauleon, 2023-11-15:
reply from ODM owner...
I noticed ME version in customer's system already be 15.0.45.2411(from BIOS setup). I think ME was upgraded successfully already. But the problem is why was Linux reading the wrong power status and not reading the ME version in system. But I don't know what is the mechanism for Linux to read version information. 

@hughsie how does fwupd get the power status? is this from ESRT table as well?

kmauleon, 2023-12-11:
@hughsie how does fwupd get the power status? is this from ESRT table as well? please advise... thank you

hughsie, 2023-12-11:
@kmauleon no, the capsule is probably talking to the EC directly.

kmauleon, 2023-12-12:
> Maybe the issue is not about the power but about getting the current version.
>
> I think `fwupd` is trying to install this: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34RN.firmware, which says [...]

@DanBenHa can you please try to download the cab file and reinstall manually... please execute as below command... thank you very much

fwupdmgr --allow-older --allow-reinstall install <cab file>
restart the system
wait for firmware installation to complete

kmauleon, 2024-04-05:
hi @plodocus any update from your side?

plodocus, 2024-04-06:
@kmauleon Sorry, I lost track of this. Thanks for the ping.

For some reason the Intel Management Engine doesn't show up in `fwupdmgr get-devices`:

```
20VXCTO1WW
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.44
│     Minimum Version:    0.1.44
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               2a5d161c-0ba2-4136-94b3-c027f0a7e5fc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    61.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              d6d1a39a-3fbc-5ad1-ab40-336ec0776a38
│                         f8d483f9-9284-51b6-9363-7e6d59c40a97
│     Device Flags:       • Updatable
│   
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3478575
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   GUIDs:              09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                       8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:        73e606488fec47b3e3f9288094f66fded0051446
│       Summary:          Fingerprint reader config
│       Current version:  0030
│       Minimum Version:  0030
│       Vendor:           Synaptics (USB:0x06CB)
│       GUIDs:            7c5a1e62-38fa-5859-9337-09dbac6377e4
│                         8a12a1dd-5e54-5cb8-9607-c3a4e3e9a7e4
│       Device Flags:     • Updatable
│                         • Supported on remote server
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─SKHynix HFS001TDE9X081N:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    41730C20
│     Vendor:             SK hynix (NVME:0x1C5C)
│     GUIDs:              34f9d6fa-e1d6-5d7d-909e-0406644298c5
│                         34e06406-f620-5b9d-a9cd-7f3a4e648633
│                         4406b8dc-20c5-5f3a-bc9c-8afdcee48684
│                         0ee57466-ca9d-5f10-a005-ec6bacf6fb82
│                         867711b5-67ef-56e4-8085-0aa777d4de72
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
[... 84 lines cut ...]
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               9dec5ab8-3484-42ea-883a-1b12ace95d37
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
      Summary:            UEFI ESRT device
      Current version:    24856
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               4b3d2033-ae0c-4f25-bc3f-96f0ed347569
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
```
I have no idea what happened in the meantime.

kmauleon, 2024-04-07:
hi @plodocus can you please try to run the following

fwupdmgr refresh --force
fwupdmgr update     (please update one firmware at a time if there are available updates for BIOS, ECFW, or MEFW)
restart system

kmauleon, 2024-10-11:
closing as no update.... let me know if need to reopen... thank you very much...

closed 2024-10-11 by kmauleon