fwupd/firmware-lenovo #381: Latest system firmware for T14 Gen3 AMD (R23ET71W) appears to brick device in certain configuration(s)

aleatoricLilac, 2023-11-19:
Note: the actual update took place under Windows so I'm not exactly sure if this would be the correct place for the issue, but seeing as the version was recently pushed to LVFS as well I believe it should be.

Device type: ThinkPad T14 Gen 3 AMD (21CFA000CD)

A couple days ago I attempted to update the firmware on my laptop to 1.47/1.31 and the process completed successfully, but after it automatically rebooted the laptop appeared to be stuck in a reboot loop without reaching POST. All LEDs on the keyboard would light up for a second, the fan would spin up, then the Caps Lock would remain on until the laptop automatically reboots again. I contacted local support and had the device repaired (motherboard replacement, presumably) and just received the repaired laptop today.

Assuming that the issue was a one-off incident, I updated the firmware again and it did boot without issue for a couple times. Then I proceeded to reset the FW settings to default, and after a successful reboot went on to change the settings back to the pre-repair state: voila, bricked again after saving & restarting.

I suspect this might be related to either the Secure Boot settings (the device came back with Secure Boot disabled; it booted correctly after resetting settings to defaults and the failure came after resetting the keys), disabling the Computrace agent, or configuration differences between Chinese and non-Chinese versions of the device (as Chinese versions don't include a discrete TPM and relies on the fTPM only?)

aleatoricLilac, 2023-11-19:
Changes I made pre-bricking that I can remember, with the BIOS simulator as reference:

Config
* Network:
> Lenovo Cloud Services: disabled
> UEFI IPv4/IPv6 Network Stack: disabled

* Keyboard/Mouse:
> Trackpad: disabled
> F1-F12 as Primary Function: enabled

Security
* Security Chip:
> Clear Security Chip: cleared

* Virtualization:
> Enhanced Windows Biometric Security: enabled

* Absolute Persistence Module
> Current Settings: permanently disabled

* Secure Boot:
> Clear All Secure Boot Keys
> Restore Factory Keys
> Allow Microsoft 3rd Party UEFI CA: enabled

aleatoricLilac, 2023-11-19:
Sorry, misclicked

masema67, 2023-11-21:
almost same issue here on a ThinkPad P14s AMD G3 21J6S05000​
I tried to update the bios to 1.47  stuck in a reboot loop without reaching POST, Lenovo changed the mainboard, my P14s was booting and I tried to updated the bios again and it failed, with the same error as before.

georgiou, 2023-11-22:
Same issue for my ThinkPad T14 Gen 3 AMD updated the firmware from linux (fedora 39) and it now fails to post in exactly the same way

hughsie, 2023-11-22:
@mrhpearson do you want me to pull this from the LVFS until we've worked out what's going on? It looks like the WU deliverable is broken too.

aleatoricLilac, 2023-11-22:
Something else I've noticed is that after another motherboard replacement, the option to permanently remove the Absolute anti-theft module is not present in either the original firmware that came with the board (1.29/1.21) or the latest previous version (1.40/1.30). In addition, if my memory is correct then I don't recall the original motherboard of this laptop presenting said option either.

Speculation here, but since for the second attempt the system booted once just fine before I permanently disabled the persistence module (among a slew of other changes unfortunately) and the original board presumably didn't even include it, could this be related to the issue?

shilin-da, 2023-11-22:
If only I saw this issue two hours ago 😕 
Neither reset nor battery removal helps. Will contact support in the morning.

mrhpearson, 2023-11-22:
> almost same issue here on a ThinkPad P14s AMD G3 21J6S05000​

mrhpearson, 2023-11-22:
> @mrhpearson do you want me to pull this from the LVFS until we've worked out what's going on? It looks like the WU deliverable is broken too.

Hi @hughsie - yes please. 

Japan is off on holiday until Monday so I'll escalate this as a critical issue but anything to avoid bricked systems :(

Sorry to everyone who hit this.

mrhpearson, 2023-11-22:
Internal ticket is LO-2760
(and apologies for the ticket close/open - not sure how that happened...some strange finger trouble maybe?)

hughsie, 2023-11-22:
@mrhpearson I've moved https://fwupd.org/lvfs/firmware/59193/ back to embargo -- it'll take 2h to disappear.

mrhpearson, 2023-11-29:
As a note, FW team have identified the issue and are working on a fix for the next release version.
Did those impacted have Absolute persistence set to permanently disabled in the BIOS?

Hope those impacted were able to recover their systems :(

shilin-da, 2023-11-29:
Thanks for an update here!

> Did those impacted have Absolute persistence set to permanently disabled in the BIOS?

Computrace one? Yes, it was permanently disabled in mine.

> Hope those impacted were able to recover their systems :(

Awaiting for service center response, hope they will be able to reflash the firmware :confounded:

mrhpearson, 2023-11-29:
Thanks for the confirm. If support give you any grief at least you can point at this being a flawed FW update release (you're welcome to give my name as an internal contact if they have any questions)
Hopefully they get you fixed up quickly.

georgiou, 2023-11-30:
Absolute persistence was set to permanently disabled in the BIOS on my machine as well.
My motherboard was replaced (under warranty) and everything is working again.

mrhpearson, 2024-05-16:
Probably should have closed this when 71W was pulled - but as an update there is updated FW available now (76W)

closed 2024-05-16 by mrhpearson