fwupd/firmware-lenovo #383: 1.50 firmware is missing for T480

JNKL, 2023-11-23:
Lenovo Vantage on Windows is already distributing it.

https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480-type-20l5-20l6/downloads/ds502355-bios-update-utility-bootable-cd-for-windows-10-64-bit-linux-thinkpad-t480

mrhpearson, 2023-11-29:
It always takes a week or two before the LVFS release is done. Unfortunately the way the FW team process works is they complete the main release, and then start the LVFS testing/release.
I checked and 1.50 is available in the testing channel if you don't want to wait. It should move to stable soon.
Mark

ralphm, 2023-12-12:
Is there a way to expedite this? This versions of the firmware for both flavors (20L5, 20L6 and 20L7, 20L8) fix a [critical vulnerability (LogoFAIL)](https://support.lenovo.com/us/en/product_security/ps500590-bios-image-parsing-function-vulnerabilities-logofail) and were released on 2023-11-20. We're now three weeks further.

mrhpearson, 2023-12-12:
Escalated - internal ticket is LO-2789

mrhpearson, 2023-12-15:
FW released - let me know if any issues

closed 2023-12-15 by mrhpearson