fwupd/firmware-lenovo #384: Lenovo P14S: ACPI error on Ubuntu 23.10

kred-jon, 2023-11-24:
I have this error:

[    3.035831] ACPI Error: Needed [Integer/String/Buffer], found [Package] 000000002d9253bd (20230331/exresop-469)
[    3.035864] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [ToHexString] (20230331/dswexec-433)

[    3.035888] No Local Variables are initialized for Method [ADBG]

[    3.035891] Initialized Arguments for Method [ADBG]:  (1 arguments defined for method invocation)
[    3.035892]   Arg0:   000000002d9253bd <Obj>           Package 000000002d9253bd

[    3.035901] ACPI Error: Aborting method \ADBG due to previous error (AE_AML_OPERAND_TYPE) (20230331/psparse-529)
[    3.035920] ACPI Error: Aborting method \_SB.HIDD._DSM due to previous error (AE_AML_OPERAND_TYPE) (20230331/psparse-529)
[    3.035942] ACPI: \_SB_.HIDD: failed to evaluate _DSM b356ecee-4244-8f40-a792-4edd4d758054 (0x3003)

Somebody can help me explain or fix it? 

Cannot mirroring screen using HDMI port anymore, but in settings the eternal monitor is detected. 

Thank you!

hughsie, 2023-11-24:
Is this the result of a firmware upgrade? if so, what was the old version (that worked) and what is the new version (that doesn't work)? Thanks.

kred-jon, 2023-11-27:
Not sure with the version. But here the history version:

fwupdmgr get-history
LENOVO 20S4CTO1WW
│
├─Thunderbolt host controller:
│ │   Device ID:          34daa8029459c45596fbccc13f5354639ad7d945
│ │   Previous version:   22.00
│ │   Update State:       Success
│ │   GUID:               d6591a43-ce08-57d7-8f00-a4f0cb625e01
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │ 
│ └─ThinkPad T14/P15s Thunderbolt Controller Update:
│       New version:      22.0
│       Remote ID:        lvfs
│       Release ID:       10899
│       Summary:          Lenovo ThinkPad T14/P15s Thunderbolt Firmware
│       License:          Proprietary
│       Size:             262.1 kB
│       Created:          2021-11-29
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Trusted metadata
│       Description:      
│       This stable release fixes the following issue:
│       
│       • change NVM version from 21 to 22.
│       Checksum:         8970160deb2dd6a092855bc89b86659ff0aae2a7246b672ad20436d8df0bb250
│     
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Previous version:   225.67.2046
│ │   Update State:       Success
│ │   Last modified:      2023-08-30 01:45
│ │   GUID:               0d7baa94-7f7f-4896-b1d3-8a6c594ff635
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 1/ P14s Gen 1/ T15 Gen 1/ P15s Gen 1/ T14 Gen 1 Healthcare Edition Consumer ME Update:
│       New version:      225.70.2228
│       Remote ID:        lvfs
│       Release ID:       50930
│       Summary:          Lenovo ThinkPad T14 Gen 1/ P14s Gen 1/ T15 Gen 1/ P15s Gen 1/ T14 Gen 1 Healthcare Edition Consumer ME Firmware
│       License:          Proprietary
│       Size:             3.1 MB
│       Created:          2023-07-26
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N2XRN14W
│       Vendor:           Lenovo
│       Release Flags:    • Trusted metadata
│       Description:      
│       • 1 Intel Platform Update
│       
│       Version 14.1.70.2228 (LVFS: 225.70.2228)
│       
│       Problem Fixes
│       
│       • Intel CSME IPU 2023.3:
│       • Mitigated the following security vulnerabilities under issues.
│       Issues:           CVE-2022-29871
│                         CVE-2022-36392
│       Checksum:         14a535c1e0413dea62317a4395fc597caa62ee7dc36c620e7753e79dda9b0327
│     
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   0.1.28
  │   Update State:       Success
  │   Last modified:      2023-10-04 00:51
  │   GUID:               9d255c4b-2d88-4861-860d-7ee52ade9463
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen 1/ P14s Gen 1 / T15 Gen 1 / P15s Gen 1 / T14 Gen 1 Healthcare Edition System Update:
        New version:      0.1.29
        Remote ID:        lvfs
        Release ID:       58655
        Summary:          Lenovo ThinkPad T14  Gen 1/ P14s Gen 1 / T15 Gen 1 / P15s Gen 1 / T14 Gen 1 Healthcare Edition System Firmware
        License:          Proprietary
        Size:             25.0 MB
        Created:          2021-03-29
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
        Description:      
        Lenovo ThinkPad T14 Gen 1 System Firmware Version 1.29
        
        This stable release fixes the following issues:
        
        • Update includes a security fix.
        • Updated the CPU microcode.
        Issues:           CVE-2022-38076
                          CVE-2022-36351
                          CVE-2022-40964
                          CVE-2022-27635
                          CVE-2022-36765
                          CVE-2022-36764
                          CVE-2022-36763
                          CVE-2023-28005
                          CVE-2022-43505
                          CVE-2022-38083
                          CVE-2022-40982
                          CVE-2022-4304
                          CVE-2022-27879
                          CVE-2022-44611
                          CVE-2022-37343
                          CVE-2022-0778
                          CVE-2021-38575
        Checksum:         06aef6a9e29c85d382e658eb49dcc23c2f0a6f909843307735cc3c212eabd525

kred-jon, 2023-11-27:
HDMI-2 connected 1920x1080+0+0 (0x99a) normal (normal left inverted right x axis y axis) 476mm x 268mm
	Identifier: 0x45
	Timestamp:  3755831
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:     DP-2
	CRTC:       1
	CRTCs:      0 1 2
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	_MUTTER_PRESENTATION_OUTPUT: 0 
	EDID: 
		00ffffffffffff0009d1e97845540000
		2e1f010380301b782eb065a656539d28
		0c5054a56b80d1c081c081008180a9c0
		b30001010101023a801871382d40582c
		4500dc0c1100001e000000ff0047424d
		30323035363031390a20000000fd0032
		4c1e5311000a202020202020000000fc
		0042656e51204757323238330a200152
		020322f14f901f051404130312071615
		01061102230907078301000065030c00
		1000023a801871382d40582c4500dc0c
		1100001e011d8018711c1620582c2500
		dc0c1100009e011d007251d01e206e28
		5500dc0c1100001e8c0ad08a20e02d10
		103e9600dc0c11000018000000000000
		00000000000000000000000000000081
	HDCP Content Type: HDCP Type0 
		supported: HDCP Type0, HDCP Type1
	Content Protection: Undesired 
		supported: Undesired, Desired, Enabled
	max bpc: 12 
		range: (8, 12)
	content type: No Data 
		supported: No Data, Graphics, Photo, Cinema, Game
	Colorspace: Default 
		supported: Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGB_Theater
	aspect ratio: Automatic 
		supported: Automatic, 4:3, 16:9
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	audio: auto 
		supported: force-dvi, off, auto, on
	link-status: Good 
		supported: Good, Bad
	CTM: 0 1 0 0 0 0 0 0 0 1 0 0 0 0 0 0 
		0 1 
	CONNECTOR_ID: 125 
		supported: 125
	non-desktop: 0 
		range: (0, 1)
  1920x1080 (0x99a) 148.500MHz +HSync +VSync *current +preferred
        h: width  1920 start 2008 end 2052 total 2200 skew    0 clock  67.50KHz
        v: height 1080 start 1084 end 1089 total 1125           clock  60.00Hz