fwupd/firmware-lenovo #385: Thinkpad T14 G3 - Updating does not work: "Capsules not found"

dlime, 2023-11-30:
I'm trying to do this upgrade:
```
System Firmware (0.1.41 → 0.1.46)
```

Firmware installs correctly but fails once rebooted with:
```
fwupd-efi version 1.4
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System
```

```
$ sudo fwupdmgr get-devices --show-all-devices
21CF004PGE
│
├─Battery:
│     Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
│     Summary:            UEFI ESRT device
│     Current version:    1.3.7
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               893039b2-553d-4e41-b03d-895a25aca9e5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
│     Summary:            UEFI ESRT device
│     Current version:    0.1.31
│     Minimum Version:    0.1.31
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               6be552b4-eda3-46a7-ab1b-d8ba1ff451bd
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated RGB Camera:
│     Device ID:          91c9a317b30acf7c0cd967f2ce6925d518ef19a7
│     Current version:    10.24
│     Vendor:             8SSC21D67422V1SR26725G6 (USB:0x30C9)
│     Serial Number:      01.00.00
│     GUIDs:              e34f2caa-1e04-54b6-8d2e-fc1af8a8d989 ← USB\VID_30C9&PID_0050&REV_1024
│                         1ef3e35c-cfc6-5028-8eb1-8026af65fab8 ← USB\VID_30C9&PID_0050
│     Device Flags:       • Updatable
│   
├─Ryzen 7 PRO 6850U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│                         52f8f9af-1ca9-5352-bef4-ceb232c888a5 ← CPUID\PRO_0&FAM_19&MOD_44
│                         e94372a3-3ffb-5d1c-a579-c415b7313e52 ← CPUID\PRO_0&FAM_19&MOD_44&STP_1
│     Device Flags:       • Internal device
│   
├─SSD 980 PRO 1TB:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    5B2QGXA7
│     Vendor:             Samsung (NVME:0x144D)
[... 123 lines cut ...]
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device
      Current version:    1
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               75d78aef-5e7f-0afe-b68a-aef4c52ccddd
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
________________________________________________

Devices that were not updated correctly:

 • System Firmware (0.1.41 → 0.1.46)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
```

hughsie, 2023-11-30:
@dlime what distro and fwupd version please?

dlime, 2023-12-01:
> @dlime what distro and fwupd version please?

Distro: `Ubuntu 22.04.3 LTS x86_64`
Fwupd: `fwupd version: 1.7.9`

hughsie, 2023-12-01:
@dlime that's super old -- can you try with something newer please? e.g. at least fwupd-efi 1.3 and fwupd 1.9.x -- thanks.

dlime, 2023-12-02:
> @dlime that's super old -- can you try with something newer please? e.g. at least fwupd-efi 1.3 and fwupd 1.9.x -- thanks.

Thanks for pointing to the right direction. As the latest version provided by https://packages.ubuntu.com/jammy/fwupd is too old I upgraded `fwupd` via [Snap](https://github.com/fwupd/fwupd/wiki/fwupd-snap):
```
sudo apt purge fwupd
sudo snap install fwupd
```

Then refreshed & started updates:
```
sudo fwupdmgr get-devices
sudo fwupdmgr refresh
sudo fwupdmgr get-updates
sudo fwupdmgr update
```

I then encountered this issue:
```
/boot/efi does not have sufficient space, required 90.1MB, got 31.8MB
```
Which I solved by:
- cleaning up leftover `/boot/efi/EFI/ubuntu/fw`
- temporary moved Windows boot directory to `/boot/`
- re-run `sudo fwupdmgr update`
- restored EFI Windows directory

(I'll leave the adventure of extending `/boot/efi/` partition size for another day)

Once I reboot unfortunately I still got the same error:
```
fwupd-efi version 1.4
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System
```

Don't know if this is expected: I noticed that `fwupd` is now located in `/boot/efi/EFI/` and not in `/boot/efi/EFI/ubuntu/` as it was previously.

dlime, 2023-12-12:
I've managed to increase the size of my `/boot/efi` partition (originally only `100MB`) and successfully rerun:
```
fwupdmgr get-devices
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr update
```
All updates got installed with no error 👍

closed 2023-12-12 by dlime