fwupd/firmware-lenovo #386: Upgrade Intel Management Engine fails on T14P Gen1

newbie997, 2023-12-01:
**Describe the bug**
I execute `fwupdmgr update` on T14P Gen1, and the screen displays as follows after reboot:
```
fwupd-efi version 1.4
Error 366: Failed to verify signature of OEM or RoT key manifests. For example: Error on update from Production to Pre-Production.
Reset System
``` 
After restarting the system I executed `fwupdmgr get-devices`
```
LENOVO 21J70002CD
│
├─ELAN0681:00 04F3:3209:
│     Device ID:          1fdcb2a8b1674688af852e4127cb75b35906bb2f
│     Summary:            Touchpad
│     Current version:    0x0004
│     Bootloader Version: 0x0003
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              38897f13-3a41-5b8a-9cc8-9370a2b41291 ← HIDRAW\VEN_04F3&DEV_3209
│                         4b69bffd-d456-54ae-baee-e11b517fa3bd ← HIDRAW\VEN_04F3&DEV_3209&MOD_0037
│                         3313b7ba-c61b-5183-a9c4-3411cc892998 ← ELANTP\ICTYPE_14&MOD_0037
│                         a97baf40-d178-5cf6-81e6-874542d698e2 ← ELANTP\ICTYPE_14&MOD_0037&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│
├─Unknown Device:
│     Device ID:          c15c2d5f642ca8dc5b20a79e7fd0672546063f20
│     Current version:    0.0.0.0
│     Vendor:             USB:0x1D5C
│     Install Duration:   15 seconds
│     GUIDs:              9252ea97-2539-599e-b1f7-d33c7f253339 ← USB\VID_1D5C&PID_7112
│                         4a86f85b-78c1-577b-abed-292e9b5f8ba7 ← USB\VID_1D5C&PID_7112&CID_00
│     Device Flags:       • Updatable
│
├─13th Gen Intel Core™ i7-13700H:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000411c
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         c3f97ae6-7da2-54cc-ad26-3225f9517203 ← CPUID\PRO_0&FAM_06&MOD_BA
│                         557a4f44-6981-5889-8676-684b9727b697 ← CPUID\PRO_0&FAM_06&MOD_BA&STP_2
│     Device Flags:       • Internal device
│
├─Integrated RGB Camera:
│     Device ID:          aa4b7bf130da693bbf1a8a785ad830190788233b
│     Current version:    0.7
│     Vendor:             8SSC21K12297V1SR33129AR (USB:0x30C9)
│     Serial Number:      01.00.00
│     GUID:               c8615157-0fc4-5175-ae3e-4de204933573 ← USB\VID_30C9&PID_00A8
│     Device Flags:       • Updatable
│
├─Intel Management Engine:
│     Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
│     Summary:            UEFI ESRT device
│     Current version:    1.25.2091
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               d91340eb-9ddf-42a0-8134-c77b957802d3
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
......and more .....
|
|
Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
Successfully uploaded 1 report
```
According to the information in the last link, I confirmed my device `Disable Boot Order Lock` and `turning off secure boot`, but it still doesn't work.

**Steps to Reproduce**
```
# fwupdmgr update
Devices with no available firmware updates:
 • ELAN0681:00 04F3:3209
 • (null)
 • Integrated RGB Camera
 • Internal SPI Controller
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • WD PC SN810 SDCQNRY-512G-1201
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 1.25.2091 to 1.27.2176?                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update includes a security fix.                                            ║
║                                                                              ║
║ 21J70002CD must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • UEFI dbx
An update requires a reboot to complete. Restart now? [y|N]:
```
**fwupd version information**
```
# fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.9
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.4
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.Passim        0.1.4
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.6.3-arch1-1
runtime   org.freedesktop.fwupd         1.9.9
```

newbie997, 2023-12-02:
My laptop is probably not a pre-production machine. I bought it from an official Chinese online store. The link for this model is as follows: https://newthink.lenovo.com.cn/searchResults.html?q=VGhpbmtQYWQgVDE0cCBHZW4x  ,I can't seem to find any information related to t14p or t14p gen1 on lvfs. How can I troubleshoot this problem, thank you very much

kmauleon, 2023-12-19:
@newbie997 can you please provide a clear screenshot for Setup Menu Main Page including BIOS version and CPU version --ODM owner's request
thank you very much...

newbie997, 2023-12-20:
@kmauleon 
[image]
Is this the information?

kmauleon, 2023-12-20:
thank you @newbie997 will forward to owner

kmauleon, 2023-12-21:
@newbie997 owner suggest to update to latest BIOS via ISO thru this link >> https://think.lenovo.com.cn/support/driver/driverdetail.aspx?DEditid=120103

please check if the issue still happens after upgrade.  thank you very much

jczhang02, 2024-01-09:
@newbie997 请问你有这样的问题吗: CPU满载情况下频率限制到800MHZ左右.

newbie997, 2024-01-10:
@jczhang02 是的，就是这个问题，我暂时用thermald调整了，全核满载3.7左右，但瞬间有核心满载的话也会先降下去，然后在慢慢提升上来。

kmauleon, 2024-04-05:
@newbie997 have you tried this one https://github.com/fwupd/firmware-lenovo/issues/386#issuecomment-1865428036

jczhang02, 2024-04-05:
> @jczhang02 是的，就是这个问题，我暂时用thermald调整了，全核满载3.7左右，但瞬间有核心满载的话也会先降下去，然后在慢慢提升上来。

您可以给我个 联系方式吗？ 我们可以交流下这个笔记本的一些使用问题等等 @newbie997

newbie997, 2024-04-08:
@kmauleon I've tried the iso file in the link and some older versions. They all report the same error.
[image]

jczhang02, 2024-04-08:
@newbie997 how u update the iso bios update?

jczhang02, 2024-04-08:
update from linux maybe not supported and will cause error.
IBM thinkpad make it but lenovo does not!

newbie997, 2024-04-08:
I checked how to update the bios. I first used an open source tool called Ventoy (Ventoy is an open source tool to create a bootable USB drive for ISO/WIM/IMG/VHD(x)/EFI files) and then I used USB Image Writer directly to write the bios to the USB stick and it worked. Thanks for your help. @jczhang02 @hughsie

closed 2024-04-08 by newbie997

jczhang02, 2024-04-08:
那你在使用新的bios后，锁频问题和性能是否有改进呢？

Yours, Chengrui

newbie997, 2024-04-09:
这个我不太确定，我和你发邮件了，私聊吧 @jczhang02