fwupd/firmware-lenovo #387: Thinkpad P1 Gen 5: fwupdmgr causes "Failed to Recover ME Firmware"

wtogami, 2023-12-04:
Last night fwupdmgr attempted to upgrade the system firmware, ME firmware, and IIRC something else I don't remember exactly what.

ME updater said:
* "48% Reading ME Firmware...Please do not power off!"
* "56% Recovering ME Firmware...Please do not power off!"
* "60% Failed to recover ME Firmware..."

Subsequently
* "BIOS Self Healing backup progressing" was stuck overnight at 0%.
* Forced power off, stuck at 0% again forever.

Pulled nvme drive
* "BIOS Self Healing backup progressing" was able to succeed.

Now
* "Failed to recover ME Firmware" displays for a few seconds every boot then I'm able to boot into Windows or Linux grub.

### Bugs

1. It shouldn't be offering that bogus ME firmware update.
2. The combination of the bogus ME firmware update and a legitimate system firmware update can nearly brick the device.
3. I'm stuck with this ME failure during every boot. How do I fix this?

```
root@p1gen5:~# rpm -q fwupd
fwupd-1.9.9-1.fc39.x86_64
root@p1gen5:~# fwupd
fwupdmgr   fwupdtool  
root@p1gen5:~# fwupdmgr update
Devices with no available firmware updates: 
 • Fingerprint Sensor
 • Integrated Camera
 • Intel Management Engine
 • SSD 980 PRO 2TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 1.27.2176 to 257.27.2176?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Corporate ME Firmware, for Hybrid    ║
║ Vpro                                                                         ║
║                                                                              ║
║ 21DDS6SA00 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: n
 • System Firmware
 • UEFI dbx
root@p1gen5:~# fwupdmgr get-history
LENOVO 21DDS6SA00
│
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Previous version:   10.01.3333223
│ │   Update State:       Success
│ │   Last modified:      2023-10-13 11:32
│ │   GUID:               659f7e45-8d45-528d-b3c7-0695eed055f6
│ │   Device Flags:       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Updatable
[... 131 lines cut ...]
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen 5/X1 Extreme Gen 5 System Update:
        New version:      0.1.21
        Remote ID:        lvfs
        Release ID:       73337
        Summary:          Lenovo ThinkPad P1 Gen 5/X1 Extreme Gen 5 System Firmware
        License:          Proprietary
        Size:             22.2 MB
        Created:          2023-11-07
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
        Description:      
        Changes in this release version 1.21
        
        Important updates - Enhancement to address security vulnerability.
        
        New functions or enhancements - Nothing
        
        Problem fixes - Nothing
        Checksum:         64fafb8d3864b5e6ba9dcdfcca145d76cb2a269ebb36d135e6c5bc3dc9c3c67e
      
```

wtogami, 2023-12-04:
I'm able to boot without ME but things are broken like audio.

https://thesofproject.github.io/latest/getting_started/intel_debug/suggestions.html#make-sure-the-me-is-enabled
Huh why does this require ME ...

Krotki, 2023-12-06:
I deal with exactly same problem on P1 gen 5. This is very annoying.

On top of above description my machine freezes from time to time and I have to reboot it which brings me back the ME firmware load. It's a vicious cycle and the machine is barely usable.

I tried to install cab directly (found on another issue): ```fwupdmgr --allow-older --allow-reinstall install 68b320dd67db5717062ddbc6d1ad7e42c7c4e82d90fab41c73d87685fe6446fd-Lenovo-LenovoThinkPadP1Gen5X1ExtremeGen5-CorporateMEFirmware-257.27.2176.cab``` but did not help.

Is there a way to fix it ?

kmauleon, 2023-12-07:
@wtogami and all,  latest version of MEFW is 1.27.2176.... Please ignore update notice to 257.27.2176
this is a known issue and ODM owner still discussing the fix with BIOS dev.  thank you very much

kmauleon, 2023-12-07:
most likely the fix will be in BIOS but I am not entirely sure... I am still waiting for an update from ODM MEFW owner... will update here once I received any information from them...

wtogami, 2023-12-09:
#360 sounds like the same issue. This has been going on since August where you've been trying to get answers from the LDM MEFW owner? It doesn't sound like this will happen. Sigh need to go with warranty service.

wtogami, 2023-12-11:
I just filed a Lenovo support ticket for depot repair. Should I cancel the support request because I will be able to instead flash the firmware to fix it myself? @mrhpearson knows my support ticket details.

wtogami, 2023-12-11:
Can I test the fix? I mean I have nothing to lose given the machine would go to the depot either way.

kmauleon, 2023-12-12:
@wtogami can you please send me a note >> kmauleon at lenovo dot com
so I can share the cab file with you... thank you very much

kmauleon, 2023-12-12:
@wtogami sent the file and instruction... please check.  thank you very much

kmauleon, 2024-03-20:
ODM says they will target to set the firmware to stable by next week....

kmauleon, 2024-04-02:
ODM just set the firmware to stable ... closing this ticket for now.  please reopen if needed..  thank you very much

closed 2024-04-02 by kmauleon