fwupd/firmware-lenovo #388: Thinkpad P1 Gen5: trackpoint unresponsive after resume

padamstx, 2023-12-04:
I have a P1 Gen5 running Fedora 38 with all the latest linux and firmware updates installed.

After installing some firmware updates a few weeks ago, my trackpoint mouse pointer is completely unresponsive after a resume.  The touchpad and touchscreen functionality seem to be fine after the resume, it's just the trackpoint that is dead.

When this problem occurs, only a complete power off/on reboot will get the trackpoint back.  I tried an experiment where I did a "restart" 4 times and that didn't help.  When I then did the power recycle, the trackpoint came back.

The firmware updates from a few weeks ago included the install of the System Firmware from version 0.1.19 to 0.1.21.  I've experimented a bit by downgrading this back to 0.1.19 but that didn't help either.  I subsequently re-installed the latest (0.1.21).

I'm completely baffled by this one.  Any ideas on what to do next to diagnose would be appreciated!

Thanks in advance!

hughsie, 2023-12-04:
Random question: Does upgrading to Fedora 39 change the behaviour?

padamstx, 2023-12-04:
I don't know as I haven't tried that.  I will ask other colleagues if they have enountered the same problem on F39.

padamstx, 2023-12-04:
I've received word from a colleage that he is running F39 and also sees this problem.

padamstx, 2023-12-04:
To add a little bit more data...
On F38, I'm seeing this problem with at least the three most recent kernels (6.6.2-101.fc38, 6.5.12-200.fc38, and 6.5.11-200.fc38)

erezh16, 2023-12-04:
P1 Gen5 / F39: confirming the described behavior.

Toby-Haynes, 2023-12-05:
P1 Gen5 / F39 - also confirmed. All firmware is at the latest available levels.

erezh16, 2023-12-06:
P1 Gen5 / F39. Don't know if this helps. I compared the output of `sudo libinput list-devices` when the trackpoint is working (after off/on) and when it is not (after resume), without any external mouse attached:
1. In *both* cases there are two input devices detected: "ELAN0686:00 04F3:320D Mouse" and "ELAN0686:00 04F3:320D Touchpad".
2. When the trackpoint works (after off/on), there is an *additional* input device detected: "TPPS/2 Elan TrackPoint"  
See both outputs attached.
[trackpoint_off.txt](https://github.com/fwupd/firmware-lenovo/files/13577721/trackpoint_off.txt)
[trackpoint_on.txt](https://github.com/fwupd/firmware-lenovo/files/13577723/trackpoint_on.txt)

Toby-Haynes, 2023-12-06:
Similar and possibly related bug reports - all Lenovo models with the Elan trackpoint.

https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1791427
https://bugzilla.redhat.com/show_bug.cgi?id=1560723
https://forums.lenovo.com/t5/ThinkPad-Z-series-Laptops/Z13-trackpoint-stops-working-after-lid-is-closed/m-p/5166498?page=5

This last link on forums.lenovo.com concludes with a message that Lenovo is working with Elan, and that Elan has a reproduction of the issue. That post is dated 20 November 2023

padamstx, 2023-12-07:
In the most recent comment in @Toby-Haynes's first link above (https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1791427/comments/225) the commenter mentioned that for his Thinkpad model (X1 Extreme Gen 5), downgrading the BIOS from 1.20 to 1.19 fixed the issue.
Just for grins, I downgraded my BIOS from 1.21 to 1.18 (I had already experimented with downgrading to 1.19 with no luck).  Unfortunately, this didn't help :(

erezh16, 2023-12-07:
> In the most recent comment in @Toby-Haynes's first link above (https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1791427/comments/225) the commenter mentioned that for his Thinkpad model (X1 [...]

Maybe if we find a BIOS version for P1 Gen5 with a publishing date close to that of v1.19 of the X1 Gen5, this should be the right one..? Long shot, I know.

padamstx, 2023-12-07:
> Maybe if we find a BIOS version for P1 Gen5 with a publishing date close to that of v1.19 of the X1 Gen5, this should be the right one..? Long shot, I know.

I was using fwupdmgr to do the downgrade and the versions it offered only went back to 1.18, but perhaps there's a way to obtain even older versions.
I don't want to mess around too much for fear of bricking this machine :joy:
My understanding is that Lenovo has reached out to Elan and they've been able to reproduce so hopefully a solution will be found soon.

erezh16, 2023-12-12:
> Perhaps one other thing worth mentioning... I ran `sudo libinput list-devices` when the trackpoint was working fine, and then also after resuming when the trackpoint was dead, and the only difference is that this entry is not listed when the trackpoint is dead:
> 
> ```
> Device:           TPPS/2 Elan TrackPoint
> Kernel:           /dev/input/event4
> Group:            9
> Seat:             seat0, default
> Capabilities:     pointer 
> Tap-to-click:     n/a
> Tap-and-drag:     n/a
> Tap drag lock:    n/a
> Left-handed:      disabled
> Nat.scrolling:    disabled
> Middle emulation: disabled
> Calibration:      n/a
> Scroll methods:   *button
> Click methods:    none
> Disable-w-typing: n/a
> Disable-w-trackpointing: n/a
> Accel profiles:   flat *adaptive custom
> Rotation:         n/a
> ```
> 
> Not surprising but apparently the system doesn't even recognize the trackpoint after a resume.

Same as my findings above. All I added is that there are ELAN input devices (mouse, touchpad) that are always present (trackpoint working or not) but at least the "mouse" seems to be passive - does not generate events from using the trackpoint, so it cannot serve as a workaround.

erezh16, 2023-12-12:
One more bit of information: when I install firmware using `fwupd`, it seems to trigger a "hard" reset that makes the trackpoint work again, without physically turning the machine off and on again.

TheForgottenKing, 2024-01-14:
Replying to state that I think this is purely a BIOS issue and not an issue with fwupd or Linux at all.

I run Windows 11 on my P1 Gen 5 and I use S3 sleep since S0i3 was causing issues. BIOS version 1.21 caused the same issue (complete loss of trackpoint and buttons until full power off and on) and this was resolved upon downgrading to version 1.19.

frasertajima, 2024-01-15:
There is a BIOS setting for sleep where you can select between the more modern "Windows and Linux" sleep mode and the older Linux S3 mode. 

- Selecting the older S3 mode ensures the ThinkPad, P16 in my case, stays sleeping but it borks the red trackpoint when resuming from suspend (the trackpad is fine). 

- Selecting the more modern "Windows and Linux" suspend mode keeps the trackpoint running after suspend but the ThinkPad keeps waking up during suspend and just becomes annoying after a while.

So I am waiting for a BIOS fix on this issue and avoiding the trackpoint until then as a working suspend is more important than a working trackpoint.

padamstx, 2024-01-16:
>     * Selecting the older S3 mode ensures the ThinkPad, P16 in my case, stays sleeping but it borks the red trackpoint when resuming from suspend (the trackpad is fine).
> 
> ...
> 
> So I am waiting for a BIOS fix on this issue and avoiding the trackpoint until then as a working suspend is more important than a working trackpoint.

I have seen the same exact behavior.  I'm running Fedora 38 on my P1 Gen 5, however.   I have switched to the "Windows & Linux" sleep mode, but I make sure that I unplug everything except for the power cord.  This seems to prevent inadvertent resumes from happening, although now and then the display won't come back from sleep and I need to do a force-power-off :(

I am also hoping for a BIOS fix for this :)

padamstx, 2024-01-16:
@kmauleon is this issue on anyone's radar screen yet?

kmauleon, 2024-01-18:
@ChiWei-Chen should this be trackpoint or BIOS owner?

ChiWei-Chen, 2024-01-19:
@padamstx Thanks, this is a known issue our FW team working with ELAN, will keep you posted. (Tracking by LO-2039 internally)

padamstx, 2024-02-13:
Any news on this one?
This problem is especially problemmatic because it occurs when the laptop's S3 sleep mode is enabled in the BIOS.
This problem does not occur when the newer "Windows & Linux" sleep mode is enabled, but lately my laptop rarely recovers properly from a resume when this mode is enabled.  The primary display will either remain blank and never come back after the resume (power recycle needed to recover), or the primary display comes back after result but the keyboard/trackpointer/touchpad is completely unresponsive (power recycle needed).
So, at this point suspend/resume is effectively non-functional for P1 Gen 5 laptops running Fedora linux.
Any priority that can be associated with this would be appreciated.  This problem has existed for several months now.  When can we expect a fix for this?

ChiWei-Chen, 2024-02-14:
@padamstx, the "Windows & Linux" sleep mode (S0ix) issue may related to graphics, could you let us know NVIDIA driver version? Could you try the latest version if it hasn't been updated?
Thanks!

padamstx, 2024-02-15:
Hi @ChiWei-Chen, prior to yesterday I was using the latest version of the nvidia driver on Fedora 38.  With that setup, I was experiencing the problems that I mentioned in the previous comment above (i.e. resume was not working very well at all when the "Windows & Linux" sleep mode was configured in the BIOS).

Yesterday I upgraded my P1 Gen 5 to Fedora 39.  Initially, I had the latest version of the nvidia driver installed and observed basically the same behavior as before with F38 (S3 mode -> unresponsive trackpointer after resume; W&L mode -> resume essentially didn't work well at all).
I then completely removed the nvidia driver and now with W&S sleep mode, resume appears to be working more reliably and the trackpointer is also working correctly after a resume.  With S3 sleep mode enabled, I still see the same behavior as on F38 (display finally comes back in ~75 seconds after resume was triggered, and the trackpointer is unresponsive).  So the trackpointer issue would appear not to be related to the nvidia display drivers.

goekce, 2024-02-15:
I have a similar issue. On my X13 Yoga 1st gen (20SY) sometimes the trackpoint does not work after waking up from suspend. My workaround: suspending again and waking it up.

castagnag, 2024-02-15:
I have the same issue on my X11 Carbon Gen5 but cannot find where to change
the sleep mode in BIOS. I have the latest Lenovo System Firmware Version
1.61

padamstx, 2024-02-15:
At least in the BIOS for my P1 Gen5 laptop, it's in the Config -> Power settings

castagnag, 2024-02-16:
This is where I find it in my X11 Carbon Gen9, but it is not on the X11
Carbon Gen5 (yep, I'm addicted to X11 Carbon), though there is a Config >
Powersetting menu. I guess it is too old to hope for a firmware update on
it. Thanks all the same.

joihn, 2024-02-24:
thinkpad t14s gen1, bios 1.28
I have disable auto suspend on lid close.
Sometimes when trackpoint is not working, closing/reopening the lid a few time solve it. 
I think Elan need to fix their firware.

DAVID9SE, 2024-03-24:
I have this issue on my gen6 x1 (Fedora WS 39). Suspending the laptop again always helps, but is quite annoying.

erezh16, 2024-03-24:
I have a strange feeling that Lenovo/Elan are not really picking up on this. 
Anyway, my solution: switch to hibernation. Stop depending on this broken BIOS suspend. Much shorter suspend/resume cycles, and everything is working.

mrhpearson, 2024-04-05:
We are still looking at this, there's been some progress behind the scenes but it's taking a while to root cause and determine the right fix - and whether it needed to be done in the touchpad FW or in the BIOS.

I had missed that this was only seen when S3 was being used. I've reproduced the issue on my system and highlighted this to the FW team. I can't reproduce the issue with S0ix. Having a repro should make this easier (we'd been looking at a similar issue on the Z-series, but there it happens with S0ix)

joihn, 2024-04-09:
here is my workaround fix script, which I manually launch when my my trackpoint become unresponsive and usually it brings it back
```
#!/bin/bash
set -ex
sudo sh -c 'echo -n none > /sys/devices/platform/i8042/serio1/drvctl'
sleep 1
sudo sh -c 'echo -n reconnect > /sys/devices/platform/i8042/serio1/drvctl'
sleep 1
```

dmig, 2024-04-14:
Thanks for the workaround @joihn!

Will integrate it using this https://wiki.archlinux.org/title/Power_management/Suspend_and_hibernate#Combined_sleep/resume_unit

I have ThinkPad L13 Gen2.

dmig, 2024-04-16:
Here is my workaround service:
```

joihn, 2024-04-29:
Sometime, my previously posted command does not work,
In suh case, I've had some success with this
```
sudo sh -c 'echo -n "elantech"> /sys/bus/serio/devices/serio1/protocol'
```

M0rtale, 2024-05-20:
P1 Gen 5 on kubuntu 24.04 also experiencing this issue on bios version 1.22

mrhpearson, 2024-05-23:
P1G5 fix will be in EC version 25W EC (likely 1.12). Expected to land mid-June - it's under testing currently.

dmig, 2024-05-23:
@mrhpearson great news!

Anything about L13? Hit by this bug badly, drifting also occurs.

NostalgiaJohn, 2024-05-25:
Hi there, I encounter the same issue on my Thinkpad P1 Gen5 running Windows 11 after waking up from S3 sleep mode. I've tried to roll back to previous BIOS version (from 1.22 to 1.21) and TrackPoint driver (from 31.21.28.2 to 31.21.22.1), but still not fixed. This problem emerged after I found my system couldn't boot and reinstalled Windows 11 last week.

Hope my comment may help:) If further info. is needed or you get a solution, please contact me!

mrhpearson, 2024-05-28:
@dmig - checking on the L13 G2, I had missed that
@NostalgiaJohn - see above for P1G5. You're waiting for an EC update to 1.12. Issue is not in the BIOS or trackpoint driver.

NostalgiaJohn, 2024-05-29:
> @dmig - checking on the L13 G2, I had missed that
> @NostalgiaJohn - see above for P1G5. You're waiting for an EC update to 1.12. Issue is not in the BIOS or trackpoint driver.

I get it! I'll wait for the update.

By the way, where can I locate the future EC(Embedded Controller) update? I've found [a page on ThinkWiki](https://www.thinkwiki.org/wiki/Embedded_Controller_Firmware), it refers 'BIOS/EC version'. If so, why rolling back BIOS version doesn't help? If not, where future EC update is  on the page [Thinkpad P1 Gen5 Manual Driver Update](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p1-gen-5-type-21dc-21dd/downloads/driver-list), Motherboard Devices?

mrhpearson, 2024-05-29:
Sorry - afraid I only deal with Linux so anything I say should probably be ignored for Windows. 
I would expect Lenovo Vantage application to offer you the update when it is available (or check the Lenovo support pages)
Mark

NostalgiaJohn, 2024-05-29:
That's fine, thanks again for your comments. I've checked out my EC version and found 1.11. 
Hope you a nice day!

mrhpearson, 2024-05-31:
L13G2 FW has been released (1.32). It's in the stable channel so should be offered as an update on your system.

dmig, 2024-05-31:
Installed a system firmware update 1.32 (from 1.29) on my L13 g2: bug is still there - just experienced random cursor jumps and clicks.
Is EC even updatable on this model?

mrhpearson, 2024-06-03:
Crud - I should have caught that. Given the fix was in the EC for the P1 I assume the same is needed here.
You're right - no EC updates seem to have been uploaded. I've raised LO-3112 specifically for the L13 G2 to request that gets updated ASAP.

erezh16, 2024-06-03:
I got a P1 Gen 5. Haven't noticed any pending firmware updates recently. Should there be one already?

mrhpearson, 2024-06-03:
P1G5 - not yet. Target is mid June (the FW release process with testing/review/etc is quite lengthy)

ingemaradahl, 2024-06-12:
I'm now running 0.1.22, but the trackpoint is still unresponsive coming out from sleep.

NostalgiaJohn, 2024-06-19:
> P1G5 - not yet. Target is mid June (the FW release process with testing/review/etc is quite lengthy)

I've installed the update for almost a week, the trackpoint performs really fine now. Thanks for your group's work! :)

However, I encountered twice stuck-issue this week after come back from S3 sleep. It not a 100% problem but can happen now and then. I haven't figured out where is the source, maybe I shall raise a new issue later...

mrhpearson, 2024-06-21:
For the L13G2 - the EC is contained in the BIOS update so should have been updated as well. 
@dmig - Can you confirm your BIOS and EC version please?

dmig, 2024-06-21:
Here it is:
```
├─System Firmware:
│ │   Device ID:          0dcb5df743a10d80fd5f4dfe23028d5aa2e12513
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    0.1.32
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               78ad74e1-d048-4abd-b566-9e4e7fbd6e7c
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message

System Firmware:
  Device ID:            0dcb5df743a10d80fd5f4dfe23028d5aa2e12513
  Previous version:     0.1.29
  Update State:         Success
  Last modified:        2024-05-31 14:31
  GUID:                 78ad74e1-d048-4abd-b566-9e4e7fbd6e7c
  Device Flags:         • Internal device
                        • Updatable
                        • System requires external power source
                        • Supported on remote server
                        • Needs a reboot after installation
                        • Reported to remote server
                        • Cryptographic hash verification is available
                        • Device is usable for the duration of the update
```

mrhpearson, 2024-06-24:
EC too please :)

dmig, 2024-06-24:
> EC too please :)

Which is ...? Where do I get that version?
EC is not in the devices `fwupdmgr` can see.

ChiWei-Chen, 2024-06-25:
@dmig could you find the Embedded Controller version in BIOS menu?

dmig, 2024-06-25:
@ChiWei-Chen yes, it seems to be the only option:

EC version is R1FHT43W 1.21

ChiWei-Chen, 2024-06-26:
@dmig , thanks!
It looks the EC has been upgraded succesfully.
> BIOS1.32  (R1FET58W)  EC 1.21  (R1FHT43W)

erezh16, 2024-07-01:
[image]
Hi folks. P1 Gen5 with F39. This firmware update (see attached image) popped up on my Gnome Software Update this morning. Is this the long-awaited firmware update from Lenovo for this github issue? Is it ALL OF IT (i.e., no further updates expected for now)?

mrhpearson, 2024-07-03:
Yep - the EC update to 1.12 has the fix in it.

tobias-ahlstrom, 2024-07-04:
Hey! I'm on a T490s with Fedora 40 having similar issues. I've been following this thread for a while, hoping that this could potentially fix my issues as well. Can I expect an update as well?