fwupd/firmware-lenovo #390: Can't install update ECFW T14/P14s Gen 4

tststired, 2023-12-06:
Hello, 

Since  https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3QHT.firmware are now on lvfs 
I thought I'd be able to update them but unfortunately when I run 
`fwupdmgr refresh`  and `get-updates`  nothing seems to happen. 

`fwupdmgr get-updates`
```
Devices with no available firmware updates:
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • WD PC SN740 SDDQNQD-512G-1201
 • Integrated Camera
Devices with the latest available firmware version:
 • UEFI dbx
No updates available
```

`fwupdmgr get-devices  --show-all-devices`
```
LENOVO 21K5CTO1WW
│
├─AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704103
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │
│ ├─GPIO controller:
│ │     Device ID:        f685512aa07369c9e77742acef941d779d31e766
│ │     GUID:             37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│ │
│ ├─Graphics Processing Unit (GPU):
│ │ │   Device ID:        75dbd857f37f0f1aa3d5bf2fea25b876288fb415
│ │ │   Current version:  022.012.000.016.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUIDs:            74ecfd82-e883-519c-a56c-f7cd77dfc66e ← PCI\VEN_1002&DEV_15BF
│ │ │                     e6171682-816b-52d0-b219-40c5cf6a3619 ← PCI\VEN_1002&DEV_15BF&SUBSYS_17AA50DA
│ │ │                     6c23af4f-d6cd-5ffc-a502-0b85e472e7cb ← AMD\113-PHXGENERIC-001
│ │ │   Device Flags:     • Internal device
│ │ │
│ │ └─NE140WUM-NY1:
│ │       Device ID:      aec1a869eb0df71b7cea6b3ac71d39b830faf164
│ │       GUID:           1257dc02-3738-59ed-bcc4-87027cb604c2 ← DRM\VEN_BOE&DEV_0AB7
│ │       Device Flags:   • Internal device
│ │
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       ccdf3a32-442f-5c0b-ad12-2e7709eb8444 ← PCI\VEN_1022&DEV_15C7&SUBSYS_17AA50D9
│ │     Device Flags:     • Internal device
│ │
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 4
│       Current version:  76.25.107
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             3ac3159e-6eef-5f6b-bc8d-67686b238747 ← /sys/devices/platform/AMDI0009:00
│       Device Flags:     • Internal device
│
[... 153 lines cut ...]
│                         • Updatable
│
└─WD PC SN740 SDDQNQD-512G-1201:
      Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Summary:            NVM Express solid state drive
      Current version:    73116101
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      23276Q806892
      Problems:           • Device requires AC power to be connected
      GUIDs:              1524d43d-ed91-5130-8cb6-8b8478508bae ← NVME\VEN_15B7&DEV_5017
                          87cfda90-ce08-52c3-9bb5-0e0718b7e57e ← NVME\VEN_15B7&DEV_5017&SUBSYS_15B75017
                          8635fc59-d0a3-58d7-b16b-a460dd07b59a ← WD PC SN740 SDDQNQD-512G-1201
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
```

Manual installation of the cab file shows no supported devices found 
```
fwupdmgr install 06d854e328983af525e018c8d6c1d5bafb9955cec2464433730bb218dc4af204-N3QET38W.cab
Decompressing…           [            /                          ]
No supported devices found
```

Oposum01, 2024-01-11:
To my understanding the referenced "N3QHT.firmware" is not suitable for P14s Gen 4 AMD (21K5) - N3QHT.firmware is for the Intel variants for the P14s.
The right BIOS Package ID [R2FUJ36W](https://download.lenovo.com/pccbbs/mobiles/r2fuj36w.txt) containing v1.16 (R2FET36W) with v1.14 (R2FHT30W) has been released on 2023/11/24 on Lenovos site, but I can not find any reference to this in the fwupd/LFVS metadata.
Maybe the entry for the P14s Gen 4 AMD (21K5, which I am having too) is still missing? @mrhpearson
The GUIDs for the System Firmware on my P14s Gen4 AMD (21K5) are identical to the ones from @tststired :18cced7e-d108-41c4-9189-b0355c41450d and 230c8b18-8d9b-53ec-838b-6cfc0383493a

Oposum01, 2024-01-19:
I'm pretty sure "N3Q" is for the Intel devices, only.

Nevertheless I'm able to confirm fwupdmgr found the R2FET v1.16 update now (which has been released 4 days ago according to your link. thanks!) on a P14s Gen4 AMD.

tststired, 2024-01-19:
Hmm I'm not sure because I always assumed that EC and BIOS have separate firmware, can @mrhpearson confirm this? 
https://fwupd.org/lvfs/search?value=t14+gen+4 because here when i search, there's 2 BIOS system firmware update paths, and only one EC path.

Oposum01, 2024-01-19:
There is no Embedded Controller Update for the AMD variants available on LVFS for now, as far as I understand.
See the titles in https://fwupd.org/lvfs/search?value=t14+gen+4:
- [ThinkPad T14 Gen 4 / P14s Gen 4 / T16 Gen 2 / P16s Gen 2 System Update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3QET.firmware) -> (**Intel**) "System Update" containing [N3QET39W.cab](https://fwupd.org/downloads/2f28abef4bb7851704ffccc8f685c84a84d79a401a3d9ea5011ff59b2875f66b-N3QET39W.cab) right now
- [ThinkPad T14 Gen 4 / P14s Gen 4 / T16 Gen 2 / P16s Gen 2 Embedded Controller Update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3QHT.firmware) -> (**Intel**) "Embedded Controller Update" containing "N3QHT20W 1.10"
- [ThinkPad T14/T16 Gen 4 P14s/P16s Gen 2 **AMD** System Update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR2FET.firmware) -> (**AMD**) "System Update" containing [Lenovo-Thinkpad-T14-T16-P14s-P16s-AMD-SystemFirmware-1.16.cab](https://fwupd.org/downloads/3b29b209117a1edbf391cd2fb98bb23684956d28955cc90b8d2974fcabf06da8-Lenovo-Thinkpad-T14-T16-P14s-P16s-AMD-SystemFirmware-1.16.cab)

tststired, 2024-01-19:
https://pcsupport.lenovo.com/au/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p14s-gen-4-type-21k5-21k6/21k5/21k5cto1ww/

Ah appears you're right in the details of the 1.16 release it mentions the "This package updates the UEFI BIOS (including system program and Embedded Controller program) stored in the ThinkPad computer to fix problems, add new functions, or expand functions." 

Interesting that intel variants get a split EC update

mrhpearson, 2024-01-22:
Just to clarify on the nomenclature:
The first three chars are the platform - N3Q is the intel variant and R2F is the AMD variant.
The next two chars is the FW type. ET is BIOS, HT is EC (and others have different codes)
After that is the version.

For the G4 AMD EC FW - I just checked and version 1.15 was uploaded 6 days ago but is still under embargo and hasn't been tested yet. Can someone bug me if it's not been released by the end of the week and I'll nag them for an update, but it looks like it's just working it's way thru the process right now.