fwupd/firmware-lenovo #392: Update Lenovo 40B0 refuses to update after failed update attempt

caniko, 2023-12-09:
```
ASUSTeK COMPUTER INC. ROG Zephyrus G14 GA402XY_GA402XY
│
└─ThinkPad Thunderbolt 4 Dock:
  │   Device ID:          ef28b72a23f91bade7e8dae062bd77b9fdb2c629
  │   Previous version:   10.7
  │   Update Error:       Device has been removed
  │   GUID:               8d30b09f-bcc5-5379-bc65-9ccceeece1f4
  │   Device Flags:       • System requires external power source
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Updatable
  │                       • Signed Payload
  │   Device Requests:    • Image (custom)
  │ 
  └─(null) Update:
        New version:      10.16
        Remote ID:        lvfs
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.
```
Refuses to update after the device randomly disconnected during update, `fwupdmgr get-updates` gives no devices to update. What should I do? Would it help if I tried with a real lenovo laptop?

caniko, 2023-12-09:
Fixed, I used my friends Lenovo laptop with W11 to upgrade. Maybe there was a simple, yet elusive, method to do it on Linux. Keeping the issue open because it is still unsolved.

billyvg, 2025-04-29:
Odd for me, I can't even get it to recognize that there are updates available, same version:

```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          6cb4e69ad6ba19631db59c938866bbf97b25b8b6
│ │   Current version:    10.7
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │                       ce170cf8-f4a5-52d6-8a6c-f206071a70ba ← USB\VID_17EF&PID_30B4&CID_40B0&DMCVER_10.7
│ │                       fe1be6a9-3027-5135-8e1b-737015e2b5dd ← USB\VID_17EF&PID_30B4&CID_40B0&REV_0042
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   Device Requests:    • Message
```