fwupd/firmware-lenovo #396: ThinkPad L14 Gen 2 AMD is a few versions behind

valpackett, 2023-12-13:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1KET.firmware is currently at 0.1.27 `R1KET42W`

https://pcsupport.lenovo.com/ar/en/products/laptops-and-netbooks/thinkpad-l-series-laptops/thinkpad-l14-gen-2-type-20x5-20x6/downloads/ds549928-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-l14-gen-2-types-20x5-20x6-thinkpad-l15-gen-2-types-20x7-20x8?category=BIOS is currently at 0.1.30 `R1KET45W` which includes security updates

```
R1KUR16W
 UEFI: 1.30 / ECP: 1.13
-  [Important] Enhancement to address security vulnerability, CVE-2023-5058.

R1KUR15W
 UEFI: 1.29 / ECP: 1.13
-  [Important] Some security code update. 
```

ulf5, 2023-12-17:
T14s gen 4 AMD is also behind:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR2EET.firmware
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-4-type-21f8-21f9/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C

1.09 vs 1.16

mrhpearson, 2023-12-19:
Created internal ticket for L14 G2 AMD - LO-2793

Can you create a separate thread for the T14s G4 AMD please (different platform, so different FW team and I need to track separately). I did create internal ticket  LO-2794. 
As a note I can see 1.17 there but under embargo so I believe they are working on this one and maybe there were issues with LVFS and 1.16? Will find out what is going on.

mrhpearson, 2024-01-09:
BIOS and EC have been released for the L14 G2 AMD so I think this can be closed. Let me know if any concerns

closed 2024-01-09 by mrhpearson