fwupd/firmware-lenovo #397: Thinkpad T14 Gen3 AMD a white line appears randomly at the top of the screen

lantsevrot, 2023-12-15:
Hi all. I bought a new T14 Gen 3 AMD. But a problem immediately appeared: a white line periodically appears at the upper left border of the screen. The problem exists on both Windows 10 and Linux (Debian and Gentoo).

I updated the bios and reset it, but it didn't help. It is not possible to send it in for repair under warranty. What should I do? Just wait for a bios update and hope it gets fixed someday?

This guy has the same problem: https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T-series-Laptops/White-line-appears-randomly-on-the-screen/m-p/5205116

``
gentoo-tp ~ #  dmidecode -s bios-version
R23ET71W (1.47 )
``

hughsie, 2023-12-15:
@mrhpearson I'm pretty sure this isn't the right place to track this -- but maybe you know about the root cause?

mrhpearson, 2023-12-15:
I would try disabling PSR2 - just because that's been causing some headaches: add "amdgpu.dcdebugmask=0x10" to your kernel bootargs.
Otherwise - it sounds like a HW issue and recommend contacting customer support for an RMA.

As an FYI - issues like this (not fw update related) are good for the Lenovo linux forums: http://forums.lenovo.com/linux :)

closed 2023-12-15 by hughsie