fwupd/firmware-lenovo #398: Thinkpad T14s Gen 4 AMD is a few versions behind

ulf5, 2023-12-19:
T14s gen 4 AMD is behind:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR2EET.firmware
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-4-type-21f8-21f9/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C

1.09 vs 1.16

mrhpearson, 2023-12-19:
Thanks :)
Internal ticket LO-2794 updated.

For historical notes (copied):  I can see 1.17 there but under embargo so I believe they are working on this one and maybe there were issues with LVFS and 1.16? Will find out what is going on.

frittentheke, 2024-01-10:
> Thanks :) Internal ticket LO-2794 updated.
> 
> For historical notes (copied): I can see 1.17 there but under embargo so I believe they are working on this one and maybe there were issues with LVFS and 1.16? Will find out what is going on.

Do you have any update yet @mrhpearson ?

mrhpearson, 2024-01-10:
Afraid not yet - waiting for an ETA from the FW team.

ulf5, 2024-01-15:
Now it seems no version at all is available, unless it moved?

alexlud, 2024-01-19:
1.16 was published today. 
I was able to upgrade from 1.11 to 1.16 (fwupd-1.9.11-1.fc39.x86_64).

ChiWei-Chen, 2024-03-19:
Hi @ulf5 ,
Both 1.18 BIOS/ 1.13 ECFW moved to stable now, please let us know if this doesn't work for you.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR2EET.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR2EHT.firmware

closed 2024-03-28 by mrhpearson