fwupd/firmware-lenovo #400: Failed to downgrade Thunderbolt firmware version

Anifyuli, 2023-12-26:
Hello mister. I have some trouble in downgrading proccess of Thunderbolt firmware version from 23.00 to 20.00. If I execute command for downgrade it, I get this error.

```bash
# fwupdmgr install f52c4809a9e30e4a911351f5d9daf063e0a92515 --allow-older --force
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade Thunderbolt host controller from 23.00 to 20.00?                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X280 Thunderbolt Firmware                                    ║
║                                                                              ║
║ Thunderbolt host controller and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Writing…                 [************************************** ] Less than one minute remaining…
could not start thunderbolt device upgrade: could not write to /sys/devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:00.0/0000:04:00.0/domain0/0-0/nvm_authenticate: Invalid argument
```

I wish this problem can solved soon as possible. Thanks for read my issue

srgzx, 2024-04-16:
Did you ever figure this out?

Anifyuli, 2024-04-16:
No, I haven't idea how to figure it

hieu-cove, 2024-09-12:
It's probably the same case as this https://github.com/fwupd/fwupd/issues/6688
Ran into the same think so I think we're out of luck there.