fwupd/firmware-lenovo #406: fwupd.org/downloads/d9fc50ac93a4233be78b872bf0313efcb54801fc32ce3e6296c60642c8769006-r23ul73w.cab -> firmware signature missing or not trusted

davidvoit, 2024-01-09:
Hi,

I'm trying to install the newest firmware update using fwupdmgr update, but I'm getting the signature error stated in the title

Full output of the command with verbose
```
[david@torus ~]$ LANG=C.utf-8 fwupdmgr --verbose update 
(fwupdmgr:6306): GLib-DEBUG: 22:46:26.449: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:6308): GLib-GIO-DEBUG: 22:46:26.453: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6306): GLib-GIO-DEBUG: 22:46:26.457: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.457: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.457: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.457: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.457: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.457: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:6306): GLib-GIO-DEBUG: 22:46:26.457: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:6306): GLib-GIO-DEBUG: 22:46:26.458: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6306): GLib-GIO-DEBUG: 22:46:26.458: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.458: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.458: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.459: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.459: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:6306): dconf-DEBUG: 22:46:26.459: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:6306): Fwupd-DEBUG: 22:46:26.459: Emitting ::status-changed() [idle]
Devices with no available firmware updates: 
 • Integrated Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
(fwupdmgr:6306): FuMain-DEBUG: 22:46:26.470: current version is 0.1.31: 0.1.31=same
 • Prometheus
(fwupdmgr:6306): FuMain-DEBUG: 22:46:26.473: current version is 10.01.3478575: 10.01.3478575=same
 • SKHynix HFS001TDE9X081N
(fwupdmgr:6306): FuMain-DEBUG: 22:46:26.476: current version is 41730C20: 41720C20=older, 41730C20=same
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.46 to 0.1.49?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This BIOS includes the following updates:                                    ║
║                                                                              ║
║ • Notice that BIOS can't be downgraded to older BIOS version after           ║
║ upgrade to r23ul73w(1.49).                                                   ║
║ • Enhancement to address security vulnerability.                             ║
║ • Change to permit fan rotation after fan error happen.                      ║
║ • Fixed non-bootable issue when Absolute Persistence Module Activation is    ║
║ disabled permanently.                                                        ║
║ • Fixed issue that boot time is long when system is attached with            ║
║ DisplayLink Dock.                                                            ║
║ • Fixed platform profile stuck on power-saver under linux modern standby     ║
║ resume.                                                                      ║
║                                                                              ║
║ 21J5002TGE must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Fwupd-INFO: 22:46:27.828: downloading https://fwupd.org/downloads/d9fc50ac93a4233be78b872bf0313efcb54801fc32ce3e6296c60642c8769006-r23ul73w.cab
(fwupdmgr:6306): Fwupd-DEBUG: 22:46:27.828: Emitting ::status-changed() [downloading]
Downloading…             [                                -      ]Fwupd-INFO: 22:46:29.079: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 22:46:29.243: download progress: 0%
Downloading…             [   -                                   ]Fwupd-INFO: 22:46:29.347: download progress: 1%
Downloading…             [                                       ]Fwupd-INFO: 22:46:29.391: download progress: 2%
Downloading…             [                                       ]Fwupd-INFO: 22:46:29.434: download progress: 3%
Downloading…             [*                                      ]Fwupd-INFO: 22:46:29.475: download progress: 4%
Downloading…             [*                                      ]Fwupd-INFO: 22:46:29.513: download progress: 5%
[... 99 lines cut ...]

firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/fwupd.conf ONLY if you are a firmware developer

```

Found the following snippet in an other issue, but I'm just an enduser here, maybe I'm using the tools wrong, bug sharing the output anyway

```
[david@torus bla]$ LANG=C jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/
firmware.bin:
    PASSED sha1: OK
    PASSED sha256: OK
    FAILED pkcs7: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
    FAILED gpg: '48A6D80E4538BAC2' is not a valid signature
    FAILED: Validation failed
firmware.metainfo.xml:
    PASSED sha1: OK
    PASSED sha256: OK
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
Validation failed
```

Regards,
David

hughsie, 2024-01-10:
> maybe I'm using the tools wrong

Can you try using:

    LANG=C jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/ --public-keys /etc/pki/fwupd

There are two sets of keys, one for metadata and one for the payload.

I'm also tracking this in https://github.com/fwupd/fwupd/issues/6646 which might be worth subscribing to.

xsuchy09, 2024-01-10:
```shell
LANG=C jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/ --public-keys /etc/pki/fwupd
firmware.bin:
    PASSED sha1: OK
    PASSED sha256: OK
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
firmware.metainfo.xml:
    PASSED sha1: OK
    PASSED sha256: OK
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
```

hughsie, 2024-01-10:
Lets close this one, and track in https://github.com/fwupd/fwupd/issues/6646

hughsie, 2024-01-10:
Okay, all firmwares have been resigned, and metadata regenerated, and so can you please try fwupdmgr refresh --force and then fwupdmgr update please. Feedback either way appreciated.

davidvoit, 2024-01-10:
I refresh --force and Update now did work, thanks for the fast solution!