fwupd/firmware-lenovo #408: 'authentication signing error' for Intel Thinkpad L14/L15 Gen 3

VictorGob, 2024-01-12:
Thinkpad L14/L15 Gen 3. Unable to update:

- **Lenovo Embedded Controller Firmware** from version 0 to 1.36 
- **Lenovo ThinkPad L14/L15 Gen 3 Consumer ME: R1XMA13W/R1XMA13N Firmware** from from 1.27.2176 to 257.27.2176
- **BIOS and ECFW firmware Update For Reference code N3AET77W** from 0.1.10 to 0.1.23

```bash
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.5.0-14-generic
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```
<details>
<summary> fwupdmgr get-devices --show-all-devices </summary>

```
21C1002SSP
│
├─12th Gen Intel Core™ i7-1255U:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000430
│     Vendor:             Intel
│     GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         ab855c04-4ff6-54af-8a8a-d8193daa0cd8
│                         d49f8285-de61-572f-ae6d-b83795576416
│     Device Flags:       • Internal device
│   
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               4d8d65f2-65ba-5333-b5ec-ab226494fed7
│                         5ec24b2a-3bc3-548b-befa-7f65a3dec395
│                         a67b547a-8ece-5f7d-a489-0d75d0f2eaf6
│                         7fc247b0-a7a8-55a7-b29c-ffb9725436dc
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:          ae1abd099407b1d95698d69b7273f7fed5c6f35c
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Failed
│     Update Error:       failed to update to 0: authentication signing error
│     GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          54dda0fbfd5ac0dee5efc0ec4baefc9190d666ac
│     Current version:    56.20
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUID:               b687616d-e0bc-58d3-9022-8f0065eed4ca
│                         bcd2c020-e315-55aa-b7e1-7d2d7108037e
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
[... 264 lines cut ...]
│                         6df6872d-19f7-5035-b85c-7c760034c77d
│                         1d6b808f-3aac-5c20-8496-8a1433de5b80
│                         918e52f5-17f2-5b89-b030-6eef6ff9ffd9
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB4 host controller:
      Device ID:          b69e952bfa5a911d1b49b7aad938e33e06abcda2
      Summary:            Unmatched performance for high-speed I/O
      Vendor:             INTEL (THUNDERBOLT:0x8087)
      GUID:               d651f364-21d8-5cc3-9790-dd019a8958ef
                          5041b15d-6e1c-5681-86b3-bb1d56993357
                          e72e778e-94f7-5ed2-b560-1c1262ee217c
      Device Flags:       • Internal device
                          • System requires external power source
                          • Signed Payload
```

</details>

OS: Ubuntu 22.04.3 LTS
CPU: 12th Gen Intel(R) Core(TM) i7-1255U

hughsie, 2024-01-13:
@mrhpearson are we tracking this already?

mrhpearson, 2024-01-15:
Thanks Richard.

Two separate issues going on here I think

1 - @kmauleon - I'm assuming you're already working on the ME update issue as that's common across a lot of platforms. Please confirm.

2 - For the EC: The EC version 0 is very wrong - your system cannot have version zero. Can you please go into the BIOS setup screen (F1 during early boot) and confirm what you see there and we'll figure out the next steps based on that.

Just an advance note that I'm travelling this week so replies may be delayed.

kmauleon, 2024-01-16:
for MEFW, ODM owner still working on this... i'll check with them again... thank you

VictorGob, 2024-01-29:
**Lenovo ThinkPad L14/L15 Gen 3 Consumer ME: R1XMA13W/R1XMA13N Firmware** has been updated (or at least no longer appears as updatable). 

Any update on the rest of the issues? Do they need more information from me?

kmauleon, 2024-02-16:
@VictorGob can you please confirm MEFW version via fwupdmgr get-devices shows the correct one >>> 1.27.2176

VictorGob, 2024-02-16:
> ├─Intel Management Engine:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               23192307-d667-4bdf-af1a-6059db171246
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message

mrhpearson, 2024-02-16:
What happens if you try and update only the BIOS?

I checked the FW release notes and I couldn't see anything highlighted as update issues or steps needed. If you're not able to update the BIOS from LVFS we can work thru doing it manually as you are on a quite old version.

Maybe - if the BIOS update to latest doesn't work you can try BIOS 1.11 - https://fwupd.org/downloads/0d328386f2dd531d2189b89ccde0e5b1dddffb4912d219d2bc4d44f728cd48d6-Lenovo-ThinkPad-L14Gen3-SystemFirmware-0.1.11.cab
I think we need to get the BIOS updated first so the EC version is not reported as zero.

Mark

VictorGob, 2024-02-16:
I've tried to install it, but it throws an error:
>$ fwupdmgr install 0d328386f2dd531d2189b89ccde0e5b1dddffb4912d219d2bc4d44f728cd48d6-Lenovo-ThinkPad-L14Gen3-SystemFirmware-0.1.11.cab
invalid handle

hughsie, 2024-02-17:
> invalid handle

@superm1 is that a snap issue?

superm1, 2024-02-17:
That's probably https://github.com/fwupd/fwupd/issues/6730

superm1, 2024-02-17:
Put it in the local automatic repo instead for a workaround.

superm1, 2024-02-17:
> I've tried the "copy to root" workaround
> 
> >$ sudo fwupdmgr install /root/0d328386f2dd531d2189b89ccde0e5b1dddffb4912d219d2bc4d44f728cd48d6-Lenovo-ThinkPad-L14Gen3-SystemFirmware-0.1.11.cab
> Decompressing…           [                -                      ]
> No supported devices found

This is no longer a snap issue afaict. It means the cab isn't matching the device or device not found.

mrhpearson, 2024-02-17:
Hmmm - your GUID doesn't match what is in the capsule metadata.

├─System Firmware:
│ │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│ │   GUID:               562a0522-d1a0-4b88-ad34-3cfda8788412
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a

vs (from the capsule metadata)

  <provides>
    <firmware type="flashed">5dc5d695-c24f-4723-aa17-1b30f79c6d19</firmware>
  </provides>

I'll need to ask the FW team about this I'm afraid (I'm not on the VPN right now - will try and remember to do it later on with my regular work PC - but nag me if I haven't posted a ticket number by Tuesday).
I'm guessing we'll need to update from UEFI shell, but would like to get the FW team to review first.

If there's any history to the updates that might explain why this system is in this strange state that would be useful.
Mark