fwupd/firmware-lenovo #409: Battery not charging on ThinkPad X1 Yoga Gen1 (after installing ubuntu 20.04 updates)

cronzr, 2024-01-15:
Hi Team!
  Could you please help? I just installed updates from update center in ubuntu 20.04 and battery stops to charge.

cronzr, 2024-01-15:
`# fwupdmgr get-devices 
20FRS2T500
│
├─1100 SATA 256GB:
│     Device ID:          e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:            ATA drive
│     Current version:    M0DL022
│     Vendor:             Micron (ATA:0x1344, OUI:00a075)
│     Serial Number:      18111C21BE72
│     GUID:               6aa08236-ffc8-5d25-891e-a0470346f3a1 ← IDE\Micron_1100_SATA_256GB__________________M0DL022
│                         6ac6374d-4066-5042-9a20-048e5159e695 ← IDE\0Micron_1100_SATA_256GB__________________
│                         4bf727a1-d975-5528-9c56-1eac662e9ec8 ← Micron 1100 SATA 256GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.19
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               4c2e5b5c-6467-43af-afeb-7bc72d96b9c3
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.92.4222
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               95f9a8f4-9531-483c-afee-c3f12774ff4b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.52
  │   Minimum Version:    0.1.31
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               81cba30b-c2d4-4e66-bcf3-69d81ba890b0
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI revocation database
        Current version:  83
        Minimum Version:  83
        Vendor:           UEFI:Linux Foundation
        Install Duration: 1 second
        GUID:             14503b3d-73ce-5d06-8137-77c68972a341 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649
                          5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
                          c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
        Device Flags:     • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Only version upgrades are allowed
                          • Signed Payload
`

cronzr, 2024-01-15:
# upower -d
Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              Monday 15 jan 2024 12:26:56 (2270 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               SMP
  model:                00HW029
  serial:               2041
  power supply:         yes
  updated:              Monday 15 jan 2024 13:02:57 (109 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              46,85 Wh
    energy-empty:        0 Wh
    energy-full:         47,12 Wh
    energy-full-design:  52,06 Wh
    energy-rate:         0 W
    voltage:             16,769 V
    percentage:          99%
    capacity:            90,5109%
    technology:          lithium-polymer
    icon-name:          'battery-full-charging-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Monday 15 jan 2024 12:28:57 (2149 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               pending-charge
    warning-level:       none
    energy:              46,85 Wh
    energy-full:         47,12 Wh
    energy-rate:         0 W
    percentage:          99%
    icon-name:          'battery-full-charging-symbolic'

Daemon:
  daemon-version:  0.99.11
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  yes
  critical-action: PowerOff

hughsie, 2024-01-15:
> I just installed updates from update center

Do you remember what update you applied? "fwupdmgr history" might help.

cronzr, 2024-01-15:
Hi @hughsie!
  Thanks a lot for your quick reply. There were a lot of updates - quite a long listing.

# fwupdmgr get-history
No history

hughsie, 2024-01-15:
@mrhpearson any ideas? I don't see a specific battery entry, so I assume it's an EC regression? The update details were not exactly awesome: https://fwupd.org/lvfs/components/3446/description

mrhpearson, 2024-01-16:
Strange there's no history - is that normal?
Ack on the update description; that not good. 

I will try and have a look thru what is on the FW server and see what I can find in the internal notes but in the meantime have created internal ticket  LO-2833 for review by the FW team.
If it's not a battery update then EC is the most likely candidate. Hopefully they have some suggestions.

Mark

cronzr, 2024-01-16:
Hi Mark!
  Thanks a lot for your quick reply. Hope that FW team can help on this.
  Thank you!

cronzr, 2024-01-18:
Richard, Mark, hi!
  Could you please advice if there were any news from FW team?

cronzr, 2024-01-18:
By the way - is it possible to rollback firmware? So, I hope in this case the battery will start to charge again...
Could you please help?

hughsie, 2024-01-18:
> is it possible to rollback firmware

    fwupdmgr downgrade

but it might be prevented by the vendor -- and also relies on the "old" firmware also being on the LVFS.

cronzr, 2024-01-18:
Richard, thanks a lot for the advice! I'll try to do that.

cronzr, 2024-01-18:
I tried to downgrade to prior versions when possible, but no result....

```
# fwupdmgr downgrade
Choose a device:
0.	Cancel
1.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
2.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
3.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
1
No downgrades for Embedded Controller: current version is 0.1.18: 0.1.18=same

# fwupdmgr downgrade
Choose a device:
0.	Cancel
1.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
2.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
3.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
2
Choose a release:
0.	Cancel
1.	184.90.3987
2.	184.80.3746
3.	184.79.3722
0
Request canceled

# fwupdmgr downgrade
Choose a device:
0.	Cancel
1.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
2.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
3.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
3
No downgrades for System Firmware: current version is 0.1.52: 0.1.56=newer, 0.1.55=newer, 0.1.53=newer
```

cronzr, 2024-01-22:
Hi Mark!
  Could you please advice if FW have updates on this ticket?

Thanks in advance.

> Strange there's no history - is that normal? Ack on the update description; that not good.
> I will try and have a look thru what is on the FW server and see what I can find in the internal notes but [...]

cronzr, 2024-01-24:
@hughsie @mrhpearson
Richard, Mark, hi!

Battery not charging and it's not possible to use notebook outside the office.
Could you please help?

mrhpearson, 2024-01-24:
The issue has been flagged to the FW team, but we haven't had any feedback yet I'm afraid

I have to note that the platform went end of support a year ago and the FW on LVFS is over a year old - so it's hard to get comments from anybody.

There are also, seemingly, 15k successful downloads so I'm wondering if you battery just died and the timing is unfortunate (or something in the update stressed it and kicked it over the edge?). It's quite possible that changing FW isn't going to help.

Will send a nudge - but wanted to set expectations.

cronzr, 2024-01-24:
Hi Mark!
  Thanks a lot for ther information and your quick reply.

cronzr, 2024-01-26:
Hi Mark, Richard!
  TodayI tried to downgrade intel management controller and after that arrived kernel updates in ubuntu. I've installed them and now battery is charging. Weird, but seems like it was a bug in kernel, but not sure what was exact reason.
   Thank you very much for you help!

mrhpearson, 2024-01-29:
Thanks for the update!

closed 2024-01-29 by mrhpearson