fwupd/firmware-lenovo #410: X1 Nano Gen2 doesn't seem to expose SPI controller though Gen1 did

arusakov-rh, 2024-01-15:
Using the kernel from Fedora 40:
```
$ ls -ld /sys/class/mtd
drwxr-xr-x. 2 root root 0 Jan 13 20:31 /sys/class/mtd
```
However,
```
$ ls -l /sys/class/mtd
total 0
```
Any chance of getting access to it?

hughsie, 2024-01-15:
@arusakov-rh can you paste the output of lspci please.

arusakov-rh, 2024-01-15:
```
00:00.0 Host bridge: Intel Corporation 12th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P GT2 [Iris Xe Graphics] (rev 0c)
00:04.0 Signal processing controller: Intel Corporation Alder Lake Innovation Platform Framework Processor Participant (rev 02)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 02)
00:07.0 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #0 (rev 02)
00:07.2 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #2 (rev 02)
00:08.0 System peripheral: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator (rev 02)
00:0a.0 Signal processing controller: Intel Corporation Platform Monitoring Technology (rev 01)
00:0d.0 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 USB Controller (rev 02)
00:0d.2 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #0 (rev 02)
00:0d.3 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #1 (rev 02)
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
00:14.3 Network controller: Intel Corporation Alder Lake-P PCH CNVi WiFi (rev 01)
00:15.0 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #0 (rev 01)
00:16.0 Communication controller: Intel Corporation Alder Lake PCH HECI Controller (rev 01)
00:1f.0 ISA bridge: Intel Corporation Alder Lake PCH eSPI Controller (rev 01)
00:1f.3 Multimedia audio controller: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
04:00.0 Non-Volatile memory controller: Shenzhen Unionmemory Information System Ltd. RPETJ512MKP1QDQ PCIe 4.0 NVMe SSD 512GB (DRAM-less) (rev 03)
50:00.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 03)
51:00.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 03)
51:01.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 03)
51:02.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 03)
51:03.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 03)
51:04.0 PCI bridge: Intel Corporation Thunderbolt 4 Bridge [Goshen Ridge 2020] (rev 03)
```

arusakov-rh, 2024-01-15:
Hm, kinda yes:
```
$ sudo dmesg | rg spi
[   46.757890] spi-nor spi0.0: unrecognized JEDEC id bytes: f7 30 30 09 03 00
```

hughsie, 2024-01-15:
Hmm, we've not seen this before. That message is from `drivers/mtd/spi-nor/core.c` -- according to that, it comes from `drivers/mtd/spi-nor/core.h` and that gets populated from files like `drivers/mtd/spi-nor/winbond.c` -- although I don't see anything like that.