fwupd/firmware-lenovo #412: ThinkPad P14s Gen 2a battery at 240%

anayrat, 2024-01-19:
Hello,
Last year I updated laptop firmware and since then, the battery show 240% capacity.
I tried to recalibrate it by discharging to 0% and recharging, but the full capacity is still at 240%:
```
Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               LGC
  model:                5B10W51826
  serial:               2132
  power supply:         yes
  updated:              Fri 19 Jan 2024 03:55:11 PM CET (22 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              65.29 Wh
    energy-empty:        0 Wh
    energy-full:         121.21 Wh
    energy-full-design:  50.5 Wh
    energy-rate:         35.92 W
    voltage:             12.119 V
    charge-cycles:       50
    time to full:        1.6 hours
    percentage:          53%
    capacity:            100%
    technology:          lithium-polymer
    icon-name:          'battery-good-charging-symbolic'
```

The energy-full is two times larger than the real capacity.
I looked at my history:

```
├─Battery:
│ │   Device ID:          f95c9218acd12697af946874bfe4239587209232
│ │   Previous version:   1.0.8523
│ │   Update State:       Success
│ │   GUID:               37132c80-5c8b-4726-a502-6f8c456da9b4
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─(null) Update:
│       New version:      1.1.10246
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
```

I didn't have any issue during the update.

joaquinco, 2024-04-22:
The same happens to me on a t14 gen2. It seems to be related to this battery model.

From linux-hardware.org: <https://linux-hardware.org/?id=bat:lgc-5B10W51826>, there are some probs that show the same output regarding capacity, to link a few:

- <https://linux-hardware.org/?probe=1247f8ba53&log=upower>
- <https://linux-hardware.org/?probe=285b51551c&log=upower>
- <https://linux-hardware.org/?probe=748298f0c8&log=upower>

anayrat, 2024-04-23:
Good catch! Maybe someone from Lenovo could help us @mrhpearson  ?

mrhpearson, 2024-04-24:
I've flagged this to the FW team to get their advice on next steps - internal ticket  LO-2999

Not sure if this will require a battery recalibration or not. We don't have an automated way of doing that in Linux (it's been on my todo for far too long ...); but if they think that's what is needed I think we can manually work through the steps.

Will update when I hear back from the team.

mrhpearson, 2024-05-31:
Sorry for the slow reply on this one - took some back and forth with the FW team (and there was a holiday in Japan).

They're recommending doing a battery recalibration. Best way of doing this is from Windows with the Lenovo Vantage tool. If that's possible (and I know it sucks) let me know.
If you do boot Windows, they wanted a screen shot of the battery details (similar to attached) so they can confirm it's not a Linux data display issue.

[image]

If Windows is not an option, we can try and manually work through the calibration algorithm on Linux - but I've never done it before and it's going to be very manual and comes with no guarantees (I have a flow chart of operations to follow - and think it can be done using the battery controls).
We've had getting a Linux calibration utility implemented for a long time...really need to get that done :(

anayrat, 2024-06-02:
Hello,
Thanks for your reply, unfortunately, I don't have Windows :( 
I already tried to discharge the battery and  charge again, without success.
It seems it is only a reporting issue : 
```
upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               LGC
  model:                5B10W51826
  serial:               2132
  power supply:         yes
  updated:              Sun 02 Jun 2024 05:56:33 PM CEST (5 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               fully-charged
    warning-level:       none
    energy:              121.21 Wh
    energy-empty:        0 Wh
    energy-full:         121.21 Wh
    energy-full-design:  50.5 Wh
    energy-rate:         0 W
    voltage:             12.321 V
    charge-cycles:       53
    percentage:          100%
    capacity:            100%
    technology:          lithium-polymer
    icon-name:          'battery-full-charged-symbolic'

cat /sys/class/power_supply/BAT0/status
Full
cat /sys/class/power_supply/BAT0/capacity
100
cat /sys/class/power_supply/BAT0/capacity_level 
Full
cat /sys/class/power_supply/BAT0/energy_full
121210000
cat /sys/class/power_supply/BAT0/energy_now 
121210000
cat /sys/class/power_supply/BAT0/energy_full_design 
50500000
```
energy_full and energy_now  are wrong.

mrhpearson, 2024-06-03:
OK - I'll let them know.
If you're interested, we can try and walk thru the recalibration process manually...depending on your willingness to be a guinea-pig. Though I guess I should try it first....I need to go and dig them up :)
If you want to try it ping me off forum - mpearson-lenovo a t squebb dot ca

joaquinco, 2024-06-14:
Followup.

I don't know what changed but full capacity is showing correct numbers now.

I'm not exactly sure what changed, I upgraded my linux distro and the kernel to 6.8 a few weeks ago but that didn't fix the issue, at least immediately.  And I'm pretty sure the firmware update from last month (0.1.26) didn't fix it either. 

Until today I noticed it was resolved : - )

daroczig, 2024-06-14:
Interestingly, I'm experiencing this issue only after suspend/resume :zipper_mouth_face: 
After a fresh boot, the `energy-full` and `energy-full-design` values are normal, but later those become inflated numbers.

mrhpearson, 2024-06-17:
Interesting - not sure why that would make a difference, but asking the battery team.

mrhpearson, 2024-07-03:
Would you be able to dump your EC registers? 
sudo modprobe ec_sys
hexdump -C /sys/kernel/debug/ec/ec0/io

Notes: you may need to install the hexdump utility (it's available with most distro's) and to get the ec_sys kernel module on Fedora you need to install the debug kernel (sudo dnf install kernel-debug)

daroczig, 2024-07-03:
This is on Arch:

```
$ sudo hexdump -C /sys/kernel/debug/ec/ec0/io
00000000  a6 05 20 c5 00 86 05 00  00 08 47 00 00 01 80 00  |.. .......G.....|
00000010  00 00 ff ff d0 bc 00 7d  7b ff 00 00 ff ff 1d 06  |.......}{.......|
00000020  00 00 00 00 00 00 00 00  00 00 00 01 7c 00 00 80  |............|...|
00000030  00 00 02 00 70 04 04 00  84 00 60 18 00 5f 00 00  |....p.....`.._..|
00000040  00 00 00 00 00 00 14 84  51 0c 00 00 00 00 04 00  |........Q.......|
00000050  00 80 02 03 00 00 00 20  00 00 00 00 00 00 00 00  |....... ........|
00000060  dc b0 04 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 04 00 00 00  32 31 00 00 00 00 00 00  |........21......|
00000080  00 00 05 06 aa 09 02 00  00 00 00 00 00 00 2b 00  |..............+.|
00000090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000a0  c6 1b c6 1b ff ff 64 00  00 00 e2 42 ff ff e0 40  |......d....B...@|
000000b0  00 00 00 00 00 00 00 00  00 00 87 05 00 96 01 00  |................|
000000c0  00 00 41 00 0d 00 00 00  01 58 00 04 08 00 00 00  |..A......X......|
000000d0  29 c0 00 00 00 00 03 01  1f 68 3c 00 00 00 00 00  |)........h<.....|
000000e0  00 00 00 00 00 00 00 00  12 20 0c 24 e4 ae 44 03  |......... .$..D.|
000000f0  4e 32 4f 48 54 33 35 57  28 7a 74 d4 00 00 00 00  |N2OHT35W(zt.....|
00000100
```

anayrat, 2024-07-04:
Hello, 
Here are mine:

```
00000000  a6 05 3c 46 00 86 00 00  00 04 00 00 8a 41 80 00  |..<F.........A..|
00000010  00 00 00 00 00 00 01 00  00 00 00 00 00 00 00 00  |................|
00000020  80 00 00 00 00 00 00 00  00 00 00 00 00 00 00 80  |................|
00000030  01 00 00 00 70 04 00 00  c4 00 30 00 00 00 00 00  |....p.....0.....|
00000040  00 00 00 00 00 00 04 00  00 00 00 00 00 00 00 00  |................|
00000050  00 80 02 03 02 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 04 00 00 00  2d 00 00 00 00 00 00 80  |........-.......|
00000080  00 01 00 06 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000090  00 00 00 00 00 e0 e4 00  00 00 00 00 00 00 00 00  |................|
000000a0  20 e0 18 00 3b 00 eb fc  ff 0b e2 00 0f 0c 00 00  | ...;...........|
000000b0  00 00 00 00 00 00 00 26  00 00 24 00 00 00 00 00  |.......&..$.....|
000000c0  00 00 00 00 00 00 00 03  00 00 00 00 00 00 10 00  |................|
000000d0  00 c0 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000e0  00 00 00 00 00 00 00 00  12 60 00 20 04 a0 06 03  |.........`. ....|
000000f0  52 31 4d 48 54 35 36 57  30 65 48 44 00 31 00 00  |R1MHT56W0eHD.1..|
00000100
```

mrhpearson, 2024-07-04:
@anayrat - did you use my ecreg tool that's on github? If so - it needs tweaking as in this case we don't want offset 0x81 set to 1. It needs to be set to 0 for this case (this changes what values are displayed in the later registers).
If you didn't use that tool, then I think you'll have to wait a few seconds (30?) and try again - and make sure offset 0x81 is clear.

Can you collect another dump, when the issue is seen, and make sure offset 0x81 = 00.

Thanks

anayrat, 2024-07-05:
Hello, I did not used your tool. I waited more than one hour, and it seems there is still 1 at 0x81 :
```
sudo  hexdump -C /sys/kernel/debug/ec/ec0/io
00000000  a6 05 3c 46 00 86 00 00  00 04 00 00 8a 41 80 00  |..<F.........A..|
00000010  00 00 00 00 00 00 01 00  00 00 00 00 00 00 00 00  |................|
00000020  80 00 00 00 00 00 00 00  00 00 00 00 00 00 00 80  |................|
00000030  01 00 00 00 70 04 00 00  84 00 30 00 00 00 00 00  |....p.....0.....|
00000040  00 00 00 00 00 00 14 00  00 04 00 00 00 00 00 00  |................|
00000050  00 80 02 03 02 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 04 00 00 00  3b 00 00 00 00 00 00 80  |........;.......|
00000080  00 01 00 06 54 0b 00 00  00 00 00 00 00 00 00 00  |....T...........|
00000090  00 00 00 00 00 e0 e4 00  00 00 00 00 00 00 00 00  |................|
000000a0  20 e0 18 00 3c 00 00 00  f0 0b ff ff 07 0c 00 00  | ...<...........|
000000b0  00 00 00 00 00 00 00 26  00 00 24 00 00 00 00 00  |.......&..$.....|
000000c0  00 00 42 00 00 00 00 03  00 41 00 00 00 00 10 00  |..B......A......|
000000d0  00 c0 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000e0  00 00 00 00 00 00 00 00  12 60 00 20 04 a0 06 03  |.........`. ....|
000000f0  52 31 4d 48 54 35 36 57  30 65 48 44 00 31 00 00  |R1MHT56W0eHD.1..|
00000100

```

mrhpearson, 2024-07-05:
Huh - how curious. It's supposed to be 00 by default. That might be useful for the EC team so I will let them know

Would you mind trying this:

 - Take the code at https://github.com/mrhpearson/ecreg-dump (this utility is normally used for dumping some of the thermal registers in the EC)
 - Change the code at line 31 (https://github.com/mrhpearson/ecreg-dump/blob/3a39c225fa46b4b51f90720d60e6f3e34580c7d8/ecreg-dump.c#L31) so that it writes 0x00 instead of 0x60.
 - Build and run it and see what that gives you. Hopefully we will get a sensible value at 0xA6

anayrat, 2024-07-05:
Here is the result:
```
00: a6 05 3c 46 00 86 00 00 00 04 00 00 8a 41 80 00 
10: 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 
20: 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 
30: 01 00 00 00 70 04 00 00 84 00 30 00 00 00 00 00 
40: 00 00 00 00 00 00 14 00 00 04 00 00 00 00 00 00 
50: 00 80 02 03 02 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 00 00 00 04 00 00 00 33 00 00 00 00 00 00 80 
80: 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 e0 e4 00 00 00 00 00 00 00 00 00 
a0: 59 2f 59 2f ff ff 64 00 00 00 10 32 ff ff e0 00 
b0: 00 00 00 00 00 00 00 26 00 00 24 00 00 00 00 00 
c0: 00 00 42 00 00 00 00 03 00 41 00 00 00 00 10 00 
d0: 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 12 60 00 20 04 a0 06 03 
f0: 52 31 4d 48 54 35 36 57 30 65 48 44 00 31 00 00 
```

mrhpearson, 2024-07-05:
Excellent - that confirms that the values returned from the battery FW are out of whack. I will forward to the battery team for their review.
Thank you!

mrhpearson, 2024-07-05:
Sorry - let me retract the above. Writing the details to the bug for the FW I realised that counting is hard and I'd goofed.
0xA6 is set to 0x64 which equals 100%

So the EC registers are correct (assuming your system is fully charged).
Can you confirm that when you get this that cat /sys/class/power_supply/BAT0/capacity is reporting > 100% please?

If we confirm that then I _think_ it means the problem is the BIOS not reporting the correct value to the OS. If so the good news is your battery is fine and this is just the BIOS being stupid.

anayrat, 2024-07-06:
Hello,
Here are the results : 
```
sudo ./ecreg-dump
EC register collector

00: a6 05 3c 46 00 86 00 00 00 04 00 00 8a 41 80 00 
10: 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 
20: 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 
30: 41 00 00 00 70 04 00 00 84 00 30 00 00 00 00 00 
40: 00 00 00 00 00 00 14 00 00 04 00 00 00 00 00 00 
50: 00 80 02 03 02 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 00 00 00 04 00 00 00 36 00 00 00 00 00 00 80 
80: 00 00 00 06 28 0a 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 e0 e4 00 00 00 00 00 00 00 00 00 
a0: 59 2f 59 2f ff ff 64 00 00 00 13 32 ff ff e0 00 
b0: 00 00 00 00 00 00 00 26 00 00 24 00 00 00 00 00 
c0: 00 00 42 00 00 00 00 03 00 3a 00 00 00 00 10 00 
d0: 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 12 60 00 20 04 a0 06 03 
f0: 52 31 4d 48 54 35 36 57 30 65 48 44 00 31 00 00 

cat /sys/class/power_supply/BAT0/capacity
100

upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               LGC
  model:                5B10W51826
  serial:               2132
  power supply:         yes
  updated:              Sat 06 Jul 2024 10:42:19 AM CEST (30 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               fully-charged
    warning-level:       none
    energy:              121.21 Wh
    energy-empty:        0 Wh
    energy-full:         121.21 Wh
    energy-full-design:  50.5 Wh
    energy-rate:         0 W
    voltage:             12.818 V
    charge-cycles:       60
    percentage:          100%
    capacity:            100%
    technology:          lithium-polymer
    icon-name:          'battery-full-charged-symbolic'

```

daroczig, 2024-07-08:
When I [posted above](https://github.com/fwupd/firmware-lenovo/issues/412#issuecomment-2206051078), my battery was indeed 100%. Now after a few suspend/resume, battery stats are skewed:

```
$ upower -i /org/freedesktop/UPower/devices/battery_BAT0

  native-path:          BAT0
  vendor:               Celxpert
  model:                5B10W&!!&
  serial:               9761
  power supply:         yes
  updated:              Mon 08 Jul 2024 12:18:55 PM CEST (16 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               discharging
    warning-level:       low
    energy:              73.42 Wh
    energy-empty:        0 Wh
    energy-full:         654.45 Wh
    energy-full-design:  328.96 Wh
    energy-rate:         17.756 W
    voltage:             16.35 V
    charge-cycles:       N/A
    time to empty:       4.1 hours
    percentage:          11%
    capacity:            100%
    technology:          lithium-polymer
    icon-name:          'battery-caution-symbolic'
  History (rate):
    1720433935  17.756  discharging
    1720433905  18.052  discharging
    1720433875  18.294  discharging
    1720433845  18.110  discharging
```

Note the inflated `energy-full` value.

The EC registers at the time being:

```
00000000  a6 05 20 c5 00 86 05 00  00 08 47 00 00 01 80 00  |.. .......G.....|
00000010  00 00 ff ff d0 bc 00 7d  7b ff 00 00 ff ff 1d 06  |.......}{.......|
00000020  00 00 00 00 00 00 00 00  00 00 00 01 63 00 00 80  |............c...|
00000030  40 01 02 00 70 04 04 00  c4 00 60 18 00 5f 00 00  |@...p.....`.._..|
00000040  00 00 00 00 00 00 04 84  50 10 00 00 00 00 04 00  |........P.......|
00000050  00 80 02 03 00 00 00 20  00 00 00 00 00 00 00 00  |....... ........|
00000060  dc b0 04 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 04 00 00 00  31 30 00 00 00 00 00 00  |........10......|
00000080  00 00 05 06 00 00 02 00  00 00 00 00 00 00 2b 00  |..............+.|
00000090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000a0  41 1c a5 1d 08 01 5f 00  e1 fb 75 3f ff ff e0 00  |A....._...u?....|
000000b0  00 00 00 00 00 00 00 00  00 00 87 05 00 96 01 00  |................|
000000c0  00 00 00 00 0d 00 00 00  01 00 00 04 08 00 00 00  |................|
000000d0  29 c0 00 00 00 00 00 01  1f 68 3c 00 00 00 00 00  |)........h<.....|
000000e0  00 00 00 00 00 00 00 00  12 20 0c 24 e4 ae 44 03  |......... .$..D.|
000000f0  4e 32 4f 48 54 33 35 57  28 7a 74 d4 00 00 00 00  |N2OHT35W(zt.....|
00000100
```

daroczig, 2024-07-08:
That's awesome. Current status:

```
# cat /sys/class/power_supply/BAT0/capacity
9

# hexdump -C /sys/kernel/debug/ec/ec0/io

00000000  a6 05 20 c5 00 86 05 00  00 08 47 00 00 01 80 00  |.. .......G.....|
00000010  00 00 ff ff d0 bc 00 7d  7b ff 00 00 ff ff 1d 06  |.......}{.......|
00000020  00 00 00 00 00 00 00 35  00 00 00 01 63 00 00 80  |.......5....c...|
00000030  40 00 02 00 70 04 00 00  a4 00 60 18 00 5f 00 00  |@...p.....`.._..|
00000040  00 00 00 00 00 00 14 84  50 04 00 00 00 00 00 00  |........P.......|
00000050  00 80 02 03 00 00 00 20  00 00 00 00 00 00 00 00  |....... ........|
00000060  dc b0 04 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 04 00 00 00  2d 2b 00 00 00 00 00 00  |........-+......|
00000080  00 00 05 06 ab 09 02 00  00 00 00 00 00 00 2b 00  |..............+.|
00000090  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000a0  f6 18 a5 1d ff ff 54 00  a1 06 34 43 3f 00 80 00  |......T...4C?...|
000000b0  00 00 00 00 00 00 00 00  00 00 87 05 00 96 01 00  |................|
000000c0  00 00 01 00 0d 00 00 00  01 41 00 04 08 00 00 00  |.........A......|
000000d0  29 c0 00 00 00 00 00 01  1f 68 3c 00 00 00 00 00  |)........h<.....|
000000e0  00 00 00 00 00 00 00 00  12 20 0c 24 e4 ae 44 03  |......... .$..D.|
000000f0  4e 32 4f 48 54 33 35 57  28 7a 74 d4 00 00 00 00  |N2OHT35W(zt.....|
00000100
```

And just in case, current power output:

```
# upower -i /org/freedesktop/UPower/devices/battery_BAT0                                                                                                                                                              
  native-path:          BAT0                                                                                                                                                                                                                   
  vendor:               Celxpert
  model:                5B10W&!!&
  serial:               9761
  power supply:         yes
  updated:              Mon 08 Jul 2024 02:41:34 PM CEST (0 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              63.72 Wh
    energy-empty:        0 Wh
    energy-full:         654.45 Wh
    energy-full-design:  328.96 Wh
    energy-rate:         0 W
    voltage:             16.482 V
    charge-cycles:       N/A
    percentage:          9%
    capacity:            100%
    technology:          lithium-polymer
    icon-name:          'battery-caution-charging-symbolic' 
  History (rate):
    1720442491  0.000   charging
    1720442473  20.956  discharging
    1720442443  22.169  discharging
    1720442413  25.690  discharging
    1720442402  0.000   discharging
    1720442402  24.128  discharging
    1720442397  26.370  charging
```

vibuz, 2024-07-15:
Same problem on my Lenovo ThinkPad P15g Gen 1. I see `121.21 Wh` is a popular number in here. 

```
$ upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               LGC
  model:                5B10W13958
  serial:               1146
  power supply:         yes
  updated:              Mon 15 Jul 2024 09:55:32 PM CEST (7 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              79.71 Wh
    energy-empty:        0 Wh
    energy-full:         121.21 Wh
    energy-full-design:  94 Wh
    energy-rate:         40.843 W
    voltage:             12.403 V
    charge-cycles:       69
    time to full:        1.0 hours
    percentage:          65%
    capacity:            100%
    technology:          lithium-polymer
    icon-name:          'battery-full-charging-symbolic'
```

Pelly1980, 2024-08-21:
Hi everybody!
So I just happened to found I have the same issue on a T14G2 intel: battery with P/n 5B10W51826 (LGC) with designated capacity of 50Wh is now displaying 121.21Wh capacity.
As I have a T14 G1 AMD, I tried to install it on that machine and the capacity remain unchanged: 121,21Wh are displayed also on T14G1 AMD model.

The battery firmware as well as the BIOS (in both PCs) are up-to-date: I tried to run a battery firmware update but I received an error "The device's firmware version on the system is equal or greater than the given versions."...

mrhpearson, 2024-08-21:
I have a trial BIOS (T14 G2 /P14s G2 AMD) with a potential fix for anybody who is willing to try it. 
Please drop me a note (mpearson-lenovo at squebb dot ca) and I'll share the details - I can't post the BIOS publicly.

bneils, 2025-02-16:
Having the exact same issue here on BIOS 1.30 on the P14s G2a. Battery model is 5B10W51826. Some people reported that getting a different battery w/ model 5B10W51830 makes this go away. However, this battery is relatively new with only 60 charge cycles. It came to me flat and cold when I bought it so I think either that or because I updated my BIOS to 1.30 that this is happening.

anayrat, 2025-09-04:
Hello,
It's been a while. Any news on this issue ?
Thanks

anayrat, 2025-09-10:
That unfortunate .... My laptop is no longer under warranty