fwupd/firmware-lenovo #413: Lenovo E14 Gen 2 Type C PD Firmware update fails

BasicComputers, 2024-01-24:
I have a Lenovo E14 Gen 2 device that does not charge anymore due to an issue with the previous firmware version for the USB C PD. Installing the latest firmware update R1ETR04W_WEB would resolve this issue an allow the device to charge again but when installing the update it states that "This process requires power to avoid accidental power-off during an update." Is there anyway to bypass this error in order to get the firmware installed?

[image]

hughsie, 2024-01-24:
@mrhpearson I think that warning is from the capsule, not from fwupd -- any way around it?

mrhpearson, 2024-01-24:
Afraid I don't know - sorry. @ChiWei-Chen  - any tricks you know of from the battery issues on the certified platforms we had previously

And as the E14 G2 was not Linux certified and has no FW on LVFS I assume you are updating via some alternate mechanism? It's recommended you contact Lenovo support in this case.

As it's out of scope for this forum I will close the ticket here.

closed 2024-01-24 by mrhpearson

ChiWei-Chen, 2024-01-25:
@BasicComputers please try the instruction "What do I do if my computer stops responding." from the manual https://download.lenovo.com/pccbbs/mobiles_pdf/e14_r14_e15_s3gen2_ug_v2_en.pdf

If this doesn't work for you, it's recommended contact Lenovo support. Thanks!

BasicComputers, 2024-01-26:
@ChiWei-Chen Its an issue with the firmware for sure and Lenovo wont do anything about it, All they need to do is release an update that allows for the firmware to run without power but I guess they wont do that. Anyways thanks for trying guys, I guess the only thing possible would be to modify the capsule file to allow for no power being needed.

arekm, 2025-03-25:
Some people just flash firmware directly on PD controller eeprom:

https://www.youtube.com/watch?v=OwRNi0KOfNY

Just where to get correct firmware for devices ... (like I'm looking for 20TA00F7GE e14 gen2 model PD fw).