fwupd/firmware-lenovo #414: ThinkPad Universal USB-C Dock v2 - firmware missing

lengau, 2024-01-26:
This looks similar to #238 , but it's for the [second generation dock](https://pcsupport.lenovo.com/us/en/solutions/acc500295).

```
│   
├─Vendor Interface:
│     Device ID:          05285288b47a52d357895905caed1c00e37e22f0
│     Current version:    1.0.1.4
│     Vendor:             Wistron (USB:0x17EF)
│     GUIDs:              78d3f11b-6ad9-5739-a650-2f0d2955a710 ← USB\VID_17EF&PID_30EF
│                         949fae77-e058-5343-bafb-a2540d69e9b5 ← USB\VID_17EF&PID_30EF&MCUID_M4521
│     Device Flags:       • Updatable
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│     Device Requests:    • Message
```

The device has no available firmware updates (including with `lvfs-testing` enabled):

```
$ fwupdmgr get-updates 05285288b47a52d357895905caed1c00e37e22f0
Devices with no available firmware updates: 
 • Vendor Interface
No updatable devices
```

But their website offers a [firmware version 1.0.21 (1.0.2.1?) installer for Windows](https://pcsupport.lenovo.com/us/en/downloads/DS563328). IDK whether they've uploaded and embargoed it or if it's simply not been made available to LVFS yet.

hughsie, 2024-01-26:
@a999153 any ideas?

c0d3z3r0, 2024-02-08:
@mrhpearson how is the Windows updater doing it then?

mrhpearson, 2024-02-08:
They didn't share that detail with me unfortunately.

c0d3z3r0, 2024-02-08:
Maybe ask again? I mean, if the Windows update can do it... fwupd can do. :-)