fwupd/firmware-lenovo #415: Foxconn X55 firmware crashes witout reason

EnzephaloN, 2024-01-30:
Thinkpad P1 gen4 with Arch (6.7.2-arch1-1)
X55 firmware: T99W175.F0.1.0.0.9.TF.008.079

````
...
Jan 30 13:17:43 thinkpad kernel: sequence number glitch prev=3 curr=0
Jan 30 13:17:51 thinkpad kernel: sequence number glitch prev=2 curr=0
Jan 30 13:18:07 thinkpad kernel: sequence number glitch prev=87 curr=0
Jan 30 13:18:51 thinkpad kernel: mhi-pci-generic 0000:08:00.0: firmware crashed (6)
Jan 30 13:18:51 thinkpad kernel: wwan0: Failed to queue TX buf (-5)
Jan 30 13:18:52 thinkpad ModemManager[1372]: <msg> [modem1] port 'wwan0mbim0' no longer controllable, reprobing
Jan 30 13:18:52 thinkpad NetworkManager[1300]: <info>  [1706617132.0900] device (wwan0mbim0): state change: activated -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Jan 30 13:18:52 thinkpad kernel: wwan wwan0: port wwan0qcdm0 disconnected
Jan 30 13:18:52 thinkpad kernel: wwan wwan0: port wwan0mbim0 disconnected
Jan 30 13:18:52 thinkpad kernel: wwan wwan0: port wwan0at0 disconnected
Jan 30 13:18:52 thinkpad ModemManager[1372]: <wrn> [/dev/wwan0mbim0] MBIM error: Device must be open to send commands
Jan 30 13:18:52 thinkpad ModemManager[1372]: <wrn> [/dev/wwan0mbim0] MBIM error: Device must be open to send commands
Jan 30 13:18:52 thinkpad ModemManager[1372]: <wrn> [/dev/wwan0mbim0] MBIM error: Device must be open to send commands
Jan 30 13:18:52 thinkpad ModemManager[1372]: <wrn> [/dev/wwan0mbim0] MBIM error: Device must be open to send commands
Jan 30 13:18:52 thinkpad ModemManager[1372]: <wrn> [/dev/wwan0mbim0] MBIM error: Device must be open to send commands
Jan 30 13:18:52 thinkpad ModemManager[1372]: <wrn> [modem1/wwan0mbim0/mbim] Couldn't properly close QMI device: Device must be open to send commands
Jan 30 13:18:52 thinkpad passimd[2919]: resetting Passim-D3AC
Jan 30 13:18:52 thinkpad passimd[2919]: Emitting PropertyChanged('Status'='running')
Jan 30 13:18:52 thinkpad passimd[2919]: signal_name: StateChanged (is)
Jan 30 13:18:52 thinkpad ModemManager[1372]: <msg> [base-manager] port wwan0qcdm0 released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:08:00.0'
Jan 30 13:18:52 thinkpad ModemManager[1372]: <msg> [base-manager] port wwan0mbim0 released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:08:00.0'
Jan 30 13:18:52 thinkpad ModemManager[1372]: <msg> [base-manager] port wwan0at0 released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:08:00.0'
Jan 30 13:18:52 thinkpad ModemManager[1372]: <msg> [base-manager] port wwan0 released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:08:00.0'
Jan 30 13:18:52 thinkpad dnsmasq[1523]: no servers found in /etc/resolv.conf, will retry
Jan 30 13:18:52 thinkpad NetworkManager[1300]: <info>  [1706617132.1380] manager: NetworkManager state is now CONNECTED_LOCAL
Jan 30 13:18:52 thinkpad NetworkManager[1300]: <warn>  [1706617132.1388] modem-broadband[wwan0mbim0]: failed to disconnect modem: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Object does not exist at path “/org/freedesktop/ModemManager1/Modem/1”
Jan 30 13:18:52 thinkpad evolution[3665]: Network disconnected.  Forced offline.
Jan 30 13:18:52 thinkpad systemd[1]: Starting Network Manager Script Dispatcher Service...
Jan 30 13:18:52 thinkpad systemd[1]: Started Network Manager Script Dispatcher Service.
Jan 30 13:18:52 thinkpad kernel: wwan wwan0: port wwan0qcdm0 attached
Jan 30 13:18:52 thinkpad kernel: wwan wwan0: port wwan0mbim0 attached
Jan 30 13:18:52 thinkpad kernel: wwan wwan0: port wwan0at0 attached
Jan 30 13:18:52 thinkpad vivaldi-stable.desktop[2939]: Fontconfig error: Cannot load default config file: No such file: (null)
Jan 30 13:18:53 thinkpad passimd[2919]: signal_name: StateChanged (is)
Jan 30 13:18:54 thinkpad ModemManager[1372]: <msg> [wwan0mbim0/mbim] MBIM device is not QMI capable
Jan 30 13:18:56 thinkpad ModemManager[1372]: <msg> [device /sys/devices/pci0000:00/0000:00:1c.4/0000:08:00.0] creating modem with plugin 'foxconn' and '4' ports
Jan 30 13:18:56 thinkpad ModemManager[1372]: <msg> [base-manager] modem for device '/sys/devices/pci0000:00/0000:00:1c.4/0000:08:00.0' successfully created
Jan 30 13:18:56 thinkpad ModemManager[1372]: <msg> [modem2/wwan0mbim0/mbim] MBIM device is QMI capable
Jan 30 13:18:56 thinkpad ModemManager[1372]: <msg> [modem2] state changed (unknown -> disabled)
Jan 30 13:18:56 thinkpad NetworkManager[1300]: <info>  [1706617136.5651] manager: (wwan0mbim0): new Broadband device (/org/freedesktop/NetworkManager/Devices/8)
Jan 30 13:18:56 thinkpad NetworkManager[1300]: <info>  [1706617136.5655] device (wwan0mbim0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Jan 30 13:18:56 thinkpad NetworkManager[1300]: <info>  [1706617136.5660] device (wwan0mbim0): modem state 'enabling'
Jan 30 13:18:56 thinkpad NetworkManager[1300]: <info>  [1706617136.5666] device (wwan0mbim0): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
````

hughsie, 2024-01-30:
> mhi-pci-generic 0000:08:00.0: firmware crashed

I don't think the right people are here -- this is more for EUFI update failures specific to Lenovo. Sorry.

closed 2024-01-30 by hughsie

hughsie, 2024-01-30:
@mrhpearson any ideas?

mrhpearson, 2024-01-30:
Hi @EnzephaloN - could you post a ticket on forums.lenovo.com/linux - and then I can pull in some folk from the WWAN team to have a look.

Afraid I have to highlight that this modem has not been certified or enabled with Linux on the P1G4. If you've unlocked the modem by alternative mechanisms...we don't know about those; and if you've not unlocked the modem then that might be why it's not working....

I did check with my colleague and he said he was using FW version T99W175.F0.1.0.0.9.AT.009 for the testing done in the US....but I know FW versions switch between carriers so not sure if that is helpful or not.

Mark

EnzephaloN, 2024-01-31:
Hello Mark.
I just did it with this way, like it's written in every documentation about WWAN-modems and linux (i.e. https://wiki.archlinux.org/title/Mobile_broadband_modem#FCC_locking ) . I did this years ago on my Thinkpad T470p too with its modem. Worked great for years. Thats the way -  you know ;-)
And the modem (as sold with my notebook) in my P1g4 also works fine for the most time. But sometimes (after hours or -in some cases- only minutes) the connection breaks because "firmware crashed".
The firmware T99W175.F0.1.0.0.9.TF.008.079 came by using ``fwupdmgr update`` some days before.

I will try to register to the n-th forum and ask there for help...

EnzephaloN, 2024-02-14:
Hey there
If created a thread at forums.lenovo.com/linux, but no replies on this :-( .
Anything here which might help me?

mrhpearson, 2024-02-14:
Please email me mpearson-lenovo at squebb dot ca