fwupd/firmware-lenovo #416: me fw update failed

issamejekkam, 2024-02-02:
Greetings everyone!

I've encountered a problem with my Lenovo ThinkPad L580 recently, and I'm seeking assistance.

I'm having difficulty understanding the issue. I suspect it might be related to interrupting the Ubuntu installation on my laptop. Every time I power it on, I see the error message "me fw update failed" on the screen. Occasionally, I can access the BIOS and encounter various other messages intermittently. However, the main issue is that the laptop shuts down directly after about 10 seconds, despite trying several troubleshooting tricks from YouTube without success. Could anyone please lend a hand in resolving this?

hughsie, 2024-02-03:
Did you recently do a firmware update with fwupd?

issamejekkam, 2024-02-03:
No, i don’t think so.

hughsie, 2024-02-03:
So,  I think this is a bug, but I don't think it's a fwupd bug I'm afraid. I think you might have more luck with the Lenovo linux forums.

issamejekkam, 2024-02-03:
Okay, thank you !

closed 2024-02-03 by hughsie