fwupd/firmware-lenovo #417: Lenovo X1 Yoga Gen 2 firmware 1.50 not in repo (required to combat LogoFAIL)

joe-clark-midsummer, 2024-02-06:
Other Lenovo products (e.g. T470s, T480s) have a firmware version recent enough on the fwupdmgr repo that simply running `fwupdmgr update` will update to a firmware version that patches LogoFAIL. However, the latest firmware version for X1 Yoga Gen 2 is currently 1.49, and the patch occurs in 1.50. Please note that this is listed as a high severity vulnerability by Lenovo, so it is important to have this up to date.

Thanks,
Joe

mrhpearson, 2024-02-06:
1.50 is on LVFS - but under embargo still.
I've created internal ticket  LO-2863 to check on the status and see if it can be moved to stable. As a note - it is Chinese New Year starting this week and into next week so it might take a little longer than normal to get updates.

joe-clark-midsummer, 2024-02-12:
> 1.50 is on LVFS - but under embargo still. I've created internal ticket LO-2863 to check on the status and see if it can be moved to stable. As a note - it is Chinese New Year starting this week and into next week so it might take a little longer than normal to get updates.

Sorry, this is probably me just me not understanding what "under emargo" means in this context, but how come the upgrade is available to download directly from Lenovo's website at https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-yoga-type-20jd-20je-20jf-20jg/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C?

hughsie, 2024-02-12:
Ohh it's less "security embargo" and more "not yet public". Sometimes the LVFS updates lag the windows updates by a few days.

mrhpearson, 2024-02-12:
Yep - it means the firmware team have uploaded the FW to the LVFS servers so they can do their testing process, before moving it to the stable channel where it's generally available for you to use.
It always takes a little longer for FW to get onto LVFS than the Lenovo site - usually a week, but sometimes longer.

joe-clark-midsummer, 2024-02-15:
Understood, thanks. Does seem to be quite a bit more than a week in this case though!

One thing I really don't understand about this particular vulnerability fix is as follows. Take the Lenovo X1 Yoga Gen 2 for example. According to https://support.lenovo.com/us/en/product_security/ps500590-bios-image-parsing-function-vulnerabilities-logofail, firmware 1.50 patches the vuln. Yet 1.50 was released on 22 Nov 2023, and LogoFAIL wasn't even discovered until Dec 6! I don't really understand this. (Sorry as you can probably tell, this area isn't really my forte.)

mrhpearson, 2024-02-15:
Fair point on it being longer than a week - it's possible the FW team have missed promoting this to stable (it has been in the embargo channel for two months which is too long). The internal ticket will nudge them on that when they're back from CNY.

For dates - I believe the vulnerability was discovered pre September 2023, but only publicly disclosed on 6th Dec (after vendors have been able to patch and release fixes). It's disappointing the LVFS FW wasn't released before the announce date :(

joe-clark-midsummer, 2024-03-11:
Has there been any update from Lenovo on this?

mrhpearson, 2024-03-12:
Hi, and yes! FW was moved to stable a few hours ago. Sorry it took so long, there were some update issues they had to work through.
Please let me know if you see any issues.

joe-clark-midsummer, 2024-03-20:
Awesome, update worked a charm. Thanks guys for getting that over the line, I can finally sleep knowing I don't have to go round 200 laptops with a USB stick...

closed 2024-03-20 by joe-clark-midsummer