fwupd/firmware-lenovo #419: "Could not apply capsule: Device Error" is reported on boot (ThinkPad T490s)

jodeleit-es, 2024-02-11:
**Describe the question**
During boot I got the following error mesage
```shell
fwupd-efi version 1.4
WARNING: Could not apply capsules: Device Error
Reset System
```
This is on a ThinkPad T490s.

Any pointers what could be the culprit here?

**fwupd version information**

```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-15-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

hughsie, 2024-02-11:
@mrhpearson any ideas? I've not seen this before.

mrhpearson, 2024-02-12:
Can I get some details on what FW was being updated (and what you're updating from) please

jodeleit-es, 2024-02-12:
This is the view of the "Software" app:
[image]

According to `fwupdmgr get-upgrades` the first "device" is "Intel Management Engine" (seems it never got an update installed ever). The second is device is labeled "System Firmware".

Does this help or do you need the output of some special commands?

mrhpearson, 2024-02-12:
I didn't expect to see ME with version 0....that can't be helping and seems suspicious
Could you boot to the BIOS setup screen (F1 during early boot) and see what is reported there please

jodeleit-es, 2024-02-12:
BIOS reports 12.0.93.2331 for the `ME Firmware Version`.

According to `fwupdmgr get-upgrades` this is the latest version, just in another format: `Version 12.0.93.2331 (LVFS: 192.93.2331)`

mrhpearson, 2024-02-12:
OK - but that likely means your BIOS is not reporting the ME version to the system.
Just so we can confirm that, can you do:
```
grep . /sys/firmware/efi/esrt/entries/entry?/fw_version
grep . /sys/firmware/efi/esrt/entries/entry?/fw_class
```
I have no idea if it's important related to your FW update problem...but I'm scratching my head a bit for next steps. I need to go look at the BIOS release notes

jodeleit-es, 2024-02-13:
Thanks for your support!

```
$ sudo grep . /sys/firmware/efi/esrt/entries/entry?/fw_version
/sys/firmware/efi/esrt/entries/entry0/fw_version:65610
/sys/firmware/efi/esrt/entries/entry1/fw_version:0
/sys/firmware/efi/esrt/entries/entry2/fw_version:65553
/sys/firmware/efi/esrt/entries/entry3/fw_version:16842759
/sys/firmware/efi/esrt/entries/entry4/fw_version:117571840
/sys/firmware/efi/esrt/entries/entry5/fw_version:26403
$ sudo grep . /sys/firmware/efi/esrt/entries/entry?/fw_class
/sys/firmware/efi/esrt/entries/entry0/fw_class:6300ba52-d9ea-4ca0-88f9-f89ef73cc07b
/sys/firmware/efi/esrt/entries/entry1/fw_class:ef5cdc85-9cf6-469d-9cb7-920b7dd6672b
/sys/firmware/efi/esrt/entries/entry2/fw_class:be9a5f93-930b-40bf-8864-815268ea4ea3
/sys/firmware/efi/esrt/entries/entry3/fw_class:23c7887b-48fc-4d5e-81c6-8f8a91874786
/sys/firmware/efi/esrt/entries/entry4/fw_class:41fc4d0b-3b42-4702-14a5-f615c4b03a42
/sys/firmware/efi/esrt/entries/entry5/fw_class:80c2e1e4-0a6a-42d0-b188-5a218e57c8de
```

mrhpearson, 2024-02-13:
OK - thanks for confirming. The BIOS is not updating the ME FW version correctly (entry1 is the ME)

I think the BIOS is going to need re-installing, but would like review feedback from the FW team. I've created internal ticket  LO-2870 for tracking - but it is Chinese New Year this week so don't expect to hear back until next week unfortunately.

It might be worth trying an install of version 1.75 in the meantime?
https://fwupd.org/downloads/d7db7fc7bcc2af6064ad87d2be3fce7795663cae678a2bffd34c8a4ae852603f-Lenovo-ThinkPad-X390T490s-SystemFirmware-1.75.cab 

Mark

jodeleit-es, 2024-02-13:
Thanks again for your effort. I think I'll stick with the current firmware until a new one gets available and ignore the misleading "update needed" visualization for the ME for now.

Do you have an idea concerning the problem with the "system fimware" (entry0 above)? Or should I create a new ticket for this? Or is this the same problem?

mrhpearson, 2024-02-13:
I don't I'm afraid - I'm hoping the FW team can explain.

kmauleon, 2024-02-21:
as per ODM owner... can you please update all available updates first.  please update firmware one at a time especially for BIOS, EC, and MEFW

ODM owner says:
"       I have checked this error on Github and I can’t duplicate it on my own machines. 
Since the ME version in setup is normal, I don’t think update BIOS can fix the “ME version 0” error.

       I confirmed with UT team. They met several similar error such as : ME version 687930 (abnormal ME version). The solution of this kind error is update all available updates.
       I believe the version of fwupd decide whether it can obtain the ME version."

jodeleit-es, 2024-02-21:
Can you specify _"update all available updates"_? There is only the "system firmware" update I cannot apply, everything else is up-to-date.

I will try this update again (isolated) later this day and report back..

jodeleit-es, 2024-02-21:
The "system firmware" update in isolation seems to have worked out, it is no longer shown as "missing update".
But the "me" update is still shown as needed and the actual version is reported as "0".
During the update boot cycle there some errors reported, not sure if these are related. 
This is from dmesg with some surrounding lines:
```
[Mi Feb 21 17:30:43 2024] ACPI: Added _OSI(Module Device)
[Mi Feb 21 17:30:43 2024] ACPI: Added _OSI(Processor Device)
[Mi Feb 21 17:30:43 2024] ACPI: Added _OSI(3.0 _SCP Extensions)
[Mi Feb 21 17:30:43 2024] ACPI: Added _OSI(Processor Aggregator Device)
[Mi Feb 21 17:30:43 2024] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.RP07.BRST], AE_ALREADY_EXISTS (20230331/dswload2-326)
[Mi Feb 21 17:30:43 2024] fbcon: Taking over console
[Mi Feb 21 17:30:43 2024] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230331/psobject-220)
[Mi Feb 21 17:30:43 2024] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.RP07.RSTX], AE_ALREADY_EXISTS (20230331/dswload2-326)
[Mi Feb 21 17:30:43 2024] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230331/psobject-220)
[Mi Feb 21 17:30:43 2024] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.RP07.RSTG], AE_ALREADY_EXISTS (20230331/dswload2-326)
[Mi Feb 21 17:30:43 2024] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230331/psobject-220)
[Mi Feb 21 17:30:43 2024] ACPI: Skipping parse of AML opcode: Method (0x0014)
[Mi Feb 21 17:30:43 2024] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.RP07.PXSX.L850], AE_ALREADY_EXISTS (20230331/dswload2-326)
[Mi Feb 21 17:30:43 2024] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230331/psobject-220)
[Mi Feb 21 17:30:43 2024] ACPI BIOS Error (bug): Failure creating named object [\_SB.PCI0.RP07.PXSX._RST], AE_ALREADY_EXISTS (20230331/dswload2-326)
[Mi Feb 21 17:30:43 2024] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230331/psobject-220)
[Mi Feb 21 17:30:43 2024] ACPI: Skipping parse of AML opcode: Method (0x0014)
[Mi Feb 21 17:30:43 2024] ACPI: 11 ACPI AML tables successfully acquired and loaded
[Mi Feb 21 17:30:43 2024] ACPI: EC: EC started
[Mi Feb 21 17:30:43 2024] ACPI: EC: interrupt blocked
[Mi Feb 21 17:30:43 2024] ACPI: EC: EC_CMD/EC_SC=0x66, EC_DATA=0x62
[Mi Feb 21 17:30:43 2024] ACPI: EC: Boot ECDT EC used to handle transactions
[Mi Feb 21 17:30:43 2024] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
```

kmauleon, 2024-02-21:
can you please share screenshot of your BIOS Setup menu @jodeleit-es thank you!

kmauleon, 2024-03-11:
@jodeleit-es any update from your side please

jodeleit-es, 2024-03-11:
Sorry for the delay
[image]
Do you need more?

kmauleon, 2024-03-11:
@jodeleit-es your MEFW is already at the latest... can you please try to execute
fwupdmgr reinstall

jodeleit-es, 2024-03-13:
```
$ fwupdmgr reinstall
Choose a device:
0.	Cancel
1.	[..]
2.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
[..]
2
Unable to locate release for Intel Management Engine version 0
```

jodeleit-es, 2024-03-13:
"reinstall" fails with this error message: "Unable to locate release for Intel Management Engine version 0"

kmauleon, 2024-03-13:
please try to download MEFW for your system [here](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRN.firmware)
and please execute below to manually reinstall

fwupdmgr --allow-reinstall --allow-older install <MEFW cab file>

jodeleit-es, 2024-03-15:
Done. The error message on boot is now gone. 
But the version of the ME is still reported as "0", even it's in fact "192.93.2331".

kmauleon, 2024-03-18:
BIOS is the one responsible for this display... you may need to reinstall BIOS or wait for the next update

kmauleon, 2024-03-24:
please try to execute as below....

fwupdmgr refresh --force
fwupdmgr update
restart system if any updates available
fwupdmgr get-devices

kmauleon, 2024-09-26:
closing as no update... let me know if need to reopen.  thank you very much.

closed 2024-09-26 by kmauleon