fwupd/firmware-lenovo #420: After a failed firmware upgrade on Lenovo L480, boot of laptop is not completed and sudden switch off

IsaiasGit1, 2024-02-11:
fwupdmgr get-history
WARNING: UEFI capsule updates not available or enable in firmware setup
      See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
###
|
|--Intel Managemente Engine:
|   |   Previous Version:      184.50.3425
|   |   Update State:            Failed
|   |   Update Error:            failed to run update on reboot
|   |- New version:           184.93.4323
|             Remote ID:          lvfs
|             License:               Unkown
...
|--System Firmware:
    |   Previous Version:      0.1.56
    |   Update State:            Failed
    |   Update Error:            failed to run update on reboot
    |- New version:           0.1.64
              Remote ID:          lvfs
              License:               Unknwon

Situation is crazy because when I boot the laptop the OS runs only for 40 seconds and then sudden switch off happens so I cannot perform any action.

Secure Boot is off
Both boot allowed, Legacy and UEFI

IsaiasGit1, 2024-02-11:
During the few seconds the systems is available, I tried to upgrade fwupdmgr but "failed to dwonload file: Could not resolve host: fwupd.org"

Update System

hughsie, 2024-02-12:
> fwupdmgr get-history
> WARNING: UEFI capsule updates not available or enable in firmware setup

How did you update the firmware?

IsaiasGit1, 2024-02-12:
OS is ubuntu desktop 22.04. OS suggested me update some packages and two Lenovo drivers

[image]

IsaiasGit1, 2024-02-12:
Ubuntu 22.04 desktop suggested the updates.

[image]

IsaiasGit1, 2024-02-12:
Sequence was (after update packages)
1. 1st boot and message: "(A7) Me FW Downgrade - Request MeSpiLock"
2. After 40 seconds, laptop switch off always. Time to execute any command is very short