fwupd/firmware-lenovo #421: P14s Gen 1 missing latest ECP firmware

sylvainfaivre, 2024-02-12:
See https://download.lenovo.com/pccbbs/mobiles/n2xuj22w.txt :

```
  Package ID            UEFI BIOS (BIOS ID)  ECP (ECP ID)      Rev.  Issue Date
  --------------------  -------------------  ---------------   ----  ----------
  N2XUJ22W              1.30 (N2XET40W)      1.16 (N2XHT26W)   01    2023/11/20
  N2XUJ21W              1.29 (N2XET39W)      1.16 (N2XHT26W)   01    2023/08/18
  N2XUJ20W              1.28 (N2XET38W)      1.15 (N2XHT25W)   01    2023/03/16
```

LVFS has the latest UEFI BIOS 1.29 and 1.30 : https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2XET.firmware

but misses the matching ECP release 1.16, only has 1.15 which matches BIOS 1.28 : https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2XHT.firmware

mrhpearson, 2024-02-12:
Thanks for letting me know.
Created internal ticket LO-2869 to ask the FW team why it hasn't been delivered

mrhpearson, 2024-02-26:
FW has been released - should be available now