fwupd/firmware-lenovo #422: Z13 AMD cannot wake from suspend

CosmicSyntax, 2024-02-13:
Hello -

I am running Fedora 39 (Kernel 6.7.4) on the z13 AMD machine. When I suspend the machine, it seem to sleep just fine.  However, I cannot seem to wake. Open the lid and press keys, and nothing.  The fan seems to start spinning, but the power light continue to blink and nothing is on the screen.

According to the OS, it seems like it's trying to use S3.  But, I don't see any options to use S3 in the bios.
cat /sys/power/mem_sleep
s2idle [deep]

Thanks for your help

hughsie, 2024-02-14:
Is this anything to do with firmware updates? If not, you might have more luck in the Lenovo linux forums I'm afraid.

LordChaos73, 2024-02-14:
I'm running Ubuntu 23.10 on this laptop, no issues at all.

```
cat /sys/power/mem_sleep
[s2idle]
```

# System Details Report
---

## Report details
- **Date generated:**                              2024-02-14 10:19:38

## Hardware Information:
- **Hardware Model:**                              Lenovo ThinkPad Z13 Gen 1
- **Memory:**                                      32,0 GiB
- **Processor:**                                   AMD Ryzen™ 7 PRO 6860Z with Radeon™ Graphics × 16
- **Graphics:**                                    AMD Radeon™ Graphics
- **Disk Capacity:**                               (null)

## Software Information:
- **Firmware Version:**                            N3GET64W (1.64 )
- **OS Name:**                                     Ubuntu 23.10
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               45.2
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.5.0-17-generic

CosmicSyntax, 2024-02-14:
Thank you both for your response. This may be an OS issue then.

closed 2024-02-14 by CosmicSyntax

winstonma, 2024-03-07:
@CosmicSyntax I had the same issue with Ubuntu Jammy. After replacing /lib/firmware with the lastest version this issue is gone.

Let me find the instructions for you and update this post later.