fwupd/firmware-lenovo #423: requirement 'hardware' also needs org.freedesktop.fwupd >= 1.0.8

jinek, 2024-02-14:
**Describe the bug**
```shell
jinek@fedora:~/Downloads$ sudo fwupdmgr --allow-older --allow-reinstall install ./n3arl02w.cab
Decompressing…           [  -                                    ]
requirement 'hardware' also needs org.freedesktop.fwupd >= 1.0.8

```

**Steps to Reproduce**
I'm just trying to install lenovo firmware update onto x1 yoga gen 7.

**fwupd version information**
1.9.13

```shell
compile   org.freedesktop.fwupd         1.9.13
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.13
runtime   org.kernel                    6.7.3-200.fc39.x86_64

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 21CDS09V00
│
├─ELAN067B:00 04F3:31F8:
│     Device ID:          4dc958f59f7f61d6e4fee930e995b76baf216980
│     Summary:            Touchpad
│     Current version:    0x0004
│     Bootloader Version: 0x0003
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              abe7b9d9-0449-51d0-b89e-49f4b54780b2 ← HIDRAW\VEN_04F3&DEV_31F8
│                         93386a84-d3f7-5ecc-bf58-7d4e14dc2a45 ← HIDRAW\VEN_04F3&DEV_31F8&MOD_012D
│                         4070ab62-370f-529a-80fc-6e60d076e04b ← ELANTP\ICTYPE_14&MOD_012D
│                         f8f4e9e5-53f1-5807-a237-6ba8807772cd ← ELANTP\ICTYPE_14&MOD_012D&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│   
├─12th Gen Intel Core™ i7-1260P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000432
│     Vendor:             Intel
│     GUIDs:              ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│                         3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│     Device Flags:       • Internal device
│   
├─Alder Lake-P GT2 [Iris Xe Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│                         f36f6bfd-1640-540f-8adb-2e6e7b3c227d ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22E6
│     Device Flags:       • Internal device
[... 371 lines cut ...]
│     Update State:       Success
│     Problems:           • Device requires AC power to be connected
│     GUID:               240cf38f-3985-42c5-b77b-8e4bdaf0fcc1
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E

```

</details>

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? Yes
- Is this a regression? Previous update worked fine

hughsie, 2024-02-14:
@mrhpearson if the Lenovo team still insist on doing this, they **do** need to set up the requirements properly: https://lvfs.readthedocs.io/en/latest/metainfo.html#using-chid

If they use the LVFS it auto-adds the correct requirements.

mrhpearson, 2024-02-14:
The uploading of cab files to the Lenovo site is being stopped - should be LVFS only. It should be in place already, but I won't be surprised if we get a few updates still - especially on older platforms.

Regardless - this one is (I think) from Nov 2022, so definitely before the change :)

@jinek - please use LVFS for updating from Linux. I did check and this version, and a newer one, are there. The reasons for the cab file being on the Lenovo site are historical and no longer apply.

hughsie, 2024-02-15:
@mrhpearson thanks!

closed 2024-02-15 by hughsie