fwupd/firmware-lenovo #429: X1 Extreme Gen 5 - Intel ME lvfs fix

Rick-Alloy, 2024-03-12:
I've still got issues from the bad lvfs firmware that was pushed out last year in that my onboard sound has not worked since, and booting takes around five minutes with the following flow:

"48% Reading ME Firmware...Please do not power off!"
"56% Recovering ME Firmware...Please do not power off!"
"60% Failed to recover ME Firmware..."

This is getting old, well past it actually.

  Is there any word on the fix that is being looked at?  If I throw a spare drive in and install Windows on it, will the MEFW update on the Lenovo site fix the issue?  Is there any known recourse for fixing this?

Rick-Alloy, 2024-03-12:
Unfortunately, no.  The ME firmware will not present as an option even with `--allow-reinstall --allow-older`.

kmauleon, 2024-03-13:
can you please show results for fwupdmgr get-devices.... thanks

Rick-Alloy, 2024-03-13:
Sure thing, here's the output.

`21DECTO1WW
│
├─ELAN0686:00 04F3:320D:
│     Device ID:          522cb97965758d9b1309d3217dfd41673f410e15
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0003
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              71de889a-ea8d-5c66-a0fb-7e11e860cd3d
│                         7e8dad8a-0b0e-527b-8810-d5dbca8804d7
│                         74e88f64-4d6b-5048-8199-fdcc7833b48a
│                         4b808d27-c833-50cc-8177-9a401f5c08fe
│                         e25cd415-0431-502a-9077-3c409b007897
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Embedded Controller:
│     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│     Summary:            UEFI ESRT device
│     Current version:    0.1.11
│     Minimum Version:    0.1.11
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               f8963b51-51b6-4059-87c4-722967c16745
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Fingerprint Sensor:
│     Device ID:          23cf6368c14a875f74c38a5a423518f38d8abbbc
│     Summary:            Match-On-Chip fingerprint sensor
│     Current version:    01000300
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     GUIDs:              6322b4f7-0571-5f07-8538-e041bbc70677
│                         14450e82-8667-54c3-98e1-098d8c6dc3b8
│     Device Flags:       • Updatable
│                         • Device can recover flash failures
│                         • Signed Payload
│   
├─Integrated Camera:
│     Device ID:          0fef0a0c55f6442bffaebd774ae771341c89571b
│     Current version:    56.20
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              b687616d-e0bc-58d3-9022-8f0065eed4ca
│                         bcd2c020-e315-55aa-b7e1-7d2d7108037e
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               23192307-d667-4bdf-af1a-6059db171246
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─MZVL4256HBJD-00BL7:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    CL2QHXC7
[... 183 lines cut ...]
│     Summary:            UEFI ESRT device
│     Current version:    18548864
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               012b2727-3f35-c20a-fb78-d7d4216ce715
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          5859fae972cb36551b299175761b33d8e6e7ce02
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.10_B0032
      Bootloader Version: BOT01.02_B0015
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                          279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Unsigned Payload`

Rick-Alloy, 2024-03-27:
I finally had a chance to look at those suggestions.  Linux Firmware Updater is already at the lowest priority.  I also reset the bios to factory defaults and even did a complete reinstall of the OS.

I still see the Reading, Recovering, Failed on ME firmware at every boot, and onboard sound is still not working in the OS.  I saw that a new firmware update is expected this week, will that address this issue?

kmauleon, 2024-04-01:
the upcoming update is for the MEFW version fix... not sure though if this will resolve your firmware update loop as well... can you please try to reinstall BIOS firmware via 
"fwupdmgr --allow-reinstall --allow-older install <[cab file](https://fwupd.org/downloads/7405ae173430321bedc5ddab73c801663261a3df40b9a65e80c6655204eab1ea-Lenovo-ThinkPad-P1Gen5X1Extreme5-SystemFirmware-1.20.cab)>"

Rick-Alloy, 2024-04-01:
I tried that update and while it appeared to succeed, the loops still occurs.  I also updated the MEFW, which is the cause of this update loop, and still ream caught in the loop.  Here is a snap of the BIOS Event log for reference.  As you can see the 16.1.xx BIOS is a factor in this problem.

[image]

Rick-Alloy, 2024-04-01:
One interesting thing I learned is that efibootmgr shows BootNext to be the Linux-Firmware-Updater even though I've tried disabling it in BIOS, and I'm wondering if that is part of the issue in the loop.  

❯ efibootmgr
BootNext: 0001
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0002,001C,001D,001E,001F,0020,0021,0022,0023,0024,0025,0026,0001
Boot0000* ubuntu
Boot0001  Linux-Firmware-Updater
Boot0002* Windows Boot Manager
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Asset Information
Boot0015  Regulatory Information
Boot0016  ThinkShield secure wipe
Boot0017  ThinkShield Passwordless Power-On Device Manager
Boot0018  Wi-Fi Configuration
Boot0019  Intel(R) MEBx
Boot001A  Startup Interrupt Menu
Boot001B  Rescue and Recovery
Boot001C* USB CD
Boot001D* USB FDD
Boot001E* NVMe0
Boot001F* NVMe1
Boot0020* ATA HDD0
Boot0021* USB HDD
Boot0022* PXE BOOT
Boot0023* LENOVO CLOUD
Boot0024* ON-PREMISE
Boot0025  Other CD
Boot0026  Other HDD
Boot0027* IDER BOOT CDROM
Boot0028* IDER BOOT Floppy
Boot0029* ATA HDD
Boot002A* ATAPI CD

kmauleon, 2024-04-01:
I noticed that your MEFW GUID is a mismatch with what is expected... Let's wait for the MEFW version fix as the GUID is related to this one.  Thank you very much.

kmauleon, 2024-04-02:
ODM just set the firmware to stable please check... thank you very much

Rick-Alloy, 2024-04-03:
I didn't see it come up in fwupdmgr, so I pulled it from lvfs manually and ran it.

The issues appear to be resolved.  No boot loop, and sound now works within the OS.  Thank you very much.

kmauleon, 2024-04-04:
great news! thanks for checking @Rick-Alloy... closing this ticket