fwupd/firmware-lenovo #430: P14s Gen2i - Missing Thunderbolt Retimer Firmware

vifino, 2024-03-12:
It seems like the [Thunderbolt Retimer Firmware](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p14s-gen-2-type-20vx--20vy/downloads/ds563937-intel-thunderbolt-retimer-firmware-for-windows-11-version-21h2-or-later-10-64-bit-version-1903-or-later-thinkpad?category=USB%20Device,%20FireWire,%20IEEE%201394,%20Thunderbolt) is missing for the P14s Gen2i.

After extracting the exe, I used `fwupdtool install-blob ./BBR.cap 408e7431-1bfc-4b53-b318-a756e7cd641c` to manually install it.
Worked fine, but of course it would be better if it was in the LVFS.

This update applies to T14 Gen 2, P14s Gen 2, T15 Gen2 and P15s Gen 2. It is marked as recommended, but the P15 Gen 2 variant (which updates the BBR to the same version) is marked as Critical.

P15 Gen 2, P16 Gen2 and T15g Gen 2 have [similar firmware available](https://support.lenovo.com/us/en/downloads/ds553088-thunderbolt-retimer-firmware-update-for-windows-10-64-bit-version-1809-or-later-thinkpad-p15-gen-2-p17-gen-2-t15g-gen-2), not sure if they are packaged, but since they are marked as Critical, they probably should.

mrhpearson, 2024-03-14:
Thanks for flagging this. I've raised an internal ticket (LO-2917) to raise this with the FW team.

vifino, 2024-04-19:
Hey @mrhpearson, any update on this?

mrhpearson, 2024-04-19:
Still being looked at by the product FW team I'm afraid - I've sent a nudge.

vifino, 2024-06-26:
@mrhpearson Any news on the ticket?

mrhpearson, 2024-06-26:
FW is built and is under testing. Once that is complete it will be released - no ETA yet, but it's progressing
Mark

vifino, 2024-12-13:
Hey @mrhpearson, could you please make sure this eventually hits?

The update has been released in July 2023, it's time.

mrhpearson, 2024-12-13:
I had an email discussion with the FW team on this, just this morning. The TBT FW team are having a lot of trouble with getting the update working - I've been trying to understand what is going on with their setup but it's not making any sense.
The issue is being worked on, I get weekly updates, it's just not going well I'm afraid :(

vifino, 2024-12-13:
That's.. odd. Really odd.
It was quite literally just downloading, extracting the exe and `fwupdtool install-blob ./<filename>.cap` for my personal device.

Thank you for your persistance and quick response though, I am glad this is not just getting lost.

vifino, 2025-02-13:
Hey @mrhpearson, new year, new poke, any updates?

mrhpearson, 2025-08-29:
Thanks for the nudge - I missed the Feb one.
The FW team seem to have abandoned this one. I have a long email thread internally with all the problems they were hitting and then it just stopped getting replied to.

I'll go and send in a nag - but I don't have many expectations of success. I have no idea why this particular update was so hard to do.

vifino, 2025-08-29:
If it helps, I'll gladly offer my services / my time at no cost, CEST.

Performing the update is so basic, given that it is just another capsule, that I simply do not understand it being any more difficult than any of the other capsules currently being shipped.

I would really like to see this happen.