fwupd/firmware-lenovo #431: Thunderbolt 4 Dock upgrade fails with "SPI state is busy [0x08]" after 15 mins

rvandegrift, 2024-03-14:
I'm trying to update a ThinkPad Thunderbolt 4 Dock using a Z16 Gen1 running fwupd 1.19.14 from Debian trixie.

It tries for 15 mins, but eventually times out.  This happened twice in row, with me powering off the dock and laptop in between.
```
ross@stgulik:~$ fwupdmgr install -v
(fwupdmgr:3666): GLib-DEBUG: 21:01:55.672: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:3668): GLib-GIO-DEBUG: 21:01:55.678: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3666): GLib-GIO-DEBUG: 21:01:55.684: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.684: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.685: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:3666): GLib-GIO-DEBUG: 21:01:55.693: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3666): GLib-GIO-DEBUG: 21:01:55.698: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:3666): GLib-GIO-DEBUG: 21:01:55.698: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.700: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.700: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.700: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.700: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:3666): dconf-DEBUG: 21:01:55.700: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:3666): Fwupd-DEBUG: 21:01:55.701: Emitting ::status-changed() [idle]
0.	Cancel
1.	a083ebc5138e5e071ef7270cc9a8280722cc7adf (Embedded Controller)
2.	0fba4a8bedba53076fc9b147655e59455bc25a87 (MZVL21T0HCLR-00BL7)
3.	36efb79c255f402f619fa9eb53cd659db51f2a04 (System Firmware)
4.	cf9a31bd537d3261c5ae2979f5db1ff4b8c29721 (ThinkPad Thunderbolt 4 Dock)
5.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-5]: 4
0.	Cancel
1.	10.16
2.	10.13
Choose release [0-2]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade ThinkPad Thunderbolt 4 Dock from 10.11 to 10.16?                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Before continuing, ensure the following:                                     ║
║                                                                              ║
║ • Your computer battery life if over 25%.                                    ║
║ • Do not unplug the dock during update.                                      ║
║ • Do not put your computer into sleep or hibernate mode during update.       ║
║ • This update typically may take up to 15 minutes depending on your          ║
║ notebook and dock firmware version.                                          ║
║ • Please ensure fwupd version is 1.9.2 or above.                             ║
║                                                                              ║
║ ThinkPad Thunderbolt 4 Dock and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Fwupd-INFO: 21:02:07.052: downloading https://fwupd.org/downloads/95d80f78213570a9816f2216f4e2d60ab454a463fb960dea932d3d6204376d0f-1016_NoIOT.cab
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:07.063: Emitting ::status-changed() [downloading]
Downloading…             [             /                         ]Fwupd-INFO: 21:02:07.538: download progress: 1%
Downloading…             [                                       ]Fwupd-INFO: 21:02:07.579: download progress: 2%
  ... [snip a bunch of downloading progress]
Downloading…             [************************************** ]Fwupd-INFO: 21:02:07.938: download progress: 100%
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:07.938: Emitting ::status-changed() [idle]
Fwupd-INFO: 21:02:07.938: status-code was 200
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:07.978: Emitting ::status-changed() [decompressing]
Decompressing…           [***************************************](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.243: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [***************************************](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.284: Emitting ::status-changed() [device-busy]
Waiting…                 [***************************************](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.284: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.284: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(92d2211a864a1abdcca3284ef4d61631a07f42e3)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(e05192ed8d0448e4016bf5e77cd97cd7860d8c47)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(e244cfaefc61eb7670b2fbd6584c6735068d4aef)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(27581012bd0cb5fbea0bcd48f66047fabf37c2ee)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(f7842176e69c01741111688a90b3f2e6eeaab373)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(a0693101b39156a6e6f529fc5efe86536034adf0)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(dd265396ced1576c38814fd623f9e607fe24333d)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(086e8a873abdd916c658621a19d1f5f272419253)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.298: Emitting ::device-changed(33fd60b075cf330b386a42c4e11c41648b986491)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.299: Emitting ::device-changed(65dd15e1f36402259279cf61ec7886779e56173f)
(fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.308: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.308: Emitting ::status-changed() [device-erase]
Erasing…                 [                                       ](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.308: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.308: Emitting ::status-changed() [decompressing]
Decompressing…           [                                       ](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.309: Emitting ::status-changed() [device-busy]
Waiting…                 [                                       ](fwupdmgr:3666): Fwupd-DEBUG: 21:02:08.329: Emitting ::status-changed() [device-erase]
Erasing…                 [                                       ](fwupdmgr:3666): Fwupd-DEBUG: 21:02:39.738: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)
(fwupdmgr:3666): Fwupd-DEBUG: 21:17:39.747: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)
(fwupdmgr:3666): Fwupd-DEBUG: 21:17:39.775: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)
(fwupdmgr:3666): Fwupd-DEBUG: 21:17:39.793: Emitting ::device-changed(cf9a31bd537d3261c5ae2979f5db1ff4b8c29721)

failed to wait for erase: failed after 30 retries: SPI state is busy [0x08]
```

rvandegrift, 2024-07-03:
From a suggestion at [1], I was able to workaround this by updating 10.11 -> 10.13 -> 10.16.  It'd be nice if fwupd could consider this constraint on updates, or at least present a release note to the user.

[1] - https://forums.lenovo.com/t5/Debian/Thunderbolt-4-Dock-upgrade-fails-with-SPI-state-is-busy-0x08-after-15-mins/m-p/5307328

hughsie, 2024-07-25:
I've asked Lenovo to add the requirement of >=10.13 on the 10.16 firmware -- which will force fwupd to do it in two steps.

soltysh, 2024-09-05:
I'm experiencing this exact problem when trying to update my dock on a t14 gen 5 with amd from 10.16 to 10.18.

tomodachi, 2024-11-06:
I have the same issue even if I'm trying to upgrade
10.6 -> 10.13 before going to 10.16

"failed to wait for erase: failed after 30 retries: SPI state is busy [0x08]"

I'm on a Lenovo Thinkpad Z16 
Fedora 41 Kernel 6.11.5-300

taisph, 2024-12-19:
We're unable to update from 10.16 to 10.18 with fwupd 1.9.24 on Ubuntu 24.04 on a Thinkpad P14s with this exact error.

matthewsecaur, 2024-12-20:
Unable to update from 10.16 to 10.18 with fwupd-1.9.27 on Fedora 41 on a ThinkPad P16v with this same error.

DelayLama, 2024-12-23:
I currently have the exact same problem with an E16 Gen2 (Intel, 21MA) and my ("new" , refurbished) 40B0 TB4 dock currently running 10.17 firmware version.

chaincrucial, 2025-01-06:
Same issue with 10.15 -> 10.16 on P16s.

TrevorBenson, 2025-01-06:
Same error trying to upgrade 10.15 -> 10.16 on a P16s, fwupd 1.9.27 and kernel 6.12.7.

Haewelmann, 2025-01-09:
I have also seen the error twice:

fwupd: 1.9.27
Kernel: 6.8.12
Device: T16 Gen 3
Dock1: 10.11 -> 10.13
Dock2: 10.18 -> 10.18 (reinstall with PCB v4 Dock)

managervcf, 2025-04-10:
```
compile   org.freedesktop.fwupd         2.0.7
```

```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          7073f7228b6ee3d493dfaa222ede7b68832d676e
│ │   Current version:    10.7
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      1S40B0ZDZA22EV
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │                       ce170cf8-f4a5-52d6-8a6c-f206071a70ba ← USB\VID_17EF&PID_30B4&CID_40B0&DMCVER_10.7
│ │                       4097279c-ef84-58dc-9e12-4d01b1bee044 ← USB\VID_17EF&PID_30B4&CID_40B0&REV_0040
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   Device Requests:    • Message
```

When I tried to update to 10.18, it returned
```
Not compatible with firmware version 10.7, requires >= 10.16
```

then when I tried to update to 10.16, it returned
```
Not compatible with firmware version 10.7, requires >= 10.13
```

next, I tried to update to 10.163, it returned
```
Not compatible with firmware version 10.7, requires >= 10.11
```

And there is no 10.11 available on LVFS anymore. Does anyone know how I can find this version or what can I do in this situation?

hughsie, 2025-04-11:
10.11 was never made public on the LVFS I'm afraid. If it helps, I'm not very happy with the process either.

managervcf, 2025-04-12:
> maybe with windows https://download.lenovo.com/pccbbs/mobiles/tbt4dkfw1011_8.exe

I've managed to update it on Windows. 

Firstly, I installed 10.9. Then successfully straight to latest 10.18. Thanks for help.

```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          7073f7228b6ee3d493dfaa222ede7b68832d676e
│ │   Current version:    10.18
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      XXXXXXXXXXXXXXX
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │                       9f163b55-830f-5baf-99de-b197219e34e8 ← USB\VID_17EF&PID_30B4&CID_40B0&DMCVER_10.18
│ │                       4097279c-ef84-58dc-9e12-4d01b1bee044 ← USB\VID_17EF&PID_30B4&CID_40B0&REV_0040
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   Device Requests:    • Message

billyvg, 2025-04-29:
I have 10.7 as well, my updates on Windows just stall out. When I try to use fwupd on macos, it says there are no available updates

```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          6cb4e69ad6ba19631db59c938866bbf97b25b8b6
│ │   Current version:    10.7
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │                       ce170cf8-f4a5-52d6-8a6c-f206071a70ba ← USB\VID_17EF&PID_30B4&CID_40B0&DMCVER_10.7
│ │                       fe1be6a9-3027-5135-8e1b-737015e2b5dd ← USB\VID_17EF&PID_30B4&CID_40B0&REV_0042
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   Device Requests:    • Message
```

taisph, 2026-01-09:
> We're unable to update from 10.16 to 10.18 with fwupd 1.9.24 on Ubuntu 24.04 on a Thinkpad P14s with this exact error.

Managed to update from 10.16 to 10.18 on Ubuntu 24.04 on a ThinkPad P15 Gen 2 using the snap version of fwupd:

❯ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.18
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            7887
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.14.0-37-generic
runtime   org.freedesktop.fwupd         2.0.18

Note that the snap version has problems updating UEFI related components so I can't currently recommend using it for anything but this specific fw upgrade.

Pakequis, 2026-08-27:
Adding some data from testing on a ThinkPad Thunderbolt 4 Dock (40B0).
I reproduced this exact error repeatedly, and can confirm the
Windows-fixes-it path @managervcf reported above
(https://github.com/fwupd/firmware-lenovo/issues/431#issuecomment-2798932752)
works (using a different Windows tool than they did). The new part here
is going back to Linux afterward to see what happens next — it's not a
clean fix.

Reproduced identically, 6/6 times on Linux (fwupd 2.0.8,
Ubuntu-based), starting from firmware 10.18: failed to write-firmware: failed to wait for erase: failed after 30 retries: SPI state is busy [0x08]. Happened on downgrade attempts (10.16, 10.13) and even on a
same-version reinstall (10.18→10.18) — same failure regardless of
target version or direction. Dock stayed healthy and fully functional
after every failure, no recovery action needed.

What actually got past it: installed a same-day-published 10.20
release via a standalone Windows utility
(https://support.lenovo.com/us/en/downloads/ds554336 — ThinkPad TBT4 Dock Firmware Update Utility, not Lenovo Vantage — a separate .exe),
which completed successfully with no erase-busy error at all.

Back on Linux afterward, I ran 4 more real fwupdmgr install attempts
from that fresh 10.20 state, in this order:

Downgrade 10.20→10.18 — succeeded (version really changed,
fwupdmgr get-history reports Success)
Downgrade 10.18→10.16 — succeeded the same way
Downgrade 10.16→10.13 — failed for real (see below)
Upgrade 10.16→10.18 — the direction that actually matters for
real-world use — succeeded the same way as 1 and 2
One important gotcha for anyone else testing this: all 3 of those
successes (1, 2, and 4) still printed an error at the end of the live
fwupdmgr install output — failed to wait for cleanup replug: device ... did not come back — even though the firmware version had
genuinely changed each time. Don't trust the live output/exit code
alone; check fwupdmgr get-devices/get-history afterward, since a
"failure" there can be purely cosmetic (the device came back after
fwupd's own wait window gave up).

Attempt 3 (the one real failure) had yet another distinct message:
failed to detach: The new FW will take effect automatically when the dock cable is disconnected from and connected to the computer again.
I disconnected/reconnected the cable exactly as instructed, and it
still did not resolve it — version stayed on the older firmware.

That message matches the official README for the tbt4dkfw1020
(v1.0.20) Windows utility
(https://download.lenovo.com/pccbbs/mobiles/tbt4dkfw1020_2.html), which
documents the cable replug for "Phase 2" as the dock's original
intended design, not a bug: "Under the original design, Phase 2 would
only start after the user unplugged the upstream cable" (v1.0.20 only
automates that away for docks on firmware v1.0.18 with serial numbers
in two specific ranges — ZDXL0001–ZDXMZZZZ, ZDXQ0001–ZDXRZZZZ — my
dock's serial wasn't in either range). So the request itself is
expected/documented behavior, but in my case following it exactly as
documented still didn't get the new firmware to actually take effect —
that part is still unexplained.

So at least for this unit: the erase-busy state seems tied to something
in the old firmware image rather than being unconditionally permanent —
a full reflash via a different tool got it unstuck — but it's not a
universal fix either, since a different failure mode showed up right
after.

mrhpearson, 2026-08-28:
I'll review this properly tomorrow (my brain is dead right now) - but please **don't** do FW downgrades on the dock.

We don't test it and you run a real risk of bricking it. Going all the way back to 10.13 is really not a good idea either FWIW

My understanding is the latest version of fwupd has some fixes in it. I'll check and confirm on the details. 
If you're having issues I'd recommend trying a Fedora live USB

Mark