fwupd/firmware-lenovo #432: fwupdmgr security reports disabled SPI write protection

lunarlettuce, 2024-03-18:
**Describe the bug**
fwdupmgr security reports that SPI write protection is disabled, when it should be enabled. Additionally, CET OS support is marked as "Not supported".

**Steps to Reproduce**
Run `fwupdmgr security`

**Expected behavior**
It is expected that SPI write protection is enabled.

**fwupd version information**

```compile   org.freedesktop.fwupd         1.9.15
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.15
runtime   org.kernel                    6.7.9-200.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Fedora Silverblue Flatpak Repository
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```LENOVO 20XF004RUS
│
├─Lenovo USB-C Mini Dock:
│ │   Device ID:          da77984c82b59c6fc69516431f467fd9a8d39a7f
│ │   Summary:            USB 3.x hub
│ │   Current version:    4.154
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              fd4b20d3-2612-5743-ad85-5c3065361c51
│ │                       f281c1df-c3d5-5f8a-984d-e9548ffc95fe ← USB\VID_17EF&PID_3094
│ │                       ce8b3f6c-9ddd-5d50-b3f8-e87e72d2aacc ← USB\VID_17EF&PID_3094&HUB_0012
│ │                       e62c5403-daa6-5482-9e9e-74666884ce43 ← USB\VID_17EF&PID_3094&SPI_C223
│ │                       75b11f2d-86b6-5ecc-912e-a2a649f334d5 ← USB\VID_17EF&PID_3094&SPI_C223&REV_04F4
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Unsigned Payload
│ │ 
│ ├─Lenovo USB-C Mini Dock:
│ │ │   Device ID:        983c3cffc6fd36d32b00b62928d30721eaeb93db
│ │ │   Summary:          USB 3.x hub
│ │ │   Current version:  4.154
│ │ │   Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration: 15 seconds
│ │ │   GUIDs:            fd4b20d3-2612-5743-ad85-5c3065361c51 ← USB\VID_17EF&PID_3095
│ │ │                     2b337b4f-fc17-520d-8d93-095a9bfd6ba8 ← USB\VID_17EF&PID_3095&HUB_32
│ │ │                     152db1ae-acd6-5b6d-aad2-178ec2af5199 ← USB\VID_17EF&PID_3095&SPI_C223
│ │ │                     8ce1ac09-39f9-51a5-9468-74433dfa575f ← USB\VID_17EF&PID_3095&SPI_C223&REV_04F4
│ │ │   Device Flags:     • Updatable
│ │ │                     • Cryptographic hash verification is available
│ │ │                     • Device stages updates
│ │ │                     • Device can recover flash failures
│ │ │                     • Unsigned Payload
│ │ │ 
│ │ ├─Lenovo USB-C Mini Dock:
│ │ │     Device ID:      d0950b8556ed65b4b8e8bfa3809fdb849005f298
│ │ │     Summary:        USB 3.x hub
│ │ │     Current version:4.93
│ │ │     Vendor:         VIA Labs, Inc. (USB:0x17EF)
[... 259 lines cut ...]
      GUIDs:              fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
                          12c86995-0b90-5ec5-98f3-7a6ed4ca50e0 ← NVME\VEN_15B7&DEV_5006&SUBSYS_15B75006
                          06b4e2aa-91af-508b-b06e-65e3b3189e97 ← WDC PC SN730 SDBQNTY-512G-1001
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • System Firmware (0.1.27 → 0.1.28)
 • UEFI dbx (371 → 371)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.

```

</details>

**Additional questions**

- Operating system and version: Fedora Silverblue 39
- Have you tried rebooting? Yes, and I have tried resetting BIOS to defaults, and resetting secure boot keys to defaults.
- Is this a regression? Unsure if this is an issue with the laptop, or with fwupd.

lunarlettuce, 2024-03-18:
Good to know, thanks for transferring

JimOswick, 2025-11-06:
Experiencing the same issue on my ThinkPad P14s Gen 2a (AMD)

isobeyan, 2026-02-01:
> I also have this same issue on my Thinkpad T14 gen 1 AMD. It marks SPI write protection as disabled, but provides no way of enabling it. Not even in the BIOS.

experiencing the same issue here.

my spec:
# System Details Report
---

## Report details
- **Date generated:**                              2026-02-02 00:09:27

## Hardware Information:
- **Hardware Model:**                              Gigabyte Technology Co., Ltd. B550M DS3H
- **Memory:**                                      16.0 GiB
- **Processor:**                                   AMD Ryzen™ 5 5600G with Radeon™ Graphics × 12
- **Graphics:**                                    AMD Radeon™ Graphics
- **Disk Capacity:**                               1.2 TB

## Software Information:
- **Firmware Version:**                            F18c
- **OS Name:**                                     Fedora Linux 43 (Workstation Edition)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               49
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.18.7-200.fc43.x86_64

CEbhNwPM, 2026-10-11:
I have this issue on my Thinkpad T14 Gen 2 as well. For me, this is the only feature listed by fwupd (aside from "Hardware disk encryption") that is disabled. So clearly, hardware security is taken seriously on Thinkpads by Lenovo.

`flashrom --programmer internal` reports "State of SpiAccessMacRomEn or SpiHostAccessRomEn prohibits full access", so there is some kind of protection in place.

Considering all this and the extensive security settings available in the UEFI, this looks like a bug to me, either in the firmware or fwupd.

I was unable to find a way of contacting Lenovo to report this. Does anyone have means to contact them?

Sapphire5381, 2026-10-15:
>      I was unable to find a way of contacting Lenovo to report this. Does anyone have means to contact them?

Perhaps the Lenovo Support Portal or forums. If you have a business/warranty contract you could escalate through that.