fwupd/firmware-lenovo #435: Updates not installed on reboot: failed to run update on reboot

devriea, 2024-03-30:
I just got a Lenovo T480 and have been trying to update the firmware on it. Almost everything updated successfully except for the embedded controller firmware. I came across [this issue](https://github.com/fwupd/firmware-lenovo/issues/411) which might be related? But I am seeing `fwupd-efi 1.4` when I reboot, and the I do see a splash screen saying it's updating.

My machine is a Lenovo ThinkPad T480, but I am missing both batteries (so I leave it plugged in).

- I run `fwupdmgr update`
- It offers `Upgrade Embedded Controller from 0.1.11 to 0.1.21?`
- I say yes, let it download etc.
- reboot machine
- see `fwupd-efi 1.4`, see a splash screen that indicates it's installing
- boot back into Linux Mint, run `fwupdmgr get-updates`
- see it was not installed correctly

output of `fwupdmgr get-history` is
```
20L6SDKD00
│
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Previous version:   184.55.3510
│ │   Update State:       Success
│ │   Last modified:      2024-03-30 21:59
│ │   GUID:               dd31d983-cf3b-4c84-acc9-70caa94e827b
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T480:
│       New version:      184.94.4494
│       Remote ID:        lvfs
│       Release ID:       57497
│       Summary:          Lenovo ThinkPad T480 Corporate ME Firmware
│       License:          Proprietary
│       Size:             7.5 MB
│       Created:          2023-08-08
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N24RM22W
│       Vendor:           Lenovo
│       Description:      
│       • 8 Intel Platform Update 2023.3 Version 11.8.94.4494 (LVFS: 184.94.4494)
│       
│       Problem Fixes Mitigated the following security vulnerabilities:
│     
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   0.1.19
│ │   Update State:       Success
│ │   Last modified:      2024-03-30 21:59
│ │   GUID:               dc7f0308-1ef8-4774-9ba5-89a58c4d731c
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T480:
│       New version:      0.1.50
│       Remote ID:        lvfs
│       Release ID:       74178
│       Summary:          Lenovo ThinkPad T480 System Firmware
│       License:          Proprietary
│       Size:             9.5 MB
│       Created:          2023-10-30
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Lenovo ThinkPad T480 System Firmware
│       
│       • Enhancement to address security vulnerability.
│     
├─Thunderbolt host controller:
│ │   Device ID:          56aa92974f8b5ada662aa41b2e82cbd25853d9eb
│ │   Previous version:   14.00
│ │   Update State:       Success
│ │   Last modified:      2024-03-30 22:00
│ │   GUID:               8487e4cb-fe6b-5cb6-ad88-46535fb110e1
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Hardware is waiting to be replugged
│ │                       • Device stages updates
│ │ 
│ └─ThinkPad T480:
│       New version:      23.00
│       Remote ID:        lvfs
│       Release ID:       26905
│       Summary:          Lenovo ThinkPad T480 Thunderbolt Firmware
│       License:          Proprietary
│       Size:             262.1 kB
│       Created:          2023-03-23
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       This stable release fixes the following issue:
│       
│       • Fixed Thunderbolt vulnerability.
│       • Intel NVM version update to 23.
│       • PD FW version update to F407.06.33.
│     
└─Embedded Controller:
  │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
  │   Previous version:   0.1.11
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2024-03-30 23:23
  │   GUID:               b87a926d-189c-49a1-b1ff-921099b1de89
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480:
        New version:      0.1.21
        Remote ID:        lvfs
        Release ID:       6890
        Summary:          Lenovo ThinkPad T480 Embedded Controller Firmware
        License:          Proprietary
        Size:             791.7 kB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad T480 System Firmware
        
        Fixed an issue where ThinkVision T24m-10 monitor might not connected properly.
```

I am happy to answer any other questions or run more diagnostics.

ratman528, 2024-03-31:
I also tried downloading version 0.1.14 (the oldest listed on LVFS) and manually installing via `fwupdmgr install ...`. That appeared to run just as successfully as the 0.1.21 update did (even did some BIOS beeps while it was happening (which didn't happen with 0.1.21)), but that also failed.

ratman528, 2024-04-02:
Also, I forgot to include my version info. I'm not sure which version is relevant, so here are all of them.

```
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-101-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

kmauleon, 2024-04-02:
hi @ratman528 ... you may want to install latest fwupd via snap... hope this helps... 

apt remove fwupd
snap install fwupd

fwupdmgr refresh --force
fwupdmgr update     (please update one firmware at a time especially for BIOS, EC, and MEFW)
restart

ratman528, 2024-04-02:
hi @kmauleon, thanks for the response! One of the batteries for this computer arrived today, and I figured I would try once more with a battery before attempting to update fwupd. I reran `fwupdmgr get-updates`, reran the firmware update, and this time it worked. I don't think I changed anything except adding a battery to my laptop. I don't see why that would matter, but apparently it did.

Thank you for the suggestion, and I will report back if it turns out the issue isn't actually fixed!

closed 2024-04-02 by ratman528

kmauleon, 2024-04-02:
thanks for the feedback @ratman528  firmware update especially for BIOS or EC requires at least 25% battery life...

ratman528, 2024-04-02:
Yeah no problem. Any thoughts on why it needed a battery installed to successfully update? If you're plugged in to AC power it's not even using the battery to power anything.

kmauleon, 2024-04-02:
I am not sure why though but It's one of their requirements >> - You need an AC adapter and a charged battery pack
sorry not that helpful

ratman528, 2024-04-03:
No worries. Thanks for the responses, I appreciate it!