fwupd/firmware-lenovo #437: Cannot boot to bios after fw update

ht990332, 2024-04-13:
[devices.txt](https://github.com/fwupd/firmware-lenovo/files/14968331/devices.txt)
This computer was running Arch Linux for about 7 years now without fwupd installed. I installed KDE Neon (an ubuntu base distribution).
It installed "secure boot dbx update". Now I cannot enter the bios again with DEL, F2, or ESC.
systemctl reboot --firmware-setup says
"cannot indicate to efi to boot into setup mode firmware does not support boot into firmware".
I've been booting to bios since I got this machine in ~2016. This is essentially disastrous. Fortunately I am still able to boot KDE neon so I still have a working computer for now.
Any idea how to fix this?

ht990332, 2024-04-14:
In the meantime, could we drop further updates to this device till an update fixes the issue?

mrhpearson, 2024-04-15:
I don't recommend it, but I guess you can delete your dbx from the BIOS setup menu and that would allow Arch to boot again.

I think the problem is your Arch grub is marked as insecure, and is blacklisted in the updated dbx.
If you do get it booting again by clearing the dbx I would update your grub install and then reinstall the dbx ASAP.

Strange though - in my experience fwupdmgr usually flags that you have insecure grub's installed, and recommends you don't go ahead with the dbx update until that is resolved. Did you not get that notification

ht990332, 2024-04-15:
- How do I access the BIOS setup menu? I can't get F1, F2, F10, F12, DEL, or ESC to do anything anymore unfortunately. It just boots KDE NEON.
- I didn't get any warning that something was insecure or dangerous.

mrhpearson, 2024-04-15:
Which platform do you have? F1 normally get's you the F1 setup screen and F12 the boot menu
Mark

ht990332, 2024-04-15:
A lenovo desktop skylake core i5. See the devices attachment from the first comment please.

mrhpearson, 2024-04-15:
That's an Ideacenter 300-20ISH right?

Afraid I have no experience with those - they're not in the Linux program. I'm surprised you were offered a FW update from LVFS for that system.

Looks like FN+F2 gets you into the BIOS setup menu (https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/300-series/300-15isk/80rs/solutions/ht500216-recommended-way-to-enter-bios-ideapad ).
I don't know if you'll be able to modify your dbx from there though. That platform isn't showing up in the BIOS simulator.

ht990332, 2024-04-15:
There is no FN key. This is not a laptop. It's a desktop as stated earlier.
F2 doesn't do anything as I already mentioned.

ht990332, 2024-04-15:
Correct, it is That's an Ideacenter 300-20ISH.

ht990332, 2024-04-16:
Hello. Would you be able to advise me how to proceed from here? Is this repairable?
Thank you.

mrhpearson, 2024-04-16:
Recommend contacting Lenovo support as they'll have experience with this platform. Afraid I don't have access to one to do any investigation with.

hughsie, 2024-04-17:
> I'm surprised you were offered a FW update from LVFS for that system.

We're offering the update to all systems -- like Windows update would presumably do too. There are a list of blocked devices, initially provided by Microsoft: https://github.com/fwupd/fwupd/blob/main/plugins/uefi-dbx/uefi-dbx.quirk -- and if you have any update to that I'd be very welcome to add more devices there.

ht990332, 2024-04-21:
Thank you. Would you be able to provide me with a direct support email address? You can send it to the email address in my profile or type it here. Thank you.

mrhpearson, 2024-04-21:
Afraid I don't know, I'm on the engineering side, not support. Probably best to start here?
https://support.lenovo.com/ca/en/contact-us

ht990332, 2024-05-13:
A local computer technician fixed this for me. This is the same store I bought this computer from in ~2016. To avoid something like this hitting anyone else in the future with the same hardware, would you be able to block this update for this motherboard model?

Thank you.

hughsie, 2024-05-14:
> would you be able to block this update for this motherboard model

Can you attach the output of `sudo fwupdtool hwids` please.

ht990332, 2024-05-14:
[hwids.txt](https://github.com/fwupd/firmware-lenovo/files/15311133/hwids.txt)
Done. Thank you.

ht990332, 2024-05-15:
Let me know please when the change is live so I can close this bug report.
Thank you.

hughsie, 2024-05-17:
@mrhpearson you're okay if I block all `Manufacturer: LENOVO`+`Family: ideacentre 300-20ISH` from the dbx update?

mrhpearson, 2024-05-17:
As I have no interface to that team I have no authority or mechanism to get it fixed - so if it's breaking stuff seems wise to me...but more a personal opinion than 'Lenovo guidance' I'm afraid.
I'll ask the ThinkCenter team if they have a way to forward it to them, and if I can find a way to get it fixed we can revert the block...

ht990332, 2024-05-17:
Also note that not all the damage was reverted. I can now press F1 to boot to BIOS but `systemctl reboot --firmware-setup` still thinks I cannot boot to bios. Also, after some searching on Google, it appears that the DBX update was mainly targeted towards Windows 8/10/11 capable systems. This is 6th generation Intel Skylake processor from the Windows 7 era. While this is unrelated, it does mean that the update was never meant for my desktop.

hughsie, 2024-05-18:
> it does mean that the update was never meant for my desktop

Nah, the dbx update is industry wide. Last years update has been deployed over 7 million times and we've only had a handful of failures.

mrhpearson, 2024-05-21:
Just as some notes on my end, as I got a reply from the Ideacenter team:
 - Product has been EOL and the BIOS maintenance was stopped.
 - Ideacenter 300-20ISH never released the DBX related updates.
 - I got some additional notes on they couldn't understand why the DBX would cause problems booting - it shouldn't. Not much we can do about it without further debugging, and as it's out of support I wouldn't spend time on that.

Summary - Blocking the update is the right thing to do. Hoping this doesn't show up on other older platforms...

ht990332, 2024-05-21:
Some updates:

- the store that fixed my desktop said that their connection at Microsoft told them that such update was not offered to my computer model when running windows 8.1 or 10.
- There were indeed many affected people. most just didn't notice because the current OS just continues to boot as normal. you only lose the ability to boot to firmware or usb/cdrom/dvd media. if you look on google, several linux mint posts complain about systemctl reboot --firmware-setup telling them their hardware doesn't support booting to bios. otherwise, their current mint installation continues to work. People on ubuntu can simply upgrade forever without the need to reinstall. They might not know their computers were hit with this issue.
- After fixing the computer, I installed windows 10 and i indeed was not offered any such update.