fwupd/firmware-lenovo #438: T14 g3 21AH has IME listed twice in get-devices

gjotten, 2024-04-15:
Since last week I've been having problems with my laptop not being powered anymore on lid open. Got a battery problem listed in UEFI diagnostics which I find mildly suspicious on a year-old machine not showing issues the days before but hey, them's the breaks.

AFAICS all this was not triggered by me doing anything out of the ordinary in the first place. I did run `fwupdmgr get-devices`, `fwupdmgr refresh`, `fwupdmgr get-update`, and `fwupdmgr update` upon encountering this problem - just to be sure I had an up-to-date system.

Ever since I´ve been getting:

```
Boot Configuration has been recovered by restoring backup.
Please confirm BIOS setup , and change setting if necessary.
```

And since this morning also the dreaded `Recovering ME Firmware...[...]` into `Failed to recover ME Firmware...` - documented in other open as well closed issues here.

The above triggered me to reinstall both System Firmware as well as IME Firmware based on solutions listed in these issues. I did this multiple times and also in reverse order to see if one maybe influenced the other.

Color me surprised when I just now discovered IME is listed twice as a device. How the heck do I handle the below? Could it be influencing the problems I seem to be having??

```
21AHCTO1WW
│   
├─Intel Management Engine:
│     Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               cb05acd1-54fe-4afe-958d-b1d515825077
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               23192307-d667-4bdf-af1a-6059db171246
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```

hughsie, 2024-04-17:
@kmauleon seem unexpected.... :/

kmauleon, 2024-04-18:
hi @hughsie I will check which ODM handles this system and inform them...

gjotten, 2024-04-18:
in the meantime I had to RMA this system so consider my input finished

kmauleon, 2024-05-06:
ODM owner already implemented CHID restrictions for T14 g3... closing this ticket

closed 2024-05-06 by kmauleon