fwupd/firmware-lenovo #440: 30GS001SMT (LENOVO_MT_30GS_BU_Think_FM_ThinkStation P3 Tower) does not update Intel  ME

joakim-tjernlund, 2024-04-16:
```
fwupdmgr update 
Devices with no available firmware updates: 
 • SKHynix HFS001TEJ9X102N
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 16.1.25.2091 to 16.1.30.2307?           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Initial Production ME release.                                             ║
║                                                                              ║
║ 30GS001SMT must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                      
```

Then I reboot and I only see the Lenovo Logo for a few secs and then system boot normally.

Latest System FW: 0.1.97

hughsie, 2024-04-17:
What fwupd-efi and gnu-efi versions?

joakim-tjernlund, 2024-04-17:
gnu-efi-3.0.18 and fwupd-efi git from 15/04/24

joakim-tjernlund, 2024-04-18:
> Could you confirm from the BIOS setup screen which ME version is installed please? 

It says: ME Firmware Version  16.1.25.2091
Still reporting the old version it seems.

joakim-tjernlund, 2024-04-18:
OS is latest Gentoo, fwupd 1.9.14 and 1.9.16
None of them works

mrhpearson, 2024-04-18:
Thanks - created internal ticket LO-2990 and asked the FW team for input.

joakim-tjernlund, 2024-04-22:
> What fwupd-efi and gnu-efi versions?

FYI, I downgraded to sys-boot/gnu-efi-3.0.17 and sys-apps/fwupd-efi-1.4 and now the ME FW upgrade worked.

mrhpearson, 2024-04-22:
Oh...that's interesting and unexpected. @hughsie - is something you've come across before?

hughsie, 2024-04-22:
That's not expected at all. @joakim-tjernlund would you be able to test the Ubuntu or Fedora gnu-efi and fwupd-efi 1.6? We don't do any testing with Gentoo at all.

joakim-tjernlund, 2024-04-22:
> That's not expected at all. @joakim-tjernlund would you be able to test the Ubuntu or Fedora gnu-efi and fwupd-efi 1.6? We don't do any testing with Gentoo at all.

Sorry, but I am not able to do that. We have a specialised install procedure here and it is for Gentoo only

mrhpearson, 2024-04-23:
FW team confirmed update worked with both Ubuntu and RHEL on their system. I've asked for confirmation on which versions they used of the OS and fwupdmgr.

joakim-tjernlund, 2024-05-02:
A coworker just tried to update a LENOVO_MT_20W0_BU_Think_FM_ThinkPad T14 Gen 2i using
latest versions: fwupd-1.9.18, fwupd-efi-1.6, gnu-efi-3.0.18 with the same result: BIOS skips/fails
to update and just boots normally.

hughsie, 2024-05-02:
@joakim-tjernlund was that gentoo too?

joakim-tjernlund, 2024-05-02:
> @joakim-tjernlund was that gentoo too?

Yes, it it the Gentoo I maintain here

mrhpearson, 2024-05-02:
Any chance we can try a Fedora or Ubuntu live USB on that system? I'd like to rule out if we have a Gentoo issue. If we can reproduce the problem with a distro that the FW team are more familiar with, then it makes a bit easier to get them engaged.

As a note, the FW team tested with Ubuntu 22.04/1.7.9 and RHEL 8.5/1.5.9

Mark

joakim-tjernlund, 2024-05-02:
> Any chance we can try a Fedora or Ubuntu live USB on that system? I'd like to rule out if we have a Gentoo issue. If we can reproduce the problem with a distro that the FW team are more familiar [...]

That is going to be awkward, the guy is mostly working from home

joakim-tjernlund, 2024-05-02:
Did something change in build configuration in fwupd-efi-1.6, gnu-efi-3.0.18 ?
If so, the build spec may need some fixing in Gentoo

joakim-tjernlund, 2024-05-02:
I tried to reinstall Intel ME but get this:
```
fwupdmgr reinstall 484ea4d334faa809334a6cfb8d86f37ca9898dc9
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall Intel Management Engine to 16.1.30.2307?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Initial Production ME release.                                             ║
║                                                                              ║
║ 30GS001SMT must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Decompressing…           [                                       ] Less than one minute remaining…
Failed to copy /usr/libexec/fwupd/efi/fwupdx64.efi to /boot/EFI/gentoo/fwupdx64.efi: Error splicing file: Operation not permitted
```

hughsie, 2024-05-03:
This is fixed in the **very**-recently released 1.9.19

joakim-tjernlund, 2024-05-03:
gnu-efi in Gentoo got an patch: https://gitweb.gentoo.org/repo/gentoo.git/commit/?id=ebaac45475e2dbf861838769c38459dd6275d1b6
Gentoo has gcc 13.2 as stable, maybe it needed some of those changes too ?

So I retested on another T490(LENOVO_MT_20N2_BU_Think_FM_ThinkPad T490) and now System/ME FW installed! :)

Also Thunderbolt vent from ver 21.00 to 23.00, here fwupd reported that TB never came back after update.
Rebooted and TB had ver 23.00 anyway so the update worked in the end.

hughsie, 2024-05-06:
@joakim-tjernlund sanity check; it didn't work with gnu-efi 3.0.18 but did work with gnu-efi 3.0.18+`Align RTLIB CopyMem/SetMem with normal versions`?

joakim-tjernlund, 2024-05-06:
> @joakim-tjernlund sanity check; it didn't work with gnu-efi 3.0.18 but did work with gnu-efi 3.0.18+`Align RTLIB CopyMem/SetMem with normal versions`?

Yes, I think so. Plain 3.0.18 didn't work but  RTLIB CopyMem/SetMem made it work.
I am not in a position to retest back and fourth to really confirm this though.

hughsie, 2024-05-07:
Okay, thanks. I've built the new fixes into Fedora too just in case.