fwupd/firmware-lenovo #441: Failed to recover ME Firmware on Thinkpad T15 Gen2

ilyazat, 2024-04-21:
```
OS: Ubuntu 22.04.4 LTS x86_64 
Host: 20W1S30005 ThinkPad T14 Gen 2i 
Kernel: 6.5.0-28-generic 
Terminal: gnome-terminal 
CPU: 11th Gen Intel i5-1145G7 (8) @ 4.400GHz 
GPU: Intel TigerLake-LP GT2 [Iris Xe Graphics] 
Memory: 4788MiB / 31792MiB 
```

The issue is a kind of same as [this](https://github.com/fwupd/firmware-lenovo/issues/303)

So I try to install update for System Firmware:
```
zzz@notebook:~$ fwupdmgr update  
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • KXG60ZNV512G NVMe TOSHIBA 512GB
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.58 to 0.1.61?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Enhancement to address security vulnerability.                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Updated the Diagnostics module to version 04.33.000.                       ║
║                                                                              ║
║ 20W1S30005 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.

```

So it is success but 
(1) "Reading ME Firmware... Please do not power off! 48% Completed"
(2) "Recovering ME Firmware...Please do not power off! 10% Completed"
(3) "Failed to recover ME Firmware..."
 
 fail to recover still pops up and it repeats every reboot.
 
 NOTE: I did turned off my PC after first time I set up my Ubuntu. I was afraid that the new computer just had a black screen and held the power button to turn it off. 
 
 So what do I do? Will reinstalling the system help?

kmauleon, 2024-04-22:
have you tried this one >> https://github.com/fwupd/firmware-lenovo/issues/303#issuecomment-1373192951

Also please try to update one firmware at a time only
update system firmware > restart system
update EC firmware > restart system
update MEFW > restart system

ilyazat, 2024-04-22:
> have you tried this one >> [#303 (comment)](https://github.com/fwupd/firmware-lenovo/issues/303#issuecomment-1373192951)
> 
> Also please try to update one firmware at a time only update system firmware > restart system update EC firmware > restart system update MEFW > restart system

Yes, I've tried but nothing changed. Could you please consider NOTE? 

> NOTE: I did turned off my PC after first time I set up my Ubuntu. I was afraid that the new computer just had a black screen and held the power button to turn it off.

Can it be a problem?

kmauleon, 2024-04-24:
can you please show screen shot of BIOS Setup Menu @ilyazat

ilyazat, 2024-04-27:
[image]

kmauleon, 2024-04-29:
can you please try to manually install System Firmware as below
download [here](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34ET.firmware)
execute as below
"fwupdmgr --allow -older --allow -reinstall install [cab file]"

if the above procedure did not work, please try to install firmware via USB
[BIOS](https://pcsupport.lenovo.com/ph/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14-gen-2-type-20w0-20w1/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C)

ilyazat, 2024-04-29:
so how do I do it on Ubuntu?

kmauleon, 2024-04-29:
you have to create a [bootable USB](https://download.lenovo.com/pccbbs/mobiles/n34ur28w.iso)
[ReadMe](https://download.lenovo.com/pccbbs/mobiles/n34ur28w.html)

ilyazat, 2024-04-29:
[image]

kmauleon, 2024-04-29:
please allow me to contact ODM owner and will get back to you as soon as possible.... thank you very much!

kmauleon, 2024-04-30:
@ilyazat are you willing to update the system firmware via shell?  ODM owner can give the file.... if your interested, please send me an email >>> kmauleon at lenovo dot com

ilyazat, 2024-04-30:
done sir

ilyazat, 2024-04-30:
i have no problem working with shell. I've already reinstalled my OS but nothing happened

kmauleon, 2024-05-02:
@ilyazat already forwarded the procedure to your email... please check. thank you...

ilyazat, 2024-05-02:
can't find it. You sure?

kmauleon, 2024-05-03:
yes i replied to your gmail il*******at
this is correct right?

kmauleon, 2024-05-06:
sent another mail... please check @ilyazat thanks...

ilyazat, 2024-05-06:
thank you so much

now I got it

ilyazat, 2024-05-12:
For those who follows this thread and met the same problem

After long conversation with the company that makes fw updates I've came to nothing

But problem were resolved somehow after updating OS Ubuntu from 22.04 to 23.10. I don't know how :)

I really appreciate work of support here, thanks @kmauleon

closed 2024-05-12 by ilyazat

kmauleon, 2024-05-13:
thanks for the update @ilyazat !

norbertg1, 2024-10-05:
https://github.com/fwupd/firmware-lenovo/issues/360#issuecomment-2395047396

kmauleon, 2025-11-13:
@weav797 unfortunately the procedure mentioned only works for windows... please install windows to try the procedure

kmauleon, 2025-11-25:
@weav797 really sorry for this... let me try to ask ODM CO if they can still provide EFI package for this one... will get back after hearing from them... thank you

kmauleon, 2025-11-26:
@weav797 message from ODM CO... thank you...

The customer may consider sending the unit to us, as I need to verify whether the EFI package will work—but I currently don’t have an SVT unit available for testing. Moreover, units that have already shipped typically have a locked ME, so I cannot guarantee the EFI-based recovery will succeed.
Alternatively, it would be more reliable if the customer could visit an authorized service center. The engineers there can use an SF600 flash programmer to reprogram only the ME data section of the BIOS image. I’ve encountered this exact issue before, and it was successfully resolved using this method. Since the ME binary contains numerous components, it’s extremely difficult to pinpoint the exact corrupted segment—replacing the entire ME data partition is often the most effective solution.

**Does the customer happen to know the previous ME (Management Engine) version installed on the system?We’ll only be able to pinpoint the root cause accurately once we can reliably reproduce the problem.**

Based on code analysis, ME recovery can fail due to any of the following four conditions:
 
1.The ME Recovery Capsule is not present on the storage device;
2.Memory allocation failure during the recovery process;
3.Failure to read data from the ME Recovery Capsule;
4.ME firmware update failure (e.g., due to verification or flashing errors).
To determine the exact root cause in this case, we need access to the customer’s unit to capture detailed logs during the recovery attempt.