fwupd/firmware-lenovo #442: Ubuntu 22.04 Lenovo Thinkpad X1 Carbon Gen 9 - Cant installed unsigned cab

tyeth, 2024-04-21:
`firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer`

Intel Management Engine 15.0 Firmware (Consumer) Update Utility for Linux - ThinkPad X1 Yoga 6th Gen, X1 Carbon 9th Gen
2.266 MB
v15.0.42.2235
03 Nov 2022

https://download.lenovo.com/pccbbs/mobiles/n32rk04w.txt

[image]

kmauleon, 2024-04-22:
hi @tyeth please download from here >> https://fwupd.org/downloads/b9e159e7c5a68973f76b7e71640563bfd1fb7b494a46103b91ce6c7ace642043-Lenovo-ThinkPad-X1YogaGen6X1CarbonGen9-ConsumerMEFirmware-15.0.45.2411.cab

kmauleon, 2024-04-24:
hi @tyeth have you tried the link?  please advise.  thank you very much!

tyeth, 2024-04-29:
Fails due to SKU bits not matching. Same with official update from update manager

Since being repaired it had the whole mainboard replaced, and I now have a second item in my products list.
The original purchase had no operating system (to spend more money on ram :joy: ) and I think that is an unusual sku or something, or maybe totally unrelated.

[image]
Machine Type Model
20XWCTO1WW

tyeth, 2024-04-29:
<details><summary>output from get-devices:</summary>
<p>

tyeth@ubuntu447:~$ fwupdmgr get-devices
20XWCTO1WW
│
├─ELAN0672:00 04F3:3187:
│     Device ID:          0867c07f040e810fa5ea153981e63e324c29eb1f
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              940805ff-ca7a-56b2-af86-32e995004af9
│                         ab16bce2-e219-52f5-89a6-fd6b1c4244c0
│                         e4c6161e-2b1c-59cb-ad07-c8fb7feb155e
│                         4b808d27-c833-50cc-8177-9a401f5c08fe
│                         5b73c06e-3dfd-5e6c-b06a-24e3489206c2
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Embedded Controller:
│     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│     Summary:            UEFI ESRT device
│     Current version:    0.1.36
│     Minimum Version:    0.1.36
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    61.1
│     Vendor:             Azurewave (USB:0x13D3)
│     GUIDs:              ad93a8b9-464b-5ea6-9b06-59d8605805ea
│                         bc121e30-c172-5bf0-9a06-495355bea278
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
│     Summary:            UEFI ESRT device
│     Current version:    240.42.2235
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               03c27e00-8229-4c14-ba40-2afe2276555c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.4234207
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   GUIDs:              896bb9a6-a8be-5727-8a3a-43ca249f6933
│ │                       448868f0-e05d-5849-8fc4-b8fa1ec16bf5
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
[... 274 lines cut ...]
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "MissingCapsuleHeader" : "False",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "KernelVersion" : "6.5.0-28-generic",
                                 "FwupdSupported" : "True",
                                 "TpmEventLog" : "0x00000003 0000000000000000000000000000000000000000 [U3RhcnR1cExvY2FsaXR5AAM=]\n0x00000007 2d592fad11e8d304837a5188e0abcebc67bb37aa [Qm9vdCBHdWFyZCBNZWFzdXJlZCBTLUNSVE0A]\n0x00000008 1f574150bcf010d92fa433291b2a6cf00dce5b03 [TgAzADIARQBUADkAMQBXACAAAAA=]\n0x80000008 8af38da08ebb933eaf05007ec0037453d95fc840 [AADw/wAAAAAAAAEAAAAAAA==]\n0x80000008 205fecfd374af90f68f26d04f6c4402399a2465c [AADA/wAAAAAAAAgAAAAAAA==]\n0x80000008 c24a462883bd5ee63e4c3f19f50520c205299007 [AAA0/wAAAAAAAEMAAAAAAA==]\n0x80000008 f6a46ee1cb9e965d6610a2420f5ba5fa21c2459b [AACk/wAAAAAAABQAAAAAAA==]\n0x80000008 ec7a67e705bbc255e26e88a42a5f9b77b7946f89 [AACW/wAAAAAAAAoAAAAAAA==]\n0x80000008 1b792e1d644a7deb107f5aa79a07e0cbea73bbe9 [AAB3/wAAAAAAAB8AAAAAAA==]\n0x00000001 bf7439abeb5329a265dfc284b820fd67d126429c [QUNQSSBEQVRB]\n0x00000001 4a412e183453b0ffa8227ca73243c261ef6dcb08 [QUNQSSBEQVRB]\n0x00000011 120a7e4fb3323586615325680a07587c2121f1d0 [SW50ZWxDU3hFRXZlbnQwMQAAAAAEAAAAAAMAADAAAAAXODTWBn3vISrcqw8UEl0AxpU6lR7P8B3f1wM7rNLTQCeh6x3gVLRr/HpVZIhS/44CAwAAFgAAAA8AAAAqALsIAQAAAAYAAAAoAAAAAQACBQAAFgAAAA8AAAAAAFgbAAAAAP//////////AQACLQAAFgAAAAEAAAAAAAIAAQAAAAEAAAAAAAAAAQABAgAAMAAAAEcbM3IZCXshTNhBANwv+tV5kK69fUWjl+PHd3iWEb1TZcsHzxqso2MkLaMTz8EhhAICAAAWAAAAlgABABQAEQQAAAAAAAAAAAAAAAABAAMEAAABAAAAAgMAAAAEAAAAdwAAAAEpAAAwAAAAu1Djsi/uzhltIihV94XDlM45EbMu5+Vz5nGFtzNEgnzRk+8HQzg3qfRiGhP+jxFAAikAABYAAAAFAAQAAgB1EQIAAAACAAAAAAAAAAEAAQYAADAAAAAt32w8qg2tqNpJBb9s8wgeBkSnvGG8B7C1lIYi+A3+/6vXqfaIDXs+AGHY8ZfucdQCBgAAFgAAABEAFwAAAAAAAAAAAAAAAAAAAAAAAQABBwAAMAAAAIPFPeFFecrsEa6s2GMBMqEuLly821Z2Ik1YhbB4/8gqrCrpkJ/Orntd4Ds3VrqlcgIHAAAWAAAACwDhABQB+gcAAAAAAAAAAAAAAAABAAEIAAAwAAAAnPj+SI/KpD/7N8DEY+qLG+SirPF8ETzEknrZA0wm8rKBbOpO9OXZ9VNfRB/dy7GDAggAABYAAAAPAAAAAAD5EQAAAAAAAAAAAAAAAAEAAwMAAAEAAAAA]\n0x00000011 f07df548dca6dc4286ef4ce0d6db12f02ea03e06 [SW50ZWxDU3hFSW5mb0V2ZW50AAABAAAAhoDgoAEAAAA=]\n0x00000004 9069ca78e7450a285173431b3e52c5c25299e473 [AAAAAA==]\nPCR0: 75cffc96e5aad4908a558025754a281e24aa58e9\nPCR0: 0b3cbbaed63a0fe9274037c5eb [... cut ...]
                                 "Pcr0_SHA256" : "0b3cbbaed63a0fe9274037c5eb549f36f6627d160d09ea3a08681b793dd52dd1",
                                 "EfivarNvramUsed" : "42786",
                                 "DistroId" : "ubuntu",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.7.9",
                                 "LastAttemptStatus" : "0x0"
                               }
                             }
                           ]
                         }

</p>
</details>

kmauleon, 2024-04-29:
hmmm ... first time I saw this error...   maybe due to the mainboard replacement?
can you please share BIOS Setup menu screen shot and results of "fwupdmgr get-devices"

tyeth, 2024-04-30:
Get devices is in a collapsed message earlier 
[image]

kmauleon, 2024-04-30:
Oh I think you have to send this system again for repair since your OA3 ID has no value it means your system image is corrupted T_T  ODM owner won't be able to do anything about this I'm afraid.

kmauleon, 2024-05-02:
closing this as can not be fixed via firmware update

closed 2024-05-02 by kmauleon