fwupd/firmware-lenovo #443: Latest T14s gen 3 firmware/X13 gen 3 not available on LVFS

Jackaed, 2024-04-28:
The latest firmware version (1.40) for the thinkpad t14s gen 3 and x13 gen 3 is not available on LVFS at the moment. As of right now, the most up-to-date version on LVFS is 1.35. Please could this update be added to LVFS? The update can be found here: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-3-type-21cq-21cr/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C (note it was released april 14th, 2024)

hughsie, 2024-04-29:
@mrhpearson https://fwupd.org/lvfs/firmware/85577 for your reference.

mrhpearson, 2024-04-29:
Hi ,
1.40 is on LVFS, but still under embargo. I'll need to check with the FW team when it will be released to stable - but it's a holiday in China and Japan this week so I won't get an update until next week I'm afraid.
It should be soon. We've got a number of AMD platforms with FW updates lined up - and their testing, review and release process takes quite a long time. I've not heard any issues with the latest (fingers crossed).

The previous versions all got pulled for various reasons which is why they're not showing up (they're all there - but under embargo).
As a note (mostly for myself) - I'm tracking this through LO-2765. Please nudge me if I haven't done an update by end of next week.

Jackaed, 2024-04-30:
May I ask why it's under embargo on LVFS, given it's been released on the Lenovo website? Does the LVFS release mechanism still need to be tested, or is it just a matter of signing things off? 

Thank you for the response and the update!

Jackaed, 2024-04-30:
Okay, thanks for the response. On a slightly unrelated note - what's an appropriate channel to attempt to report firmware issues and ask for firmware feature requests? Feels like most "official" support channels are more designed for support people attempting to provide tech support rather than for reporting issues directly to a firmware team - but this seems like an innapropriate channel to use to attempt to achieve that, since it's more for reporting issues related to fwpud rather than the firmware itself, but the official form feels like something of a ghost town.

Glad to hear that the processes are improving though.

mrhpearson, 2024-04-30:
For FW update issues via LVFS - here is good. There's a few of us on the Lenovo Linux team who monitor.

For more general issues probably the best is the forums (forums.lenovo.com/linux). It's not official support, but again there's a few of the Linux team who will see any postings in the Linux specific channels. You often get to bypass the official support mechanism where, realistically, they don't have much direct experience with Linux.

Mark

Jackaed, 2024-05-12:
Any updates on this?

Jackaed, 2024-05-13:
Seems to have just been released! Closing, thank you for sorting this out.