fwupd/firmware-lenovo #445: p14s gen 4 AMD missing ECP 1.28  (R2FHT44W)

shtirlic, 2024-05-02:
```
  Package  (ID)            UEFI BIOS  (BIOS ID)        ECP    (ECP ID)         Rev.       Issue Date

  ----------------         --------------------        ----------------        ----       -------------

  R2FUJ55WD                1.35      (R2FET55W)        1.28  (R2FHT44W)        01         2024/03/29

  R2FUJ38WD                1.18      (R2FET38W)        1.16  (R2FHT32W)        01         2024/03/11

  R2FUJ36WD                1.16      (R2FET36W)        1.14  (R2FHT30W)        01         2023/11/24

  R2FUJ33WD                1.13      (R2FET33W)        1.11  (R2FHT27W)        01         2023/09/21

  R2FUJ31WD                1.11      (R2FET31W)        1.09  (R2FHT25W)        01         2023/08/03

```

UEFI updated to 1.35 but ECP still the 1.16

shtirlic, 2024-05-30:
@mrhpearson did you have a chance to look into this?

mrhpearson, 2024-05-31:
Sorry - I hadn't seen that.
EC 1.30 (46W) has been uploaded a few days ago but is under embargo - let's monitor that for going to stable in the next week.
Note BIOS 1.37 (57W) is the same.

shtirlic, 2024-07-18:
```
  └─ThinkPad T14/P14s Gen 4 T16/P16s Gen 2 AMD System Update:
        New version:      0.1.37
        Remote ID:        lvfs
        Release ID:       92344
```

closed 2024-07-18 by shtirlic