fwupd/firmware-lenovo #446: Fails to update ThinkCentre M70q-4 system firmware

orontee, 2024-05-09:
The computer host is a ThinkCentre M70q-4 with version 0.1.31 of the "system firmware".

When I download version 0.1.34 of [ThinkCentre-Tiny9-SystemFirmware-M4VKT22A.cab](https://fwupd.org/downloads/37c79a9279fd118f1eb67dcc5f5e7d134a2a1acaab67e27f6e580b2b3710f51a-ThinkCentre-Tiny9-SystemFirmware-M4VKT22A.cab) and run:
```sh
sudo fwupdmgr --allow-reinstall install  ~/Téléchargements/37c79a9279fd118f1eb67dcc5f5e7d134a2a1acaab67e27f6e580b2b3710f51a-ThinkCentre-Tiny9-SystemFirmware-M4VKT22A.cab
```
a message displays "Successfully installed firmware" and I am invited to reboot the computer, but **on reboot the system firmware is still 0.1.31**.

Same result when updating using `gnome-firmware`...

[firmware-history.txt](https://github.com/fwupd/firmware-lenovo/files/15263076/firmware-history.txt)

hughsie, 2024-05-09:
What does `fwupdmgr --version` say please?

orontee, 2024-05-09:
> What does fwupdmgr --version say please?

```
matthias@peitho:~ 
↳ fwupdmgr --version 
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.6.13+bpo-amd64
```

hughsie, 2024-05-10:
Could you try with fwupd-efi 1.6 and a newer fwupd (1.9.something) please? If that doesn't work we can debug a bit more.

orontee, 2024-05-10:
I installed  a recent `fwupd` using Flatpak:

```
$ sudo -i flatpak run org.freedesktop.fwupd --version
Chargement…              [************************************** ]
WARNING: Ce paquet n'a pas été validé, il peut ne pas fonctionner correctement.
compile   org.freedesktop.fwupd         1.9.19
compile   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.9
runtime   org.freedesktop.fwupd         1.9.19
runtime   org.kernel                    6.6.13+bpo-amd64
```

But I failed to install the `.cab` file:
```
$ sudo -i flatpak run org.freedesktop.fwupd --allow-reinstall install  .var/app/org.freedesktop.fwupd/data/37c79a9279fd118f1eb67dcc5f5e7d134a2a1acaab67e27f6e580b2b3710f51a-ThinkCentre-Tiny9-SystemFirmware-M4VKT22A.cab
Chargement…              [*******************                    ]
WARNING: Ce paquet n'a pas été validé, il peut ne pas fonctionner correctement.
En attente…              [*******************                    ]
missing signed bootloader for secure boot: /app/libexec/fwupd/efi/fwupdx64.efi.signed cannot be found
```

orontee, 2024-05-10:
Finally I managed to solve the missing signed bootloader:
```
[📦 org.freedesktop.fwupd ~]# fwupdtool install .var/app/org.freedesktop.fwupd/data/37c79a9279fd118f1eb67dcc5f5e7d134a2a1acaab67e27f6e580b2b3710f51a-ThinkCentre-Tiny9-SystemFirmware-M4VKT22A.cab 
Chargement…              [*******************                    ]
WARNING: Ce paquet n'a pas été validé, il peut ne pas fonctionner correctement.
Décompression…           [ -                                     ]
Specified firmware is older than the minimum required version '0.1.34 < 65567'
```

And using `get-devices` I see with this strange minimum version:
```
...
─System Firmware:
│ │   Device ID:          8a3bd64788bb0b0e8e4daa89392d48201ee98625
│ │   Résumé:             UEFI System Resource Table device (Updated via capsule-on-disk)
│ │   Version actuelle:   0.1.31
│ │   Version minimum:    65567
│ │   Fournisseur:        LENOVO (DMI:LENOVO)
│ │   État de mise à jour:Success
│ │   GUID:               d64ea3d9-4e0e-4cb5-80d2-8ee22bc34c80
│ │   Drapeaux de périphérique:• Périphérique interne
│ │                       • Mise à jour possible
│ │                       • Le système nécessite une source d'alimentation externe
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
...
```

orontee, 2024-05-11:
> I installed a recent `fwupd` using Flatpak:
>
> ```
> $ sudo -i flatpak run org.freedesktop.fwupd --version
> Chargement…              [************************************** ]
> WARNING: Ce paquet n'a pas [...]
Oh it doesn't tell it's using fwupd-efi 1.6 as you asked 😟

hughsie, 2024-05-11:
> I installed a recent fwupd using Flatpak

Hmm, the flatpak is really designed for peripherals -- I think it might need some testing to be able to deploy capsules. It also won't have a fwupd-efi that's capable of being used with secure boot -- as it's not going to be signed by someone like Red Hat.

orontee, 2024-06-30:
I finally updated through a BIOS updater on a USB key. Had to disable secure boot for this to work. May be it explains this issue...

closed 2024-06-30 by orontee