fwupd/firmware-lenovo #447: Missing firmware 1.38 for Thinkpad P53/P73

Tabby, 2024-05-23:
* Link to firmware update at OEM website
    https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p53-type-20qn-20qq/20qq/20qqs3831w/pf1by1jm/downloads/ds540999
 * Firmware release date
    18 Apr 2024
 * Link to the device page on LVFS
    https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2NETXXW.firmware

mrhpearson, 2024-05-28:
Sorry for the slow reply. Looks like the P-BIOS was released but they missed the W-BIOS.
I've created internal ticket LO-3085 to nudge them.
Thanks for reporting

mrhpearson, 2024-06-12:
FW has been moved to stable - it should be available now (make sure to do a refresh to update your fwupd cache if you don't see it, it was moved only a few hours ago.
Let me know if you see any problems!

closed 2024-06-12 by mrhpearson