fwupd/firmware-lenovo #448: Thinkpad L13 Yoga is missing csme update?

octomike, 2024-06-10:
`fwupdmgr security` reports:

```
[...]
✘ csme v0:15.0.3.1402:           Invalid
[...]
```

```
BIOS Information
        Vendor: LENOVO
        Version: R1FET58W (1.32 )
        Release Date: 04/17/2024
```

```
System Information
        Manufacturer: LENOVO
        Product Name: 20VKCTO1WW
        Version: ThinkPad L13 Yoga Gen 2
        Serial Number: R911TCDM
        UUID: <XX>
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_20VK_BU_Think_FM_ThinkPad L13 Yoga Gen 2
        Family: ThinkPad L13 Yoga Gen 2
```

Is there something else I can do?

mrhpearson, 2024-06-10:
Most strange - can you check what ME FW you have installed from the BIOS setup menu please (F1 during early boot).
Also - do you have the vPro or non-vPro model/
Thanks!
Mark

octomike, 2024-06-10:
Sure thing:

+ ME FW in BIOS: 15.0.3.1402
+ vPro: not sure, I don't think so, it's a `L13 Yoga Gen 2 (type 20VL, 20VK) Laptops (ThinkPad) - Type 20VK` with an i7-1165G7

mrhpearson, 2024-06-10:
OK - so your ME FW is way out of date - that's the problem.

The non-vPro version is in the testing channel. Looking at the history it's bounced back and forth between testing and stable a few times - which I suspect means there are issues with it (It has 32000+ downloads, 6 failed reports but zero success reports)

If you want to try it you can enable the testing channel, or manually download and install https://fwupd.org/downloads/54bca81aa689847426fc3e30c7faffdad0d042f11c0b2e717acc80291f8e9ee6-Lenovo-ThinkPadL13Gen2-ConsumerMEFirmware-240.35.1951.cab

I'll check what the status of it is with the FW team. I do see the newer version under embargo recently so hopefully they're working on it. Internal ticket is LO-3120.

Mark

mrhpearson, 2024-06-12:
Ack - your GUID is different to the version in that FW. Best not to try and force it.

I've added notes to the internal ticket and we should wait for guidance from the FW team. Looking at the latest ME FW on LVFS (under private channel for now - so not available publicly) the GUID used there matches your system and there are no HW ID limitations...so it might just be a case of waiting for that to be released first. But I've asked for confirmation
Mark

kmauleon, 2024-07-11:
ODM CO targeting to release MEFW GUID/version fix by August... thank you very much.
Please ignore MEFW update for now...

octomike, 2024-07-25:
Got an update today, but `fwupdmgr` is not happy about it / thinks it's not installed. The line

```
 Update Error:       failed to run update on reboot: expected 240.49.2573 and got 0.49.2573
```

is quite suspicious however, since the reported `0.49.2573` appears to be a substring of the target version.

```
└─Intel Management Engine:
  │   Device ID:          2713c948a41b00c225275af5b26bddd6735f1653
  │   Previous version:   0.49.2573
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot: expected 240.49.2573 and got 0.49.2573
  │   Last modified:      2024-07-25 09:06
  │   GUID:               bb2acbf1-4063-4816-9041-88bad259367a
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad L13 Gen 2 Consumer ME Update:
        New version:      240.49.2573
        Remote ID:        lvfs
        Release ID:       95975
        Summary:          Lenovo ThinkPad L13 Gen 2 Consumer ME Firmware
        License:          Proprietary
        Size:             4.1 MB
        Created:          2024-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=R1FRN22W
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        Lenovo ThinkPad L13 Gen 2 Consumer ME Firmware

        CSME Consumer15.0.49.2573.bin
        Checksum:         c00e250d76e6f01e62e9a830d90c2f958f2a7d6a018e714a411044e589974fea
```

hughsie, 2024-07-25:
@kmauleon known issue?

kmauleon, 2024-07-25:
hi @octomike , you are already on the latest MEFW...  0.49.2573 is the correct version.... 240.x.xx.xxxx is the wrong format... will clarify with ODM CO

kmauleon, 2024-07-26:
as per ODM CO...
The issue is being clarified,
MEFW2573 has been removed from LVFS web

kmauleon, 2024-09-03:
for MEFW version fix, as per ODM CO, please do the following... thank you very much.

update BIOS to 1.33 first
update MEFW to latest

ChiWei-Chen, 2024-09-19:
Hi,
Both corporate and consumer version 240.49.2573 moved stable:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FR.firmware

kmauleon, 2024-09-19:
Please update BIOS to 1.33 first
then update to latest MEFW : [vPro](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FR.firmware) / [Non-vPro](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1.firmware) versions
thank you very much!

closed 2024-09-19 by kmauleon